stefanpartheym/archlinux-ipu6-webcam #1: Not working on 6.0.9

alba4k, 2022-11-24:
First of all, I know this isn't your work, as you mention in the readme, but maybe you'll be able to help anyway, as you probably have a better understanding than me.

Basically, my webcam still doesn't seem to work.

What I find curious and that might be related is a series of errors that pop up while installing the `v4l2loopback`, `ipu6-drivers` and `ivsc-driver`  modules packages:

```
==> ERROR: Missing dev kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing lib64 kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing lost+found kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing root kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing usr kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing tmp kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing lib kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing mnt kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing sys kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing swapfile kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing var kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing home kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing opt kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing boot kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing run kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing sbin kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing etc kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing srv kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing proc kernel headers for module ipu6-drivers/r67.278b7e3d0.
==> ERROR: Missing bin kernel headers for module ipu6-drivers/r67.278b7e3d0.
```

(same errors, all that changes is the module name and version)

As you probably noticed, the list of elements is simply a list of the files and directories in my root. My competence in kernel module building is close to none.

Also, if I try to load any of those modules, I get nothing (e.g. `modprobe v4l2loopbaclk`: `modprobe: FATAL: Module v4l2loopback not found in directory /lib/modules/6.0.9-arch1-1`; `modprobe -r v4l2loopback`: `modprobe: FATAL: Module v4l2loopback not found.`)

I don't have any /dev/video* virtual devices

Tested on both linux and linux-zen, XPS 13 Plus (i7-1260P)

alba4k, 2022-11-25:
I'm an idiot, I didn't have linux-headers installed

Now I am able to compile the modules and I do have some /dev/video* devices:

```
$ v4l2-ctl --list-devices

ipu6 (PCI:pci:pci0000:00):
	/dev/video1

ipu6 (pci:pci0000:00):
	/dev/media0

Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
```

I still don't get any video output tho (I can select a "Virtual Camera" webcam)

sigboe, 2022-12-12:
I have the same issue, but I did some steps some steps to get further. And I hope that you or someone can see if these steps get you further. And maybe you can see if you can get even further, and document what you did.

I can currently get image from the `test.sh` script, here is what I did.

`/etc/modprobe.d/v4l2loopback.conf`
```
options v4l2loopback exclusive_caps=1
```
Made this file, instead of the instructions @stefanpartheym wrote [here](https://github.com/stefanpartheym/archlinux-ipu6-webcam#make-camera-work-in-chromefirefox-and-electron-based-applications) Its possible to add more settings here if they may help.

The issue is that Firefox and Cheese still does not see the camera.

The original test.sh did not work for me, several error, and I got different errors while trying different variations of the command. What worked for me is to replace the contents of `test.sh` with this

```
#!/bin/sh

sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! waylandsink
```

This works for me, if you have wayland I hope it will work for you too. If you have X11 then I believe you can replace `waylandsink` with `ximagesink` Other laptops with other sensors may require different variations of the command.

I have a Dell XPS 13 Plus 9320 with alder lake

sigboe, 2022-12-12:
I also tested this command as the test command, and it works

```
sudo -E gst-launch-1.0 icamerasrc ! video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! waylandsink
```

notable changes are that I removed `buffer-count=7` and added `,framerate=30/1` which are more similar to how v4l2-relayd standard settings.

I have yet to figure out why the test command now works but not v4l2-relayd

stefanpartheym, 2022-12-13:
Hi @alba4k,

> I still don't get any video output tho (I can select a "Virtual Camera" webcam)

For me, I also can only select "Virtual Camera". How did out test camera input? Using the test.sh script or in a chromium based browser using [this website](https://webrtc.github.io/samples/src/content/devices/input-output/)

Best regards
Stefan

alba4k, 2022-12-18:
I used that website, on firefox tho, and discord (which counts as chromium based ig)

I might try this evening in chromium

stefanpartheym, 2022-12-18:
Hi @alba4k,

I'm gonna be honest:
It doesn't work in Firefox for me either. I didn't notice, because I don't use Firefox. 

Best regards
Stefan

alba4k, 2022-12-18:
HOLY SH*T

[image]

The camera actually works in chromium! I had never seen that light turn on! This basically means that whenever I need it for something I can still use it via chromium!

Also, while still keeping my expectations low, I also tried cheese (it did not work). What was interesting tho was that is also found an "ipu6" device other than a the Virtual Camera one. huh

stefanpartheym, 2022-12-19:
Hi @alba4k,

I'm glad it works for you on Chromium :)

> Also, about the IR camera, do you think [Howdy](https://github.com/boltgolt/howdy) would now work, as that would be the main use I'd have for the cam. linux-enable-ir-emitter seems to turn on the webcam light when using /dev/video0, but I don't see anyhting flashing using my phone's camera.

Unfortunately, I don't know if (or how) Howdy works. I have not yet tried it. I didn't even know such a thing exists. But good hint, I think I'll try it out at some point ;)

Best regards
Stefan

alba4k, 2022-12-19:
@stefanpartheym I would certainty recommend trying it out.

Apart from that, let me know if you (or some other guy(e.g. on the archlinux forums) get the camera to work in more or, ideally, all scenarios.

Congrats on the nice works on creating this repository, also :)

stefanpartheym, 2022-12-19:
Hi @alba4k,

thank you! I'll let you know, if there are any news on this. ;)

I wanted to rename the packages in the repo anyways, as they always appear in the available updates list of my package manager due to the same name of the packages on AUR.

Best regards
Stefan

alba4k, 2022-12-19:
Yeah noticed that yesterday, and unfortunately the updates breaks stuff

maybe just rename the *-git or * to *-fix-git or *-fix?

The best of luck on your continuation!

stefanpartheym, 2022-12-19:
> Yeah noticed that yesterday, and unfortunately the updates breaks stuff

Uff, yeah didn't try to update yet, but good to know, that it's not an option haha.

> maybe just rename the *-git or * to *-fix-git or *-fix?

Exactly, something like that, hopefully I'll have the time for this in the next few days. :grimacing:

sigboe, 2022-12-19:
@stefanpartheym @alba4k Has Howdy been updated to python 3 yet? I uninstalled it when python 2 was deprecated on Arch. Howdy worked quite well for me after it was configured properly, but I did not want to have a security related program depend on python 2 after it was deprecated from the repos

sigboe, 2022-12-20:
Amazing, chromium does indeed work with the webcam. I didn't test it because I used cheese to test instead. 

I wonder why it doesn't work with firefox

z4lem, 2022-12-20:
Frist of all, huge thx to @stefanpartheym for your efforts!
Maybe you (or anyone else) has a clue for the following issue:

My setup: 
* According to service tag my Gen 10 Carbon has the following build-in camera:
FHD RGB + IR Camera with ThinkShutter 
* I run Arch Linux with kernel 6.0.12-arch1-1

I managed to install the repo which ended with successfully starting the v4l2-relayd.service.
This is how my dev outpul liiks line now:

[image]

However, test.sh outputs the following:

```
Setting pipeline to PAUSED ...
Failed to set pipeline to PAUSED.
Setting pipeline to NULL...
Freeing pipeline ...
```

Cheese outputs he following error:

```
(cheese:16705): cheese-WARNING **: 17:50:28.289: 
Das Gerät »/dev/video1« unterstützt keine 2:0:0:0-Farbmetrik: ../gstreamer/subprojects/gst-plugins-good/sys/v4l2/gstv4l2object.c(4125): gst_v4l2_object_set_format_full (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin35/GstV4l2Src:v4l2src1:
Device wants 2:0:0:0 colorimetry
```
Obs outputs the following error:
```
Error: v4l2-input: Initialization failed, errno: Unsuitable IOCTL (I/O-Control) for this device.
```

Despite the Chrome workaround provided, the [test website](https://webrtc.github.io/samples/src/content/devices/input-output/) and BigBlueButton allow me only to select a **Virtual Device** which results in a black screen.

Current workaround: dualbooting Windows for webcam use :'( 
Maybe there's a chance for a better workaround (without having to carry an external one with me).

 Fyi: [Here's](https://lwn.net/Articles/904776/) an article & discussion about this topic...

alba4k, 2022-12-23:
@stefanpartheym it sadly does not seem to work anymore on 6.1

Tropicao, 2022-12-27:
> @stefanpartheym it sadly does not seem to work anymore on 6.1

Hi. I have just finalized the setup on my XPS 13 9320, and indeed I have had to tweak the ipu6_driver installation, because its build is currently broken on 6.1. [The fix is ready to merge on Intel's repository](https://github.com/intel/ipu6-drivers/pull/67) but it seems to be stalled :shrug: 
While it is not merged, it is possible to pick a working copy of the driver by taking directly the source from [the fix author repository](https://github.com/jwrdegoede/ipu6-drivers/tree/ipu6-kernel6.1-build-fix): 
```
diff --git a/intel-ipu6-dkms-git/PKGBUILD b/intel-ipu6-dkms-git/PKGBUILD
index 28e0d82..2238216 100644
--- a/intel-ipu6-dkms-git/PKGBUILD
+++ b/intel-ipu6-dkms-git/PKGBUILD
@@ -1,14 +1,15 @@
 pkgname=intel-ipu6-dkms-git
 _pkgname=ipu6-drivers
 pkgver=r60.4ef83f208
 pkgrel=1
 pkgdesc="Intel IPU6 camera drivers (DKMS)"
 arch=('any')
-url="https://github.com/intel/ipu6-drivers"
+url="https://github.com/jwrdegoede/ipu6-drivers"
 license=('unknown')
 depends=('dkms')
 makedepends=('git')
-source=("git+${url}.git")
+revision="ipu6-kernel6.1-build-fix"
+source=("git+${url}#branch=${revision}")
 sha256sums=('SKIP')
 
 pkgver() {
```
I am able to get a valid stream from `test.sh` with this workaround. Thanks a lot @stefanpartheym for this packaging !

sigboe, 2023-01-02:
@Tropicao Do you want to make a pull request to this repo with this change? It works for me, in fact, both hans' master branch and the ipu6-kernel6.1-build-fix works for me on 6.1.1. But you are maybe correct in using the specific branch. 

Another merge request would be needed to revert this change, if when Intel merges this PR

Tropicao, 2023-01-02:
> @Tropicao Do you want to make a pull request to this repo with this change? It works for me, in fact, both hans' master branch and the ipu6-kernel6.1-build-fix works for me on 6.1.1. But you are maybe correct in using the specific branch in case the master branch is used for something else in the future.
> 
> Another merge request would be needed to revert this change, if when Intel merges this PR

I am not convinced it is desirable to point to [jwrdegoede](https://github.com/jwrdegoede) repository/branches "by default", because it will multiply the sources to monitor for drivers updates (it would be best to assume that its PR on Intel repository will eventually be merged), but if @stefanpartheym thinks otherwise, I'll gladly submit a PR

sigboe, 2023-01-02:
@Tropicao what about temporarily pointing the PKGBUILD file to a specific git commit in the intel repo, and then add Hans' changes as a patch file?

That is how many AUR packages do it anyway

Tropicao, 2023-01-02:
> @Tropicao what about temporarily pointing the PKGBUILD file to a specific git commit in the intel repo, and then add Hans' changes as a patch file?
> 
> That is how many AUR packages do it anyway

I am not very familiar with Arch packaging good practices, so I trust you on this. Its sounds convenient for both user and maintainer.
 
> When the PKGBUILD is updated to get a newer version, the patch can be removed at the same time, so its almost self regulating
> 
> Just thinking about making it easy for the user who just want to follow some instructions instead of finding comments here and there to compliment the instructions

Agree on the purpose. I will submit a PR with your suggestions

maharifu, 2023-01-03:
Hey, Thanks for all your effort in this! :smile: 

I believe the issue with Firefox is the lack of support for the NV12 format, as identified here - https://bugzilla.mozilla.org/show_bug.cgi?id=1761519

In the meanwhile, I got it working by forcing v4l2-relayd to convert to a supported format (YUY2). I'm sure this is not the best way to do it, but I overwrote the service ExecStart by creating the file `/etc/systemd/system/v4l2-relayd.service.d/override.conf` with content:
```
[Service]
ExecStart=
ExecStart=/bin/sh -c 'DEVICE=$(grep -l -m1 -E "^${CARD_LABEL}$" /sys/devices/virtual/video4linux/*/name | cut -d/ -f6); exec /usr/bin/v4l2-relayd -i "${VIDEOSRC}" $${SPLASHSRC:+-s "${SPLASHSRC}"} -o "appsrc name=appsrc caps=video/x-raw,format=${FORMAT},width=${WIDTH},height=${HEIGHT},framerate=${FRAMERATE} ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink name=v4l2sink device=/dev/$${DEVICE}"'
```
It just adds ` ! video/x-raw,format=YUY2` after `videoconvert` to the original ExecStart.

After `sudo systemctl status v4l2-relayd.service` it works in Firefox. :smile: It also keeps working in Chromium and Cheese.

sigboe, 2023-01-03:
@maharifu I think the theoretical "propper way" to configure this would be in `/etc/v4l2-relayd` so
```bash
sudo cp /etc/default/v4l2-relayd /etc/v4l2-relayd
sudo -e /etc/v4l2-relayd
```

I changed FORTMAT= to YUY2, the webcam looks like it works in firefox after I do that, but it just shows a single color black.

If it can work by editing that file, it can probably be shipped already working in this repo

maharifu, 2023-01-03:
@sigboe I believe that FORMAT specifies the input format from the webcam, so it should be NV12. That's why you get the black screen when changing it.

The problem is Firefox doesn't handle that format, so we need to convert it before the v4l2sink. 

Maybe we could have a patch to change the systemd unit file of v4l2-relayd when building the package?

stefanpartheym, 2023-01-08:
> Agree on the purpose. I will submit a PR with your suggestions

Hi @Tropicao , thanks a lot for your effort! I merged your PR as requested.

Best regards
Stefan

z4lem, 2023-01-17:
Unfortunately it's still not working with my setup (10. Gen, Kernel 6.1).
After installing the script and rebooting, my system shows a weird behaviour:
Each application's startup is delayed heavily (networkman, terminal, etc..) . The mouse cursor stays unresponsive all the time. 
This is what joarnalctl shows after rebooting:

```
Jan 18 00:05:55 zalem kernel: blacklist: Problem blacklisting hash (-13)
Jan 18 00:05:55 zalem kernel: blacklist: Problem blacklisting hash (-13)
Jan 18 00:05:59 zalem kernel: spi-nor spi0.0: unrecognized JEDEC id bytes: 3c 30 30 09 69 00
Jan 18 00:06:00 zalem kernel: Bluetooth: hci0: Malformed MSFT vendor event: 0x02
Jan 18 00:06:03 zalem kernel: ov2740 i2c-INT3474:01: error while getting pled gpio: -2
Jan 18 00:06:03 zalem kernel: BUG: kernel NULL pointer dereference, address: 0000000000000008
Jan 18 00:06:03 zalem kernel: #PF: supervisor read access in kernel mode
Jan 18 00:06:03 zalem kernel: #PF: error_code(0x0000) - not-present page
Jan 18 00:06:03 zalem bluetoothd[906]: src/plugin.c:plugin_init() Failed to init vcp plugin
Jan 18 00:06:03 zalem bluetoothd[906]: src/plugin.c:plugin_init() Failed to init mcp plugin
Jan 18 00:06:03 zalem bluetoothd[906]: src/plugin.c:plugin_init() Failed to init bap plugin
Jan 18 00:06:07 zalem lightdm[999]: gkr-pam: couldn't unlock the login keyring.
Jan 18 00:06:32 zalem pulseaudio[1022]: GetManagedObjects() failed: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Jan 18 00:07:07 zalem wpa_supplicant[957]: nl80211: kernel reports: key addition failed
Jan 18 00:07:18 zalem ntpd[1389]: bind(21) AF_INET6 fe80::7150:d981:8261:ec73%3#123 flags 0x11 failed: Cannot assign requested address
Jan 18 00:07:18 zalem ntpd[1389]: unable to create socket on wlan0 (5) for fe80::7150:d981:8261:ec73%3#123
Jan 18 00:09:03 zalem systemd-udevd[622]: 51F5230E-9677-46CD-A1CF-C0B23EE34DB7: Worker [626] processing SEQNUM=4767 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: input11: Worker [631] processing SEQNUM=5403 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: dmic-codec: Worker [654] processing SEQNUM=5275 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: {33AECD58-B679-4E54-9BD9-A04D34F0C226}: Worker [663] processing SEQNUM=5223 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: ljca-spi.4.auto: Worker [666] processing SEQNUM=5043 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: 0018:06CB:CEB3.0001: Worker [643] processing SEQNUM=5182 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: iTCO_wdt: Worker [658] processing SEQNUM=5174 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: 0000:00:16.0-fbf6fcf1-96cf-4e2e-a6a6-1bab8cbe36b1: Worker [641] processing SEQNUM=5069 killed
Jan 18 00:09:03 zalem systemd-udevd[622]: 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: Worker [649] processing SEQNUM=5068 killed
```
Uninstalling reverse the issue.
Any idea?

stefanpartheym, 2023-01-19:
Upgrading to kernel v6.1.6 breaks webcam for me unfortunately, even when using the new `--workaround` flag. :sweat_smile: 
I need to figure out whats going on in my case. Maybe a Wayland thing, I don't know. I'll post an update once I now more.

alba4k, 2023-01-20:
> 

That is interesting, everything still works perfectly for me on 6.1.6 (zen, but I doubt that's what makes it work).

I would be surprised if that was because of wayland, have you rebooted your system after upgrading?

z4lem, 2023-01-20:
> > 
> 
> That is interesting, everything still works perfectly for me on 6.1.6 (zen, but I doubt that's what makes it work).
> 
> I would be surprised if that was because of wayland, have you rebooted your system after upgrading?

@stefanpartheym 
Same for me: With or and without the workaround, I get a black image for all test cases as output and the cam light doesn't show up. I rebooted after each installation and I'm using X and kernel 6.1.6-arch1-3.

alba4k, 2023-01-20:
@z4lem does DKMS give you any errors when running the install script? also, even though this shouldn't matter, are you running the uninstall script first?

z4lem, 2023-01-24:
> @z4lem does DKMS give you any errors when running the install script? also, even though this shouldn't matter, are you running the uninstall script first?
> 
> Also, I just noticed that 6.1.7 just came out (it also works on my machine), so maybe check if that does anything?

@alba4k No errors during installation. And yes, in uninstalled the old version and rebooted, before trying the new one. 
My error described above appears on the newest kernel version 6.1.7 :\

alba4k, 2023-01-24:
That's interesting, I honestly have no Idea what to suggest. maybe try to ask on the arch BBS discussion?

Tropicao, 2023-01-25:
> @alba4k No errors during installation. And yes, in uninstalled the old version and rebooted, before trying the new one. My error described above appears on the newest kernel version 6.1.7 :\

Hi. I observed that DKMS build errors are quite silent during installation with the script, I only observed real errors when refreshing dkms manually (`dmks status; dkms autoinstall`), when I searched for webcam issues on my laptop. Did you go this far to check for build error ? Is the driver really loaded after executing the install script ?
Moreover, some fixes have finally been merged mainline, so we will need to revert [the hotfix to point on custom branch](https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/3) of mainline driver. I will try to test current state of river install on my machine by the end of the week, and submit the revert if it works

GorgetGuilhem, 2023-01-27:
Hi, I can't get it fully working on 6.1.8-arch1-1. No errors at the installation and `test.sh` works after installation. Problem is that no application detects the camera, tried with Chromium, Chrome, Firefox, Slack and Discord, with and without the `--workarround` flag.

Here are some informations :
```
# v4l2-ctl --list-devices
ipu6 (PCI:pci:pci0000:00):
        /dev/video1

ipu6 (pci:pci0000:00):
        /dev/media0

Virtual Camera (platform:v4l2loopback-000):
        /dev/video0

Webcam C170: Webcam C170 (usb-0000:04:00.0-2.3.2.1.3):
        /dev/video2
        /dev/video3
        /dev/media1

# sudo lshw
*-multimedia:0
             description: Multimedia controller
             produit: Intel Corporation
             fabriquant: Intel Corporation
             identifiant matériel: 5
             information bus: pci@0000:00:05.0
             version: 02
             bits: 64 bits
             horloge: 33MHz
             fonctionnalités: pciexpress pm bus_master cap_list
             configuration: driver=intel-ipu6 latency=0
             ressources: mémoireE/S:600-5ff irq:16 mémoire:603c000000-603cffffff
```

Anyway thanks a lot for your work !

Tropicao, 2023-01-28:
No major issue on my side on `6.1.8-arch1-1` kernel, after full uninstall then reinstall (using the --workaround flag). I did it with [custom patches removed and updated revision](https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/6) to make sure to get all latest fixes merged on mainline. I observe the opposite of @GorgetGuilhem : I still can not make `test.sh` work because of following error : 
```
[...]
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
[...]
[01-28 17:10:48.595] CamHAL[WAR] Failed to open file /run/camera/ov01a10-uf_VIDEO.aiqd, error No such file or directory
```
However camera works properly when testing it in Cheese (using [the proper device](https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/7)) and in Google Meet (tested in Firefox)

alba4k, 2023-01-28:
That is interesting

`test.sh` does not report any errors on my 9320, but cheese doesn't seem to work. What device are you selecting?

Tropicao, 2023-01-28:
> That is interesting
> 
> `test.sh` does not report any errors on my 9320, but cheese doesn't seem to work. What device are you selecting?

I select `Virtual Camera`, since it is the name reported by `v4l2-ctl --list-devices` on my machine

alba4k, 2023-01-29:
Nevermind, it *does* seem to work now in cheese too! I'm guessing some kernel between 6.1.6 and 6.1.8 did the trick, maybe in combination with the workaround for changing the video format (?).

Well, anyway, I think it's working basically everywhere for me now.

GorgetGuilhem, 2023-01-30:
@Tropicao tried with [custom patches removed and updated revision](https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/6) and I still cannot get applications seeing the webcam, even though `v4l2-ctl --list-devices` find it. `test.sh` works properly however.

Tropicao, 2023-01-30:
> @Tropicao tried with [custom patches removed and updated revision](https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/6) and I still cannot get applications seeing the webcam, even though `v4l2-ctl --list-devices` find it. `test.sh` works properly however.

Well, I have no clue why applications do not detect camera on your device :thinking: Since `test.sh` is working on your host, it means that your video stack is properly configured. You can not use webcam in chrome neither firefox ? Did you use the `--workaround` parameter when executing the install script ?

alba4k, 2023-01-30:
As a pretty sad sidenote, it eventually stopped working for me too.

Something weird I noticed, is that the camera still does turn off for a split second when I turn my laptop on (which it also did when the driver was working.)

Also, since this driver broke hibernation, at least for me, I think I will leave it uninstalled for now (since, again, I don't really use the webcam anyway).

On the hibernation issue: I do manage to get it apparently working by running the following script before entering hibernation, but when I power the system back on everything starts up clean.
```
#!/bin/bash

if [ "$1" = "load" ]; then
    modprobe snd_sof_pci_intel_tgl
    
    modprobe v4l2loopback
    systemctl start v4l2-relayd.service
elif [ "$2" = "unload" ]; then
    modprobe -r snd_sof_pci_intel_tgl
    
    systemctl stop v4l2-relayd.service
    modprobe -r v4l2loopback
fi
```

Don't mind that I'm also unloading and reloading the snd_sof_pci_intel_tgl module (sound), that is something that also prevents hibernation and is obviously sub-optimal, but I manage to hibernate using that (unless there's a kernel update pending), but this is completely unrelated. Is there maybe some other webcam-related module that I should unload?

GorgetGuilhem, 2023-01-31:
@Tropicao yes, I did the installation with the `--workarround` flag, and it doesn't work on Firefox and Chrome neither. And I've observed the same behavior as @alba4k about the camera led turning on half a second at system startup.

alba4k, 2023-01-31:
I think that the issue might be on the driver-side.

GorgetGuilhem, 2023-02-03:
No idea what's going on there, but I've noticed this running `sudo systemctl status v4l2-relayd` (last line) :
```bash
● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: disabled)
    Drop-In: /etc/systemd/system/v4l2-relayd.service.d
             └─override.conf
     Active: active (running) since Thu 2023-02-02 18:19:44 CET; 14h ago
   Main PID: 1182 (v4l2-relayd)
      Tasks: 2 (limit: 18692)
     Memory: 14.8M
        CPU: 59ms
     CGroup: /system.slice/v4l2-relayd.service
             └─1182 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink name=v4l2sink device=/dev/video0"

févr. 02 18:19:43 GuilhemArch systemd[1]: Starting v4l2-relay daemon service...
févr. 02 18:19:44 GuilhemArch systemd[1]: Started v4l2-relay daemon service.
févr. 02 18:19:44 GuilhemArch v4l2-relayd[1182]: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
```

Tropicao, 2023-02-03:
> EDIT : After some investigations, I found out that I needed to install gst-plugin-pipewire ... The webcam works for all applications now, thanks for your help !

Congratulations on finding the solution and making it work :) We should probably add a mention to this in the README or even automate it in the install script

alba4k, 2023-02-03:
I might do this tomorrow, if I find a couple of minutes :)

GorgetGuilhem, 2023-02-03:
> Congratulations on finding the solution and making it work :) We should probably add a mention to this in the README or even automate it in the install script

Yep, could definitively help someone else !

> I might do this tomorrow, if I find a couple of minutes :)

That would be great, thanks ! Just to mention, I'm using pipewire ;)

alba4k, 2023-02-03:
Same

I would be curious to see if the same happened if someone using pulseaudio would be experiencing the same (most likely, the answer is yes, since pipewire handles a lot of non-audio related media stuff), but I'm too lazy to install it just to check out ;P

Anyway, I think I will be opening the PR later this evening (it's 20.28 at the moment of typing for me)

alba4k, 2023-02-03:
Ok so, it's not working for me, which I expected, but if any of you can confirm that it works with `gst-plugin-pipewire` installed, then using [my branch](https://github.com/alba4k/archlinux-ipu6-webcam) _should_ be working too. I can open a pull request if someone is able to confirm that it is indeed working.

Also, the reason I suppose it is not working for me, is that the `ivsc-driver` and `ipu6-drivers` (no complaints for `v4l2loopback`) modules fail to install with DKMS printing the following (wall of) errors:

```
==> dkms install --no-depmod ivsc-driver/r35.94ecb88 -k 6.1.9-arch1-1
Module version  for ljca.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for spi-ljca.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for gpio-ljca.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for i2c-ljca.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for mei-vsc.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for intel_vsc.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for mei_csi.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for mei_ace.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for mei_pse.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for mei_ace_debug.ko.zst
exactly matches what is already found in kernel 6.1.9-arch1-1.
DKMS will not replace this module.
You may override by specifying --force.
Error! Installation aborted.
==> WARNING: `dkms install --no-depmod ivsc-driver/r35.94ecb88 -k 6.1.9-arch1-1' exited 6
```

TL;DR: DKMS doesn't seem to like those modules already (seemingly) being there.

stefanpartheym, 2023-02-04:
Hi @alba4k ,

> Also, the reason I suppose it is not working for me, is that the ivsc-driver and ipu6-drivers (no complaints for v4l2loopback) modules fail to install with DKMS printing the following (wall of) errors:

I'm experiencing the exact same issue. However, it seems like webcam does work regardless (at least in my browser, not with `test.sh` though).

> One more thing: paru warns about `intel-ivsc-driver-dkms-git` conflicting with the AUR package `intel-ipu6-dkms-git`. Altough I do recognize that this is not exactly the same as what is being installed by the script, I wonder if this recent update might cause siome issues too. So this also goes in favour of renaming the packages related to the ones in the AUR to avoid confusion.

Yes, same for me when using `yay`. I modified all `PKGBUILD`s in the repo to use a suffix in order for the AUR package managers to be able to distinguish between those packages (also specifying `provides` and `conflicts`). This will be part of the `v1.0.0` "release".
Currently, I use the suffix `-archfix`. Not sure, if it really fits, but probably not too important :sweat_smile: 
Do you have any suggestions/wishes regarding the suffix? Otherwise I'll commit and push my changes to the repo.

Best regards
Stefan

sigboe, 2023-02-04:
>Currently, I use the suffix -archfix. Not sure, if it really fits, but probably not too important :sweat_smile: 

Well my only opinion is that it's implicit this is a fix for arch, and many AUR packages have -fix suffix while no AUR package has -archfix suffix. So if you are interested in pushing this to AUR, then maybe the -fix suffix would suffice.

sigboe, 2023-02-04:
ops forgot to ping @stefanpartheym

stefanpartheym, 2023-02-04:
Hi @sigboe,

> Well my only opinion is that it's implicit this is a fix for arch, and many AUR packages have -fix suffix while no AUR package has -archfix suffix. So if you are interested in pushing this to AUR, [...]

However, I didn't plan to push those packages to AUR. With the archfix suffix I wanted to avoid any possible conflicts with packages from AUR. Hope you guys agree, otherwise I'm gonna change the suffix.

sigboe, 2023-02-04:
@stefanpartheym 
I don't think it matters if you have zero intentions to put the packages on the AUR. 

Just as an FYI here are the packaging guidelines, I guess you know about them already https://wiki.archlinux.org/title/Arch_package_guidelines

alba4k, 2023-02-04:
> Currently, I use the suffix -archfix. Not sure, if it really fits, but probably not too important 😅

I guess using `-fix` would be enough, but yeah that sounds good.

Tropicao, 2023-02-04:
> Yes, same for me when using yay. I modified all PKGBUILDs in the repo to use a suffix in order for the AUR package managers to be able to distinguish between those packages (also specifying provides and conflicts). This will be part of the v1.0.0 "release".
Currently, I use the suffix -archfix. Not sure, if it really fits, but probably not too important sweat_smile
@alba4k @Tropicao @sigboe : Do you have any suggestions/wishes regarding the suffix? Otherwise I'll commit and push my changes to the repo.

> You're right. However, I didn't plan to push those packages to AUR. With the archfix suffix I wanted to avoid any possible conflicts with packages from AUR. Hope you guys agree, otherwise I'm gonna change the suffix.

Hi @stefanpartheym. Before seeing your last answer, I was about to preach for a push to AUR : not only for a "fix" package, but to improve existing AUR packages by providing your modifications to those. IMHO, while this whole repository is currently the best way to make the webcam work on alder lake hardware, this package conflict issues (and your suggestion to add a suffix to fix this) highlights a more general issue, which is that we are kind of "forking" what already exists in AUR. While it seems valid as a short term solution (since [the webcam support mess is here to stay for a while](https://lwn.net/Articles/904776/)), any newcomer wanting to make the camera work will be confronted to multiple packages sources : the AUR packages, and your repository.
In the mean time, I totally understand if you do not want to undertake this. In this case, I do not have any strong suggestion about the best prefix :)

stefanpartheym, 2023-02-04:
Thanks @sigboe @alba4k @Tropicao for all the suggestions. I your answers give a clear direction as to what the suffix should be. Not that I would want to make a huge deal out of this, but you convinced me to use a simple `fix` suffix instead a weird unconventional `archfix`.
I'll change the suffix in the repo and release a `v1.0.1` for this.
However, I think right now, I'm not going to release those packages to AUR any time soon. Also, not sure if I would need to ask the authors of the patches and `PKGBUILD`s before doing so.

Again thanks to all of you for your input and contributions :)

stefanpartheym, 2023-02-04:
Hi @alba4k,

> Ok so, it's not working for me, which I expected, but if any of you can confirm that it works with `gst-plugin-pipewire` installed, then using [my [...]

sorry, I didn't check your fork before committing my changes. But yes, go ahead and create a PR, so I can merge your commit :)

alba4k, 2023-02-04:
I also agree on not pushing to the aur, to be honest, as it would only add complexity to the already existing mess. I could though add a note on the Arch Wiki mentioning this, as it already mentions the arch support thread from which this was created. 

I'm currently merging everything with your branch so I can open a pr.

stefanpartheym, 2023-07-30:
I think this issue could be closed now. Camera still not working in *all* programs probably. However, with the `--workaround` flag at least fixes it in some other programs, as far as I can tell.
Anyone seeing still a need for this issue, please open a new one – thanks :)

[1 more reply]

closed 2023-07-30 by stefanpartheym