stefanpartheym/archlinux-ipu6-webcam #2: Not working on 6.0.12

z4lem, 2022-12-20:
Frist of all, huge thx to @stefanpartheym for your efforts!
Maybe you (or anyone else) has a clue for the following issue:

My setup: 
* According to service tag my Gen 10 Carbon has the following build-in camera:
FHD RGB + IR Camera with ThinkShutter 
* I run Arch Linux with kernel 6.0.12-arch1-1

I managed to install the repo which ended with successfully starting the v4l2-relayd.service.
This is how my dev outpul liiks line now:

[image]

However, test.sh outputs the following:

```
Setting pipeline to PAUSED ...
Failed to set pipeline to PAUSED.
Setting pipeline to NULL...
Freeing pipeline ...
```

Cheese outputs he following error:

```
(cheese:16705): cheese-WARNING **: 17:50:28.289: 
Das Gerät »/dev/video1« unterstützt keine 2:0:0:0-Farbmetrik: ../gstreamer/subprojects/gst-plugins-good/sys/v4l2/gstv4l2object.c(4125): gst_v4l2_object_set_format_full (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin35/GstV4l2Src:v4l2src1:
Device wants 2:0:0:0 colorimetry
```
Obs outputs the following error:
```
Error: v4l2-input: Initialization failed, errno: Unsuitable IOCTL (I/O-Control) for this device.
```

Despite the Chrome workaround provided, the [test website](https://webrtc.github.io/samples/src/content/devices/input-output/) and BigBlueButton allow me only to select a **Virtual Device** which results in a black screen.

Current workaround: dualbooting Windows for webcam use :'( 
Maybe there's a chance for a better workaround (without having to carry an external one with me).

 Fyi: [Here's](https://lwn.net/Articles/904776/) an article & discussion about this topic...

sigboe, 2022-12-20:
Hey, I believe this is a duplicate of issue https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/1
I am also on 6.0.12, and i posted some steps here https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/1#issuecomment-1347066095 
I wrote there that it still doesnt work, accept the modified test, but after testing chromium it did indeed work with the additional steps there. Cheese and Firefox still doesn't work, but it beats having to use my phone to do meetings

alba4k, 2022-12-20:
Yes I' 95% sure this is the same thing.

z4lem, 2022-12-21:
Thx for reply.

As mentioned above - I created a new issue because I use another kernel version than the caption of #1 says. @sigboe I didn't know from your post, that you are using the same kernel. I don't know how relevant kernel version is for this issue...

And as mentioned above, the provided workarounds didn't work for me, using test command variant for X11 provided by @sigboe:
`sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! ximagesink`

Here's the detailed log for the new test script:

```
Setting pipeline to PAUSED ...
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov13b10.aiqb
[12-21 15:05:45.438] CamHAL[ERR] there is no aiqb file:ov13b10
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov13b10.aiqb
[12-21 15:05:45.438] CamHAL[ERR] there is no aiqb file:ov13b10
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov8856.aiqb
[12-21 15:05:45.438] CamHAL[ERR] there is no aiqb file:ov8856
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov8856.aiqb
[12-21 15:05:45.438] CamHAL[ERR] there is no aiqb file:ov8856
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov01a10.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov01a10.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov01a10.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name ov01a10.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[12-21 15:05:45.438] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name ov01a1s.aiqb
[12-21 15:05:45.439] CamHAL[INF] aiqb file name ov01a1s.aiqb
[12-21 15:05:45.439] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[12-21 15:05:45.445] CamHAL[ERR] Get entity fail for calling getEntityById
[12-21 15:05:45.445] CamHAL[ERR] Get entity fail for calling getEntityById
[12-21 15:05:45.445] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[12-21 15:05:45.445] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[12-21 15:05:45.445] CamHAL[ERR] set up mediaCtl failed
[12-21 15:05:45.445] CamHAL[ERR] @configure Device Configure failed
[12-21 15:05:45.445] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.003322614
Setting pipeline to NULL ...
Freeing pipeline ...
```

alba4k, 2022-12-21:
Yes I' 95% sure this is the same thing.

    I agree, let me update the issue title

twouters, 2022-12-22:
Can you try again with 6.1.1?

z4lem, 2022-12-22:
> Yes I' 95% sure this is the same thing.
> 
> ```
> I agree, let me update the issue title
> ```
@alba4k thank you - I will close this issue and continue yours!
@twouters I#M on it, will report in #1!

closed 2022-12-22 by z4lem

alba4k, 2022-12-22:
Let's all hope we get a way to make it work overall

sigboe, 2022-12-23:
@twouters I upgraded to `6.1.1-arch-1` and the camera stopped working for me. 
Both the test script and using it in chromium stopped working for me

z4lem, 2022-12-23:
> 
Same error here!