stefanpartheym/archlinux-ipu6-webcam #5: Camera gives black image

elhossary, 2023-01-14:
I have Thinkpad X1 carbon Gen10 (Alder lake) laptop with arch kernel >= 6.0
The camera doesn't work. after successful installation I find the "Virtual camera" entry (in Zoom for example).
But it shows black image.
Any idea how can I fix that?
journalctl output:
`
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.78] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.78] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.78] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.77] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.76] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[ERR] there is no aiqb file:ov8856
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[INF] aiqb file name ov8856.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[ERR] there is no aiqb file:ov8856
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[INF] aiqb file name ov8856.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[ERR] there is no aiqb file:ov13b10
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[INF] aiqb file name ov13b10.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[ERR] there is no aiqb file:ov13b10
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:25.75] CamHAL[INF] aiqb file name ov13b10.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.314] CamHAL[ERR] failed to config streams.
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.314] CamHAL[ERR] @configure Device Configure failed
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.314] CamHAL[ERR] set up mediaCtl failed
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.314] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.314] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.314] CamHAL[ERR] Get entity fail for calling getEntityById
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.314] CamHAL[ERR] Get entity fail for calling getEntityById
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.305] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /d>
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.305] CamHAL[INF] aiqb file name ov01a1s.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.304] CamHAL[INF] aiqb file name ov01a1s.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.304] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.304] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.304] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.304] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.304] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.303] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:25 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:24 MuHoss-PC plasmashell[8332]: QFont::setPointSizeF: Point size <= 0 (0.000000), must be greater than 0
Jan 09 21:01:24 MuHoss-PC kded5[8243]: Registering ":1.69/StatusNotifierItem" to system tray
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 21:01:23.302] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 21:01:23.301] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 21:01:23.301] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 21:01:23.301] CamHAL[INF] aiqb file name ov01a10.aiqb
[... 52 lines cut ...]
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.84] CamHAL[ERR] there is no aiqb file:ov13b10
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.86] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.86] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.86] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.86] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.85] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.84] CamHAL[INF] aiqb file name ov01a10.aiqb
[... cut ...]
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.84] CamHAL[ERR] there is no aiqb file:ov13b10
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.84] CamHAL[INF] aiqb file name ov13b10.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.84] CamHAL[ERR] there is no aiqb file:ov13b10
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:23.84] CamHAL[INF] aiqb file name ov13b10.aiqb
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:18.762] CamHAL[ERR] failed to config streams.
Jan 09 21:01:23 MuHoss-PC sh[483]: [01-09 20:59:18.762] CamHAL[ERR] @configure Device Configure failed
`

alba4k, 2023-01-29:
what programs are you testing in, and how did you install? the driver?

elhossary, 2023-01-29:
I am testing with Zoom, Chrome, and OBS studio.
I used this repo for installation only, once with ./install and once with ./install --workaround flag
both don't work

alba4k, 2023-01-29:
does `systemctl status v4l2-relayd` report anything?

elhossary, 2023-01-29:
```
$ systemctl status v4l2-relayd
● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: disabled)
    Drop-In: /etc/systemd/system/v4l2-relayd.service.d
             └─override.conf
     Active: active (running) since Sun 2023-01-29 13:25:01 CET; 3min 11s ago
    Process: 470 ExecCondition=/usr/bin/test -n ${VIDEOSRC} (code=exited, status=0/SUCCESS)
    Process: 471 ExecCondition=/usr/bin/test -n $FORMAT (code=exited, status=0/SUCCESS)
    Process: 472 ExecCondition=/usr/bin/test -n $WIDTH (code=exited, status=0/SUCCESS)
    Process: 473 ExecCondition=/usr/bin/test -n $HEIGHT (code=exited, status=0/SUCCESS)
    Process: 475 ExecCondition=/usr/bin/test -n $FRAMERATE (code=exited, status=0/SUCCESS)
    Process: 479 ExecCondition=/usr/bin/test -n ${CARD_LABEL} (code=exited, status=0/SUCCESS)
   Main PID: 485 (v4l2-relayd)
      Tasks: 2 (limit: 38136)
     Memory: 191.9M
        CPU: 1.656s
     CGroup: /system.slice/v4l2-relayd.service
             └─485 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! >

Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[ERR] there is no aiqb file:ov8856
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[INF] aiqb file name ov8856.aiqb
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[ERR] there is no aiqb file:ov8856
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[INF] aiqb file name ov01a10.aiqb
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.837] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.838] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
Jan 29 13:27:49 MuHoss-PC sh[485]: [01-29 13:27:49.838] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
lines 1-28/28 (END)
```
There is another symtopm I noticed, after reboot the WiFi takes very long to connect. The symtopm goes away only when I uninstall it

closed 2023-06-12 by elhossary