stefanpartheym/archlinux-ipu6-webcam #9: ipu6 v4l2 devices missing

avanishsubbiah, 2023-02-02:
The previous output of the command `v4l2-ctl --list-devices` gave me:
```
ipu6 (PCI:pci:pci0000:00):
	/dev/video1

ipu6 (pci:pci0000:00):
	/dev/media0

Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
```

But after updating to kernel 6.1.9 it now just outputs:
```
Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
```

When opening any app (cheese, firefox, etc.) the light does not show up and the screen is black. However on the Linux zen kernel (6.1.9) this issue is not present with the ipu6 devices showing up as well like before.

I have tried fully uninstalling, deleting yay and pacman cache, and reinstalling but ipu6 v4l2 devices seem to only be present on the Linux zen kernel (6.1.9) and not the default Arch (6.1.9) or Clear linux (6.1.7) kernels.

alba4k, 2023-02-03:
Yes this is a known issue. I will open a pull request in a couple of hours that __should__ fix it :)

avanishsubbiah, 2023-02-08:
> Yes this is a known issue. I will open a pull request in a couple of hours that _should_ fix it :)

I am still encountering this issue after the pull request

alba4k, 2023-02-08:
hi @avanishsubbiah ,does dkms report any errors when building the modules in install.sh?

avanishsubbiah, 2023-02-08:
> hi @avanishsubbiah, does dkms report any errors when building the modules in install.sh?

I get the same dkms error as in issue #11

avanishsubbiah, 2023-02-09:
So I updated from 6.1.9 to 6.1.10 and installed got the same error that dkms installation failed but now the devices show up and the webcam works regardless

I've noticed that the installation seems to work the first time I try it on new kernels, then if anything regenerates initramfs it breaks and reinstalling it doesn't fix it. Idk, how to test this exactly.

awly, 2023-08-04:
One of the recent `linux-lts` kernel updates on Arch broke this again. Current version is `6.1.39-3`.
It used to work on some slightly older `linux-lts` and `linux-zen` kernels, within `6.1.*` but I can't remember the exact version.
Also, both regular (`6.4.7.arch1-3`) and zen (`6.4.7.zen1-3`) kernel versions hit the same problem here.

Running Dell XPS 13 Plus 9320, the error is:
```
$ sudo dmesg | ag ipu6
[    1.363680] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    1.364000] intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x2)
[    1.364020] intel-ipu6 0000:00:05.0: physical base address 0x603e000000
[    1.364022] intel-ipu6 0000:00:05.0: mapped as: 0x0000000070607fa9
[    1.364073] intel-ipu6 0000:00:05.0: Unable to set secure mode
[    1.364075] intel-ipu6 0000:00:05.0: IPU in non-secure mode
[    1.364076] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    1.364078] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    1.364083] intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[    1.364084] intel-ipu6 0000:00:05.0: IPC reset done
[    1.364085] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    1.364112] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6ep_fw.bin failed with error -2
[    1.364114] intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    1.364151] intel-ipu6: probe of 0000:00:05.0 failed with error -2
```

Wondering if anyone else on those kernel versions could reproduce the issue to confirm it's not my setup?

fbrinker, 2023-08-21:
I am having the same issues on 6.4.11-arch2-1 again

stefanpartheym, 2023-11-13:
Hi @avanisubbiah and @alba4k,
are you still having issues? Current state of the repository should work for kernel versions `6.5.x` (probably with the workaround described in [this section of the README](https://github.com/stefanpartheym/archlinux-ipu6-webcam#solving-the-error-warning-erroneous-pipeline-no-element-icamerasrc)).

alba4k, 2023-11-13:
@stefanpartheym It does not appear to work in 6.6.1 (as I kind of expected) but it may also be tied to 

[image]

I guess we will have to see

I do not have that package cached (as I had already removed it)

stefanpartheym, 2023-11-14:
@alba4k I can confirm, the driver won't work on kernel version `6.6.1` for me too. I'll have a look into this once I have some more time.

stefanpartheym, 2023-11-19:
Closing this one, as drivers are now working on kernel `v6.6.1` too when following the steps mentioned in https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/53#issuecomment-1817579746

closed 2023-11-19 by stefanpartheym