stefanpartheym/archlinux-ipu6-webcam #11: Not installing and breaking hibernation

alba4k, 2023-02-05:
I'm opening a new issue on this as it's a slightly separate problem, to keep #1 clean.

Again, DKMS fails to install `ivsc-driver` and `ipu6-drivers`, giving a series of
```
Module version  for [some module].ko.zst
exactly matches what is already found in kernel [uname -r].
DKMS will not replace this module.
You may override by specifying --force.
```
followed by 
```
Error! Installation aborted.
==> WARNING: `dkms install --no-depmod [module name]/[module version] -k [uname -r]' exited 6
```

I think that this might be fixed by installing using `--force`, but that is for obvious reasons far from the best solution.

`v4l2loopback` installs correctly.

Also, even though (and probably because) two of the modules do not install, this also seems to break hibernation, both with and without running `# modprobe -r v4l2loopback && systemctl stop v4l2-relayd` before hibernating. The system shuts down (apparently) fine, but when it boots back up I'm meeted by my system however I had left it when hibernating, but completely frozen and unresponsive. I am not able to SSH into it, nor switch to an other TTY. The best thing I could do is using `Alt+SysRq+B` to reboot (holding down the power button would probably also work). I'm then greeted by a fresh boot, which works fine.

alba4k, 2023-02-09:
the webcam works, but the modules aren't there (?)

avanisubbiah, 2023-02-10:
> the webcam works, but the modules aren't there (?)

Dkms shows the messages that it failed to install but the devices show up in v4l2 ctl and the webcam works regardless

alba4k, 2023-02-10:
what happens if you try to modprobe them on manaually

alba4k, 2023-02-11:
Interestingly, nothing worked for me after installing, but it did after I reinstalled the modules manually with `--force`

```
dkms install --no-depmod ipu6-drivers/r78.f83b0747b -k 6.1.10-arch1-1
dkms install --no-depmod ipu6-drivers/r78.f83b0747b -k 6.1.10-zen1-1-zen --force
dkms install --no-depmod ivsc-driver/r35.94ecb88 -k 6.1.10-arch1-1 --force
dkms install --no-depmod ivsc-driver/r35.94ecb88 -k 6.1.10-zen1-1-zen --force
```

wellsaid, 2023-04-14:
Hi all,

I recently got a dell XPS 13 9315 and used the `./install.sh --workardound` on an archlinux installation with plasma wayland  + sddm and kernel 6.2.10-arch1-1. I can successfully use the webcam in firefox, chromium and some desktop applications (jitsi, zoom, skype) (I can't use them in kamoso or cheese as mentioned in the README).

However I have the exact same problem described by @alba4k in this issue.

My dkms status after install is as follows:

````
$ dkms status
ipu6-drivers/r86.7fdfb5eb2, 6.2.10-arch1-1, x86_64: installed (WARNING! Diff between built and installed module!) (WARNING! Diff between built and installed module!)
ivsc-driver/r36.c8db12b, 6.2.10-arch1-1, x86_64: built
v4l2loopback/r10.f94def5, 6.2.10-arch1-1, x86_64: installed
````

I did some experiments on a workaround for the hibernation problem:

First of all I have tried to remove ALL the modules installed with dkms using `sudo dkms remove -m <module_name> -k <kernel_ver>` and tried an hibernation. I was able to restore from hibernation successfully!!

Thus, I have reinstalled all the modules using `sudo dkms install -m <module_name> -k <kernel_ver>` in the following order: ipu6-drivers ivsc-driver v4l2loopback (as done in the install script).

I got the diff of lsmod after this dkms install  + reboot and wrote a script which: rmmod ALL modules loaded and then stops the v4l2-relayd service:

````
#!/bin/sh

if systemctl is-active v4l2-relayd >/dev/null; then
	systemctl stop v4l2-relayd
fi

modprobe -r \
	intel_ipu6_isys \
	intel_ipu6_psys \
	v4l2loopback \
	ov01a10 \
	v4l2_async \
	videobuf2_memops \
	spi_ljca \
	mei_vsc \
	mei_csi \
	mei_ace \
	mc \
	gpio_ljca \
	i2c_ljca \
	ljca
````

After executing this script I am able to successfully hibernate and restore from hibernation.
Of course, camera was not working at this stage. Thus, I wrote a script which restores the state as before:

````
#!/bin/sh

modules="\
	intel_ipu6_isys \
	intel_ipu6_psys \
	v4l2loopback \
	ov01a10 \
	v4l2_async \
	videobuf2_memops \
	spi_ljca \
	mei_vsc \
	mei_csi \
	mei_ace \
	mc \
	gpio_ljca \
	i2c_ljca \
	ljca"

for mod in $modules; do
	modprobe $mod
done

if systemctl is-enabled v4l2-relayd >/dev/null; then
	systemctl start v4l2-relayd
fi
````

After executing this script after a few seconds I can see the camera led blinking for a second and it is usable again :smile: 

I don't know if ALL those modules needs to be removed before hibernating.
When I have some other time I could figure out how to execute these scripts at hibernation/resume.

wellsaid, 2023-04-15:
A quick heads up on my previous post:

By reading this: https://man.archlinux.org/man/systemd-suspend.service.8.en as reference I wrote the following script:

/usr/lib/systemd/system-sleep/ipu6.sh:
````
#!/bin/sh

stage=$1
operation=$2

if [ -z "$stage" -o -z "$operation" ]; then
        echo "No stage or operation specified!" >&2
        exit 1
fi

modules="\
        intel_ipu6_isys \
        intel_ipu6_psys \
        v4l2loopback \
        ov01a10 \
        v4l2_async \
        videobuf2_memops \
        spi_ljca \
        mei_vsc \
        mei_csi \
        mei_ace \
        mc \
        gpio_ljca \
        i2c_ljca \
        ljca"

case $stage in
        pre)
                case $operation in
                        hibernate)
                                if systemctl is-active v4l2-relayd >/dev/null; then
                                        systemctl stop v4l2-relayd
                                fi

                                modprobe -r $modules
                                ;;
                        *)
                                echo "Invalid operation $operation for stage $stage" >&2
                                ;;
                esac
                ;;
        post)
                case $operation in
                        hibernate)
                                for mod in $modules; do
                                        modprobe $mod
                                done

                                if systemctl is-enabled v4l2-relayd >/dev/null; then
                                        systemctl start v4l2-relayd
                                fi
                                ;;
                        *)
                                ;;
                esac
                ;;
        *)
                echo "Invalid stage $stage!" >&2
                ;;
esac
````

I can consider opening a pull request to install this script with a `--hibernation-workaround` option for this. However I think some of the modules do not need to be removed and I don't know if its necessary to stop v4l2-relayd service.

avanisubbiah, 2023-04-24:
> A quick heads up on my previous post:
> 
> By reading this: https://man.archlinux.org/man/systemd-suspend.service.8.en as reference I wrote the following script:
> 
> /usr/lib/systemd/system-sleep/ipu6.sh:
> 
> ```
> #!/bin/sh
> 
> stage=$1
> operation=$2
> 
> modules="\
>         intel_ipu6_isys \
>         intel_ipu6_psys \
>         v4l2loopback \
>         ov01a10 \
>         v4l2_async \
>         videobuf2_memops \
>         spi_ljca \
>         mei_vsc \
>         mei_csi \
>         mei_ace \
>         mc \
>         gpio_ljca \
>         i2c_ljca \
>         ljca"
> 
> case $stage in
>         pre)
>                 case $operation in
>                         hibernate)
>                                 if systemctl is-active v4l2-relayd >/dev/null; then
>                                         systemctl stop v4l2-relayd
>                                 fi
> 
>                                 modprobe -r $modules
>                                 ;;
>                 esac
>                 ;;
>         post)
>                 case $operation in
>                         hibernate)
>                                 for mod in $modules; do
>                                         modprobe $mod
>                                 done
> 
>                                 if systemctl is-enabled v4l2-relayd >/dev/null; then
>                                         systemctl start v4l2-relayd
>                                 fi
>                                 ;;
>                 esac
>                 ;;
> esac
> ```
> 
> I can consider opening a pull request to install this script with a `--hibernation-workaround` option for this. However I think some of the modules do not need to be removed and I don't know if its necessary to stop v4l2-relayd service.

Interestingly enough this script works with systemctl hibernate but it does not work with systemctl suspend-then-hibernate?

wellsaid, 2023-05-01:
> Interestingly enough this script works with systemctl hibernate but it does not work with systemctl suspend-then-hibernate?

Hi, I have not personally tested supend-then-hibernate. Reading the docs this script will not work since you should check for operation "suspend-then-hibernate" and then check the env variable.

striczkof, 2023-06-28:
Yo I figured out the culprit.
You only need to do 'modprobe -r i2c_ljca' to hibernate without freezing on resume, then load it back with 'modprobe i2c_ljca'

After figuring this out, I checked ivsc driver repository and found this https://github.com/intel/ivsc-driver/issues/32
There is currently a PR that might fix this.

For the meantime, I might either figure out a way to build ipu6-driver with that commit merged in. If that's impossible, I'd just settle for modprobing every hibernate/resume with @wellsaid script with just i2c_ljca unloaded.
Making a PR today or tomorrow for this one.

striczkof, 2023-06-28:
Done, I can't figure how to unload the module without engaging weaponised autism and writing inhibitors in the driver itself. A script ran by systemd should do the trick with the only caveat of not working on integrity lockdowns. Check out #28

striczkof, 2023-06-30:
Whilst waiting for merge. you can add this script to your system-sleep script to fix hibernation resume, just paste this to your terminal in one line:
`echo -e '#!/bin/sh\n[ "$1" = "pre" ] && [ "$SYSTEMD_SLEEP_ACTION" = "hibernate" ] && modprobe -r i2c_ljca' | sudo tee /usr/lib/systemd/system-sleep/i2c_ljca-s2disk_test.sh && sudo chmod 744 /usr/lib/systemd/system-sleep/i2c_ljca-s2disk_test.sh`

PR #28 (2023-07-12): Added hibernation workaround, removed ipu6-driver-fix, used getopts for options
Added a temporary hack that should fix #11 caused by i2c_ljca module. This should be fixed upstream later on since there is already a PR waiting for merge.

I also removed intel-ipu6-dkms-git-fix since the latest one from the AUR is working perfectly. (with already merged ivsc-driver)

Used getopts to handle multiple options in the installer script. Now, you have to use '-a' flag for Firefox/Discord workarounds and '-s' for the hibernation hack, or '-as' for both.

Merged PR #27 so I can work with a cleaner install [... cut]
closed 2023-07-12 by stefanpartheym