stefanpartheym/archlinux-ipu6-webcam #17: Install succeds but test fails; gstreamer plugin is blacklisted

ah-OOG-ah, 2023-03-21:
Install is successful, however `test.sh` only returns `WARNING: erroneous pipeline: no element "icamerasrc"`

After clearing the plugin cache and running `gst-inspect-1.0 -b`, I get
```
(gst-plugin-scanner:54695): GStreamer-WARNING **: 00:39:48.233: Failed to load plugin '/usr/lib/gstreamer-1.0/libgsticamerasrc.so': /usr/lib/gstreamer-1.0/libgsticamerasrc.so: undefined symbol: vied_nci_mem_type
Blacklisted files:
  libgsticamerasrc.so

Total count: 1 blacklisted file
```
This appears to be the same problem as [this](https://aur.archlinux.org/packages/icamerasrc-git#comment-905923) comment; I also got the same output from `nm`.

striczkof, 2023-07-01:
Sorry just saw this. Did you manage to install icamerasrc-git successfully? Check out the latest scripts too and send logs if possible.

stefanpartheym, 2023-11-19:
Closing this one due to inactivity.

closed 2023-11-19 by stefanpartheym