stefanpartheym/archlinux-ipu6-webcam #18: Compilation of icamerasrc-git fails

alba4k, 2023-04-18:
```
gstcamerasrc.cpp: In function 'void gst_camerasrc_class_init(GstcamerasrcClass*)':
gstcamerasrc.cpp:980:67: error: expected primary-expression before ',' token
  980 |         gst_camerasrc_device_id_get_type(), DEFAULT_PROP_DEVICE_ID, (GParamFlags)(G_PARAM_READWRITE | G_PARAM_STATIC_STRINGS)));
      |                                                                   ^
gstcamerasrc.cpp: In function 'void gst_camerasrc_init(Gstcamerasrc*)':
gstcamerasrc.cpp:1258:30: error: expected primary-expression before '<' token
 1258 |   if (DEFAULT_PROP_DEVICE_ID < camerasrc->number_of_cameras) {
      |                              ^
gstcamerasrc.cpp:1259:50: error: expected primary-expression before ';' token
 1259 |     camerasrc->device_id = DEFAULT_PROP_DEVICE_ID;
      |                                                  ^
mv -f .deps/libgsticamerasrc_la-gstcameraformat.Tpo .deps/libgsticamerasrc_la-gstcameraformat.Plo
gstcamerasrc.cpp: In function 'gboolean gst_camerasrc_start(GstCamBaseSrc*)':
gstcamerasrc.cpp:2839:27: error: too many arguments to function 'int icamera::camera_device_open(int)'
 2839 |   ret = camera_device_open(camerasrc->device_id, camerasrc->num_vc);
      |         ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from gstcamerasrc.cpp:62:
/usr/include/libcamhal/api/ICamera.h:210:5: note: declared here
  210 | int camera_device_open(int camera_id);
      |     ^~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:624: libgsticamerasrc_la-gstcamerasrc.lo] Error 1
make[3]: *** Attesa per i processi non terminati....
mv -f .deps/libgsticamerasrc_la-gstcamerasrcbufferpool.Tpo .deps/libgsticamerasrc_la-gstcamerasrcbufferpool.Plo
mv -f .deps/libgsticamerasrc_la-gstcambasesrc.Tpo .deps/libgsticamerasrc_la-gstcambasesrc.Plo
make[3]: uscita dalla directory «/home/alba4k/.cache/paru/clone/icamerasrc-git/src/icamerasrc/src»
make[2]: *** [Makefile:685: all-recursive] Error 1
make[2]: uscita dalla directory «/home/alba4k/.cache/paru/clone/icamerasrc-git/src/icamerasrc/src»
make[1]: *** [Makefile:520: all-recursive] Error 1
make[1]: uscita dalla directory «/home/alba4k/.cache/paru/clone/icamerasrc-git/src/icamerasrc»
make: *** [Makefile:429: all] Error 2
==> ERRORE: Si è verificato un errore in build().
    L'operazione sta per essere interrotta...
errore: compilazione fallita per 'icamerasrc-git-r57.2d36ade-1': 
errore: compilazione fallita per i seguenti pacchetti: icamerasrc-git-r57.2d36ade-1
ERROR: Failed to install: intel-ivsc-driver-dkms-git intel-ivsc-firmware icamerasrc-git gst-plugin-pipewire
```

alba4k, 2023-04-18:
I know this issue isn't strictly related to this repository, but it affects it.

`test.sh` seems to fail after `WARNING: erroneous pipeline: no element "icamerasrc"`

GorgetGuilhem, 2023-05-09:
Same issue here, compilation of icamerasrc-git fails !

clintlombard, 2023-05-15:
Issue is with the package on AUR https://aur.archlinux.org/packages/icamerasrc-git. Workaround is to manually edit gstcamerasrc.cpp with  
```diff
diff --git a/src/gstcamerasrc.cpp b/src/gstcamerasrc.cpp
--- a/src/gstcamerasrc.cpp
+++ b/src/gstcamerasrc.cpp
@@ -2839 +2839 @@
-   ret = camera_device_open(camerasrc->device_id, camerasrc->num_vc);
+  ret = camera_device_open(camerasrc->device_id);
```

alba4k, 2023-05-15:
Still getting errors tho..

```
gstcamerasrc.cpp: In function 'void gst_camerasrc_class_init(GstcamerasrcClass*)':
gstcamerasrc.cpp:980:67: error: expected primary-expression before ',' token
  980 |         gst_camerasrc_device_id_get_type(), DEFAULT_PROP_DEVICE_ID, (GParamFlags)(G_PARAM_READWRITE | G_PARAM_STATIC_STRINGS)));
      |                                                                   ^
gstcamerasrc.cpp: In function 'void gst_camerasrc_init(Gstcamerasrc*)':
gstcamerasrc.cpp:1258:30: error: expected primary-expression before '<' token
 1258 |   if (DEFAULT_PROP_DEVICE_ID < camerasrc->number_of_cameras) {
      |                              ^
gstcamerasrc.cpp:1259:50: error: expected primary-expression before ';' token
 1259 |     camerasrc->device_id = DEFAULT_PROP_DEVICE_ID;
      |                                                  ^
make[3]: *** [Makefile:624: libgsticamerasrc_la-gstcamerasrc.lo] Error 1
make[2]: *** [Makefile:685: all-recursive] Error 1
make[1]: *** [Makefile:520: all-recursive] Error 1
make: *** [Makefile:429: all] Error 2
```

clintlombard, 2023-05-17:
Strange. No idea why it worked for me. Errors look like missing `;` somewhere. Alternatively change the PKGBUILD to point to this commit 2d36adecf1af9b44df591482aec27a32d8627999 instead of the icamerasrc_slim_api branch.

sigboe, 2023-05-19:
Hello, I also noticed this issue, since the ipu6 driver was updated to compile on Linux 6.3 and I made a merge request to update the driver I came across this issue. An older commit works to compile. I wrote my workaround in the merge request https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/19

sigboe, 2023-05-22:
Hello as I have said before there are 3 main issues getting this to compile and run today. 

1. Linux 6.3 support (PR to fix this is here https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/19 )
2. Error handling in the install script. If you already have icamerasrc-git or any other dependency, don't fail if a newer commit can't compile. (PR to fix this is here https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/20 )
3. Compiling [icamerasrc-git](https://aur.archlinux.org/packages/icamerasrc-git) for new users, they should not have to do manual work to get icamerasrc-git working.

@stefanpartheym I am open to discussion how we proceed with icamerasrc-git intel has introduced a bug [at this line in this file](https://github.com/intel/icamerasrc/blob/82cb8fc35f1ceb22ad4ff52aed7c021c417528d5/src/gstcamerasrc.cpp#L2839) that breaks compiling. The compiler says the function is passing too many arguments, removing the second (newly added argument) fixes compiling. I have not tested if any functionality is lost by doing this. This bug was introduced in commit [fe733e7](https://github.com/intel/icamerasrc/commit/fe733e7b98fc2c9824abe8b451182af38470d659) at April 3. And has issue https://github.com/intel/icamerasrc/issues/32 opened at April 6.  Our issue https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/18 was opened at April 18. Do you want us to roll our own PKGBUILD file or something? [17841ab](https://github.com/intel/icamerasrc/commit/17841ab6249aaa69bd9b3959262bf182dee74111) compiles successfully and can be compiled by replacing the source array in the PKGBUILD to fetch `"git+${url}.git#commit=17841ab6249aaa69bd9b3959262bf182dee74111"` instead.

If someone is reading this and just want it to work right now. the steps are

```bash
git clone https://aur.archlinux.org/icamerasrc-git.git
cd icamerasrc-git
sed -i 's/branch=icamerasrc_slim_api/commit=17841ab6249aaa69bd9b3959262bf182dee74111/' PKGBUILD
makepkg -sic
```
And then either Stefan has accepted my two pull requests, in case just install like normal on the newest git pull. And I am sorry I spread that out over two pull requests makes it harder for people to follow at home

```bash
cd ~ #or wherever you want to clone a repo
git clone git@github.com:stefanpartheym/archlinux-ipu6-webcam.git  #need to do this on a clean repo
cd archlinux-ipu6-webcam
git fetch origin pull/19/head:refs/heads/pr19
git merge pr19
git fetch origin pull/20/head:refs/heads/pr20
git merge pr20
#save and exit the file because a text editor will open, in vim this is done by hittin `:wq<enter>` if vim is what opened up
./install.sh
```
When these are merge, just delete the repo you cloned and clone it again to "clean up" your local changes (you merged the two PRs locally including making a commit to merge the second PR)

sigboe, 2023-05-23:
@GorgetGuilhem Okay, try to remove all the related packages, 

Then run the install.sh script, it will fail, but it will set up the correct dependencies first

then install the icamerasrc-git with the modified PKGBUILD, 

then run the install.sh again

GorgetGuilhem, 2023-05-23:
@sigboe thanks, it worked ! However the --workarround does not seem to work anymore, camera doesn't start on Firefox. But that's a start ! I'll try investigate this point when I have time.

GorgetGuilhem, 2023-05-23:
@sigboe sorry for the typo in my previous message, I checked my history and I ran the proper command without this typo so it's not coming from here.

sigboe, 2023-05-25:
@GorgetGuilhem I updated the previous commit when I noticed more people saying it works for them

GorgetGuilhem, 2023-05-31:
@sigboe perfect, thanks for your work :)

closed 2023-06-12 by stefanpartheym