stefanpartheym/archlinux-ipu6-webcam #21: ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.

hinricht, 2023-06-01:
I'm trying to get my webcam working on 

* Thinkpa X1 carbon 10th gen
* Manjao (Sway edition)
* Kernel v6.1.31-1-MANJARO
* archlinux-ipu6-webcam, main branch from today (4ae5beba4883dfa9fd3584e5e6f2d36df80d498b)

First, the installation failed during compiling `icamerasrc-git` but I got it succeeding following [these instructions patching icamerasrc-git](https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/18#issuecomment-1557720342), and I could succeed with the install script. Here's the installation log:

<details>
  <summary>install.sh log</summary>
  
```console
Sync Explicit (1): linux61-headers-6.1.31-1
warning: linux61-headers-6.1.31-1 is up to date -- skipping
 there is nothing to do
# Build and install package: intel-ipu6-dkms-git
~/work/webcam/archlinux-ipu6-webcam/intel-ipu6-dkms-git ~/work/webcam/archlinux-ipu6-webcam
==> Making package: intel-ipu6-dkms-git-fix r100.8c02a846d-1 (Do 01 Jun 2023 21:42:05 CEST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating ipu6-drivers git repo...
==> Validating source files with sha256sums...
    ipu6-drivers ... Skipped
==> Extracting sources...
  -> Creating working copy of ipu6-drivers git repo...
Reset branch 'makepkg'
==> Starting prepare()...
Cloning into 'ivsc-driver'...
==> Starting pkgver()...
==> WARNING: A package has already been built, installing existing package...
==> Installing package intel-ipu6-dkms-git-fix with pacman -U...
warning: intel-ipu6-dkms-git-fix-r100.8c02a846d-1 is up to date -- reinstalling
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) intel-ipu6-dkms-git-fix-r100.8c02a846d-1

Total Installed Size:  1,45 MiB
Net Upgrade Size:      0,00 MiB

:: Proceed with installation? [Y/n] 
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
checking available disk space...
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
==> dkms remove --no-depmod ipu6-drivers/r100.8c02a846d -k 6.1.31-1-MANJARO
==> dkms remove --no-depmod ipu6-drivers/r100.8c02a846d -k 6.3.5-1-MANJARO
:: Processing package changes...
reinstalling intel-ipu6-dkms-git-fix...
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Install DKMS modules
==> dkms install --no-depmod ipu6-drivers/r100.8c02a846d -k 6.1.31-1-MANJARO
Module version  for ov2740.ko.zst
exactly matches what is already found in kernel 6.1.31-1-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.
Module version  for hi556.ko.zst
exactly matches what is already found in kernel 6.1.31-1-MANJARO.
DKMS will not replace this module.
You may override by specifying --force.
==> dkms install --no-depmod ipu6-drivers/r100.8c02a846d -k 6.3.5-1-MANJARO
Module version  for ov2740.ko.zst
exactly matches what is already found in kernel 6.3.5-1-MANJARO.
[... 504 lines cut ...]
[06-01 21:43:23.138] CamHAL[INF] aiqb file name ov01a1s.aiqb
[06-01 21:43:23.138] CamHAL[INF] aiqb file name ov01a1s.aiqb
[06-01 21:43:23.138] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
[06-01 21:43:23.143] CamHAL[ERR] Get entity fail for calling getEntityById
[06-01 21:43:23.143] CamHAL[ERR] Get entity fail for calling getEntityById
[06-01 21:43:23.143] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[06-01 21:43:23.143] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[06-01 21:43:23.143] CamHAL[ERR] set up mediaCtl failed
[06-01 21:43:23.143] CamHAL[ERR] @configure Device Configure failed
[06-01 21:43:23.143] CamHAL[ERR] failed to config streams.
Execution ended after 0:00:00.003083925
Setting pipeline to NULL ...
Freeing pipeline ...
```
</details>

I also tried with kernel 6.3.5-1 with the same error. What am I missing ?

striczkof, 2023-06-26:
Don't need to add the udev rule manually if you used the latest.

Man why is it always Manjaro xd
I gotta install Manjaro and find it out myself

striczkof, 2023-07-01:
@hinricht dunno what to do with you case. It's working perfectly with my fresh install of Manjaro 22.1. Check out https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/26#issuecomment-1615873036 for the complete environment and logs.

I would say at the first glance that your laptop might be using a camera that requires another branch of ipu6-drivers (https://github.com/intel/ipu6-drivers/tree/ccg_plat_adlp) but then https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/22#issuecomment-1587014417 literally has the same laptop and managed to get the camera working. So yeah, no bloody idea.

Hey @elhossary If you don't mind, could you give us your laptop model and environment?

Let me try in Linux 6.3 in a bit.

elhossary, 2023-07-01:
@striczkof 
I have Alder lake X1 carbon 10 gen running Arch with kernel 6.3.9

striczkof, 2023-07-01:
> @striczkof I have Alder lake X1 carbon 10 gen running Arch with kernel 6.3.9

Thanks!

So it's neither a hardware issue (literally same laptop) nor Manjaro issue (running on my Manjaro installation right now) so idk lol
Lemme sleep then I figure this out tomorrow

hinricht, 2023-07-03:
See my comment at https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/26#issuecomment-1618251977 that I'm actually using manjaro-sway with Wayland, if this makes a difference.

hinricht, 2023-07-06:
@striczkof I saw your comment at
https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/26#issuecomment-1623003943
and gave your PR branch a try. Unfortunatly, I didn't succeed.
Here's the log from install.sh after using `uninstall.sh`:

<details>
  <summary>install.sh log</summary>

Sync Explicit (1): linux63-headers-6.3.5-2
 there is nothing to do
# Install build dependencies
Sync Explicit (1): base-devel-1-2
 there is nothing to do
# Install dependency for intel-ipu6-dkms-git
AUR Explicit (1): intel-ivsc-firmware-r2.764bd6e-1
:: [1mPKGBUILD up to date, skipping download: intel-ivsc-firmware[0m
  1 intel-ivsc-firmware                      (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
  1 intel-ivsc-firmware                      (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
[1m[32m==>(B[m[1m Making package: intel-ivsc-firmware r2.764bd6e-1 (Do 06 Jul 2023 12:41:59 CEST)(B[m
[1m[32m==>(B[m[1m Retrieving sources...(B[m
[1m[34m  ->(B[m[1m Updating ivsc-firmware git repo...(B[m
[1m[32m==>(B[m[1m Validating source files with sha256sums...(B[m
:: [1m(1/1) Parsing SRCINFO: intel-ivsc-firmware[0m
[1m[32m==>(B[m[1m Making package: intel-ivsc-firmware r2.764bd6e-1 (Do 06 Jul 2023 12:42:00 CEST)(B[m
[1m[32m==>(B[m[1m Checking runtime dependencies...(B[m
[1m[32m==>(B[m[1m Checking buildtime dependencies...(B[m
[1m[32m==>(B[m[1m Retrieving sources...(B[m
[1m[34m  ->(B[m[1m Updating ivsc-firmware git repo...(B[m
[1m[32m==>(B[m[1m Validating source files with sha256sums...(B[m
[1m[32m==>(B[m[1m Removing existing $srcdir/ directory...(B[m
[1m[32m==>(B[m[1m Extracting sources...(B[m
[1m[34m  ->(B[m[1m Creating working copy of ivsc-firmware git repo...(B[m
[1m[32m==>(B[m[1m Starting pkgver()...(B[m
[1m[32m==>(B[m[1m Updated version: intel-ivsc-firmware r8.29c5eff-1(B[m
[1m[32m==>(B[m[1m Sources are ready.(B[m
 -> intel-ivsc-firmware-r8.29c5eff-1 already made -- skipping build
[1m[32m==>(B[m[1m Making package: intel-ivsc-firmware r8.29c5eff-1 (Do 06 Jul 2023 12:42:01 CEST)(B[m
[1m[32m==>(B[m[1m Checking runtime dependencies...(B[m
[1m[32m==>(B[m[1m Checking buildtime dependencies...(B[m
[1m[32m==>(B[m[1m Starting pkgver()...(B[m
[1m[32m==>(B[m[1m Sources are ready.(B[m
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) intel-ivsc-firmware-r8.29c5eff-1

Total Installed Size:  9,45 MiB

:: Proceed with installation? [Y/n] 
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
checking available disk space...
:: Processing package changes...
installing intel-ivsc-firmware...
:: Running post-transaction hooks...
(1/1) Arming ConditionNeedsUpdate...
=> SUCCESS
# Install IPU6 driver
AUR Explicit (1): intel-ipu6-dkms-git-r78.f83b0747b-1
Sync Dependency (1): dkms-3.0.11-1
:: [1mPKGBUILD up to date, skipping download: intel-ipu6-dkms-git[0m
[... 1003 lines cut ...]
[07-06 12:48:18.980] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[07-06 12:48:18.981] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[07-06 12:48:18.981] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[07-06 12:48:18.982] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[07-06 12:48:18.982] CamHAL[INF] aiqb file name ov01a1s.aiqb
[07-06 12:48:18.982] CamHAL[INF] aiqb file name ov01a1s.aiqb
[07-06 12:48:18.983] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[07-06 12:48:18.989] CamHAL[ERR] Get entity fail for calling getEntityById
[07-06 12:48:18.989] CamHAL[ERR] Get entity fail for calling getEntityById
[07-06 12:48:18.989] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[07-06 12:48:18.989] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[07-06 12:48:18.989] CamHAL[ERR] set up mediaCtl failed
[07-06 12:48:18.989] CamHAL[ERR] @configure Device Configure failed
[07-06 12:48:18.989] CamHAL[ERR] failed to config streams.
Execution ended after 0:00:00.003559077
Setting pipeline to NULL ...
Freeing pipeline ...

</details>

Which is basically the same as before. I don't know how to proceed, happy to provide more details if needed.

striczkof, 2023-07-06:
I really don't get why it's trying to get the wrong device instead of OV01A10. Can't really think of anything other than doing a fresh install or so.

If you haven't tried already, uninstall every single package related to the camera then try installing again. Trying to think of something else that might be affecting it.

WAIT! @hinricht can you show me what comes out of `sudo modprobe -c | grep v4l2loopback` ?
After that, try also reloading v4l2loopback `sudo modprobe -r v4l2loopback && sudo modprobe v4l2loopback` and get the config again with modprobe -c

hinricht, 2023-07-06:
```
$ sudo modprobe -c | grep v4l2loopback
options v4l2loopback exclusive_caps=1 card_label="Virtual Camera"
options v4l2loopback exclusive_caps=1

~ $ sudo modprobe -r v4l2loopback && sudo modprobe v4l2loopback
modprobe: FATAL: Module v4l2loopback is in use.

~ $ sudo systemctl stop v4l2-relayd.service
~ $ sudo modprobe -r v4l2loopback && sudo modprobe v4l2loopback
~ $ sudo modprobe -c | grep v4l2loopback                       
options v4l2loopback exclusive_caps=1 card_label="Virtual Camera"
options v4l2loopback exclusive_caps=1
```

I tried starting the `v4l2-relayd.service` afterwards and tested using the `test.sh` script  but it still doesn't work, with the same output as I pasted above.

striczkof, 2023-07-08:
@hinricht can you show me what comes out of these commands?
`pacman -Q | grep "ipu6\|ivsc\|hal"`

striczkof, 2023-07-10:
@hinricht literally the same versions as mine :/

Ye, yours is an edge case. You can prolly solve it by just reinstalling Manjaro but I am genuinely interested on why your installation is not working and do something about it if possible to help others too.

hinricht, 2023-07-10:
After a while I ran `test.sh` again (still using your unmodified `add_workarounds` branch), and now I get a different result at least (and the  `Internal data flow error` is gone): 

```
<details>
  <summary>Click me</summary>

archlinux-ipu6-webcam $ ./test.sh 
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.560] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.560] CamHAL[ERR] Parse AE gain range failed
[07-10 19:31:23.561] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.561] CamHAL[ERR] Parse AE eExposure time range failed
[07-10 19:31:23.561] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-10 19:31:23.561] CamHAL[ERR] Parse AE gain range failed
Setting pipeline to PAUSED ...
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov13b10.aiqb
[07-10 19:31:23.907] CamHAL[ERR] there is no aiqb file:ov13b10
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov13b10.aiqb
[07-10 19:31:23.907] CamHAL[ERR] there is no aiqb file:ov13b10
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov8856.aiqb
[07-10 19:31:23.907] CamHAL[ERR] there is no aiqb file:ov8856
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov8856.aiqb
[07-10 19:31:23.907] CamHAL[ERR] there is no aiqb file:ov8856
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-10 19:31:23.907] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[07-10 19:31:23.908] CamHAL[INF] aiqb file name ov01a1s.aiqb
[07-10 19:31:23.909] CamHAL[INF] aiqb file name ov01a1s.aiqb
[07-10 19:31:23.909] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
[07-10 19:31:23.911] CamHAL[ERR] MediaControl init failed
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
</details>
```

I also tried with kernel versions `6.3.12-1-MANJARO` and `6.4.2-3-MANJARO` but the results were the same.

striczkof, 2023-07-10:
It keeps getting the wrong device, even though it's supposed to get OV01A10.

It either has to be the HAL or the IPU6 drivers messing up. What comes out of this?: `lsmod | grep "ivsc\|ipu\|ov"`

hinricht, 2023-07-11:
```
$ lsmod | grep "ivsc\|ipu\|ov"
intel_ipu6_isys       167936  0
videobuf2_dma_contig    28672  1 intel_ipu6_isys
videobuf2_v4l2         40960  1 intel_ipu6_isys
videobuf2_common       86016  4 videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,videobuf2_memops
intel_ipu6_psys       126976  0
intel_ipu6            139264  2 intel_ipu6_isys,intel_ipu6_psys
ov2740                 28672  0
v4l2_fwnode            32768  2 intel_ipu6_isys,ov2740
v4l2_async             28672  3 v4l2_fwnode,intel_ipu6_isys,ov2740
videodev              372736  7 v4l2_async,v4l2_fwnode,videobuf2_v4l2,v4l2loopback,intel_ipu6_isys,ov2740
mc                     86016  6 v4l2_async,videodev,videobuf2_v4l2,intel_ipu6_isys,videobuf2_common,ov2740
```

Btw, I don't remember if I added this udev rule myself or if it is placed there by the installation process, hope I didn't mess it up: 

```
$ cat 70-ipu6-psys.rules 
# This rule makes the Intel IPU6 sensor accessible to members of group "video",\n
# allowing icamerasrc to be used as an unprivileged user.\n
SUBSYSTEM=="intel-ipu6-psys",GROUP="video",MODE="0660"

```

striczkof, 2023-07-11:
Doesn't matter if you added it yourself since it's literally the same as the one added by icamerasrc PKGBUILD. And even then, it's only to let you use the camera as non-root. Not at that stage yet lol

Your modules check out with mine. So your laptop has OV2740 which is supported and the one that the HAL should get. This is mine:
<details><summary>lsmod | grep "ivsc\|ipu\|ov"</summary>

```
intel_ipu6_isys       167936  0
videobuf2_dma_contig    28672  1 intel_ipu6_isys
videobuf2_v4l2         40960  1 intel_ipu6_isys
videobuf2_common       86016  4 videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,videobuf2_memops
intel_ipu6_psys       126976  0
ov01a10                28672  1
intel_ipu6            139264  2 intel_ipu6_isys,intel_ipu6_psys
intel_vsc              16384  3 mei_ace,mei_csi,ov01a10
v4l2_fwnode            32768  2 ov01a10,intel_ipu6_isys
v4l2_async             28672  3 v4l2_fwnode,ov01a10,intel_ipu6_isys
videodev              372736  7 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov01a10,v4l2loopback,intel_ipu6_isys
mc                     86016  6 v4l2_async,videodev,videobuf2_v4l2,ov01a10,intel_ipu6_isys,videobuf2_common
```

</details> 

So now we need to figure up what causes the HAL to keep grabbing the wrong device.

Another idea. Try running this with hopes that it will give us more errors to work on: `GST_DEBUG="*:2" sudo -E LANG=C gst-launch-1.0 icamerasrc ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! fakesink`

Then after running the tests. Check this one out `sudo dmesg | grep ipu6`

Last question, does your webcam have an LED indicator? Does it light up when you boot up your PC? Mine does.

hinricht, 2023-07-12:
```
$ GST_DEBUG="*:2" sudo -E LANG=C gst-launch-1.0 icamerasrc ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! fakesink
[sudo] password for varac: 
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.348] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.348] CamHAL[ERR] Parse AE gain range failed
[07-12 06:48:21.349] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.349] CamHAL[ERR] Parse AE eExposure time range failed
[07-12 06:48:21.349] CamHAL[ERR] Malformed ET range in exposure time range configuration
[07-12 06:48:21.349] CamHAL[ERR] Parse AE gain range failed
Setting pipeline to PAUSED ...
[07-12 06:48:21.698] CamHAL[INF] aiqb file name ov13b10.aiqb
[07-12 06:48:21.698] CamHAL[ERR] there is no aiqb file:ov13b10
[07-12 06:48:21.698] CamHAL[INF] aiqb file name ov13b10.aiqb
[07-12 06:48:21.698] CamHAL[ERR] there is no aiqb file:ov13b10
[07-12 06:48:21.698] CamHAL[INF] aiqb file name ov8856.aiqb
[07-12 06:48:21.698] CamHAL[ERR] there is no aiqb file:ov8856
[07-12 06:48:21.698] CamHAL[INF] aiqb file name ov8856.aiqb
[07-12 06:48:21.698] CamHAL[ERR] there is no aiqb file:ov8856
[07-12 06:48:21.698] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-12 06:48:21.699] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-12 06:48:21.699] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-12 06:48:21.699] CamHAL[INF] aiqb file name ov01a10.aiqb
[07-12 06:48:21.699] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-12 06:48:21.700] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-12 06:48:21.700] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[07-12 06:48:21.701] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[07-12 06:48:21.702] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[07-12 06:48:21.702] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[07-12 06:48:21.703] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[07-12 06:48:21.703] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[07-12 06:48:21.703] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[... 10 lines cut ...]
0:00:00.480736222  2742 0x56421f034980 WARN                GST_PADS gstpad.c:1146:gst_pad_set_active:<camerasrc0:src> Failed to activate pad
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...

~ $ sudo dmesg | grep ipu6
[    1.788687] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    1.788890] intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x4)
[... cut ...]
[    1.805506] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6ep_fw.bin failed with error -2
[    1.805509] intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[    1.805532] intel-ipu6: probe of 0000:00:05.0 failed with error -2
```

So it looks like the firmware fails in the first place.

> Last question, does your webcam have an LED indicator? Does it light up when you boot up your PC? Mine does.

I really don't know, at least no LED does light up during boot / after powering on.

dmytri, 2023-07-21:
This seems related:

https://github.com/intel/ipu6-drivers/issues/38

I tried it, but it didn't work for me,  same test.sh output as @hinricht , tried several kernals, including standard linux and linux-lts, and linux-zen  on arch / garuda

Host: 21CB007GPB ThinkPad X1 Carbon Gen 10

striczkof, 2023-07-21:
Hey everyone, sorry for disappearing. I keep forgetting to drop an answer because I literally have no idea what could be causing it. Can't do much help until next week because my XPS 9320's screen keeps blacking out when I open the lid too far and it won't get fixed until Monday. (Probably ribbon issue)

By the way. I just like to add that this exact issue, with @alba4k logs was literally the same that I experienced before I gave up trying to install it and did a fresh reinstall months after my last attempt with this script, where it worked first try.

My idea is that if you try to install the drivers and stuff in some other way before this one. There might be leftover files or symlink. They would then need to be manually removed if the uninstallation keeps missing them.

Another idea is that there might be some package that is interacting somehow but extremely unlikely.

Gotta come up with better ideas later once I can use my laptop again without crying after the screen turns black for the nth time and have to close and reopen the lid again lol

dmytri, 2023-07-21:
Hey, though it might be left overs too, tried with a fresh install of garuda-sway and garuda-i3, both failed with the same test.sh results, though different dmesg output due to wayland vs X.

Would try with vanilla arch, but it would take some time, so last resort.

If there is anything I can do to help, let me know. Brand new X1 Carbon, so no problem reinstalling or installing anything, since there is nothing I need on it yet.

dmytri, 2023-07-21:
So the error appears when you `modprobe intel_ipu6` -- you can see it if you have `dmesg -w` open when you modprobe, using the files from https://github.com/intel/ipu6-drivers/issues/38#issuecomment-1320982931 above makes the load work and no longer give the signed firmware error, however, if you try to build camera-hal after that,  it will fail to build due to a missing method, presumably because the branch in that comment is not Adler Lake-P.

So, the error seems to come from the files shipped by intel/ipu4-camera-bins either not being signed correctly, or maybe Adler-Lake-P is different from Adler Lake

dmytri, 2023-07-22:
<details>
<summary>4l2-ctl --list-devices</summary>
Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
</details>

alba4k, 2023-07-22:
I got rid of the "original module exists" by uninstalling, deleting /var/lib/dkms and reinstalling

striczkof, 2023-07-30:
Don't think the original module exists matter here. Plus on working systems all DKMSed modules fail anyway due to the lack of keys, trying to figure out a way to fix that so I can lockdown my kernel.

Sorry guys I am really running out of ideas here if it fails to work on fresh installs too. Just need more data and people with same issues. Seriously, I literally had the same issue and exact errors before and I have no bloody idea how it got fixed.

striczkof, 2023-08-13:
The Malformed Extraterrestrial has claimed another victim in #33 .

@alba4k @dmytri @hinricht would it be possible for you guys to describe your Arch/Manjaro installation processes? I will try taking a look at it once again with the new data from the new victim being available lol

striczkof, 2023-08-17:
ye same case on @MrTrukiny 's install. My guess is some package is breaking it because his install was initially fine then an upgrade broke it. We got to pinpoint that package.

I think I might install EOS tonight and see if I manage to break my installation with `eos-upgrade`

alba4k, 2023-08-17:
Just in case it might be useful, `pacman -Q > pkgs.txt`

[pkgs.txt](https://github.com/stefanpartheym/archlinux-ipu6-webcam/files/12365763/pkgs.txt)

however, a lot of this will be duplicate across installations anyway.

striczkof, 2023-08-17:
LET'S FUCKING GO!

@dmytri @alba4k @hinricht I just managed to replicate it! Many thanks to @MrTrukiny for that `eos-update --yay` command, I think we have the offending packages already!
Although it's already 11pm here and I got a class to attend tomorrow so I'm gonna read through this and see what's what whilst I'm in the campus. For now, here is the probably useful data:

I installed EOS fresh, pulled my latest PR branch and installed it, reboot, LED lights up, then confirmed the camera working on Firefox.
Then I ran `eos-update --yay`, camera still working, then I restarted and LED still lights up.
BOOM! Camera no worky no more. Just black screen when I tried out WebRTC on Firefox. test.sh shows the same, although the Malformed ET error is gone, prolly because Intel has just added OV13B10 driver or something. The output is still long (listing all available files) and of course no window popped out.

Laptop: Dell XPS 13 Plus 9320 with Intel i7-1260P
OS: EndeavourOS Cassini Nova 03-2023 R3
Kernel: 6.4.8.arch1-1 (worked), 6.4.9-arch1-1 (after update and reboot)
DE: Xfce 4.18

<details><summary>`eos-update --yay`</summary>
<p>

```
:: Synchronizing package databases...
 endeavouros downloading...
 core downloading...
 extra downloading...
 multilib downloading...
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...

Package (62)                     Old Version          New Version          Net Change  Download Size

extra/alsa-card-profiles         1:0.3.76-2           1:0.3.77-1             0.00 MiB       0.03 MiB
endeavouros/eos-apps-info        1.7-2                23-3                   0.00 MiB       0.30 MiB
endeavouros/eos-bash-shared      23-22.5              23-22.7                0.00 MiB       0.05 MiB
endeavouros/eos-update-notifier  23-8                 23-10                  0.00 MiB       0.02 MiB
extra/firefox                    116.0.1-1            116.0.2-1              0.01 MiB      62.14 MiB
extra/fuse-common                3.15.1-1             3.16.1-1               0.00 MiB       0.00 MiB
extra/fuse3                      3.15.1-1             3.16.1-1               0.00 MiB       0.12 MiB
extra/gst-plugin-pipewire        1:0.3.76-2           1:0.3.77-1             0.00 MiB       0.05 MiB
extra/gtkmm3                     3.24.7-1             3.24.8-1               0.03 MiB       1.43 MiB
extra/gvfs                       1.50.5-1             1.50.6-1               0.03 MiB       0.97 MiB
extra/gvfs-afc                   1.50.5-1             1.50.6-1               0.00 MiB       0.04 MiB
extra/gvfs-gphoto2               1.50.5-1             1.50.6-1               0.00 MiB       0.04 MiB
extra/gvfs-mtp                   1.50.5-1             1.50.6-1               0.00 MiB       0.04 MiB
extra/gvfs-nfs                   1.50.5-1             1.50.6-1               0.00 MiB       0.02 MiB
extra/gvfs-smb                   1.50.5-1             1.50.6-1               0.00 MiB       0.03 MiB
extra/imagemagick                7.1.1.15-1           7.1.1.15-2             0.09 MiB       8.17 MiB
extra/intel-ucode                20230613-1           20230808-1             0.04 MiB       5.90 MiB
core/libinih                     56-1                 57-1                   0.00 MiB       0.02 MiB
extra/libjpeg-turbo              2.1.5.1-1            3.0.0-1                0.47 MiB       0.52 MiB
core/libldap                     2.6.4-2              2.6.6-1                0.00 MiB       0.27 MiB
extra/libpipewire                1:0.3.76-2           1:0.3.77-1             0.00 MiB       0.37 MiB
extra/libproxy                   0.5.3-1              0.5.3-2                0.00 MiB       0.03 MiB
core/linux                       6.4.8.arch1-1        6.4.9.arch1-1         -0.07 MiB     125.44 MiB
core/linux-firmware              20230625.ee91452d-5  20230804.7be2766d-2    2.30 MiB     155.73 MiB
core/linux-firmware-whence       20230625.ee91452d-5  20230804.7be2766d-2    0.02 MiB       0.03 MiB
core/linux-headers               6.4.8.arch1-1        6.4.9.arch1-1          0.02 MiB      29.41 MiB
core/perl                        5.36.1-2             5.38.0-1               3.18 MiB      20.26 MiB
extra/perl-clone                 0.46-1               0.46-2                 0.00 MiB       0.01 MiB
extra/perl-encode-locale         1.05-10              1.05-11                0.00 MiB       0.01 MiB
extra/perl-error                 0.17029-4            0.17029-5              0.00 MiB       0.02 MiB
extra/perl-file-listing          6.16-1               6.16-2                 0.00 MiB       0.01 MiB
extra/perl-html-parser           3.81-1               3.81-2                 0.15 MiB       0.15 MiB
extra/perl-html-tagset           3.20-13              3.20-14                0.00 MiB       0.01 MiB
extra/perl-http-cookies          6.10-3               6.10-4                 0.00 MiB       0.02 MiB
extra/perl-http-daemon           6.16-1               6.16-2                 0.00 MiB       0.02 MiB
extra/perl-http-date             6.05-6               6.06-1                 0.00 MiB       0.01 MiB
extra/perl-http-message          6.44-1               6.44-2                 0.00 MiB       0.07 MiB
extra/perl-http-negotiate        6.01-11              6.01-12                0.00 MiB       0.01 MiB
extra/perl-io-html               1.004-3              1.004-4                0.00 MiB       0.01 MiB
[... cut ...]
[    5.404190] iwlwifi 0000:00:14.3: WFPM_LMAC2_PD_NOTIFICATION: 0x1f
[    5.404198] iwlwifi 0000:00:14.3: WFPM_AUTH_KEY_0: 0x90
[    5.404207] iwlwifi 0000:00:14.3: CNVI_SCU_SEQ_DATA_DW9: 0x0
[    5.420502] iwlwifi 0000:00:14.3: RFIm is deactivated, reason = 4
[    5.504886] iwlwifi 0000:00:14.3: Registered PHC clock: iwlwifi-PTP, with index: 0
[    5.577458] iwlwifi 0000:00:14.3: WRT: Invalid buffer destination
[    5.729594] iwlwifi 0000:00:14.3: WFPM_UMAC_PD_NOTIFICATION: 0x1f
[    5.729605] iwlwifi 0000:00:

alba4k, 2023-08-17:
so the led does light up for you?

alba4k, 2023-08-17:
it does not for me

striczkof, 2023-08-17:
Interesting, did it light up on Windows and when you used to have it working?

alba4k, 2023-08-17:
Yes it used to light up.

I have never had windows installed on this laptop, so I can't exactly tell you about that, but I would expect that to work

striczkof, 2023-08-17:
Then that's weird because I swear it lighted up after I rebooted only to find a broken install, let me check again

striczkof, 2023-08-17:
@alba4k okay now it doesn't light up anymore.

Lights up and works before rebooting after upgrade, lights up and doesn't work after reboot, doesn't light up after second reboot.

Thoughts?

alba4k, 2023-08-18:
BTW Completely unrelated, just noticed this
[image]
[image]

How the hell did I type  \ there, its on the complete opposite side of the keyboard when compared to ?

striczkof, 2023-08-18:
That's new. Does the LED light up at least since the kernel loads the firmware now in your dmesg?

Also, I have reinstalled EOS, with internet so fully updated already, installed the packages with the script, then even ran `eos-update --yay` for good measure and the camera is still working after multiple reboots. So now, I honestly think there are offending packages in my first EOS install that broke the stack. I'm gonnna reinstall the kernel and see if that breaks it.

alba4k, 2023-08-18:
The led never lights up

striczkof, 2023-08-18:
Then that's weird. Literally the first instance of this, but I will try installing OBS in my now broken EOS camera stack and see if it appears too.

By the way, I managed to break my EndeavourOS camera stack by literally just reinstalling linux and linux-headers then reboot. What the fuck could possibly cause it?

<details><summary>pacman -S linux linux-headers</summary>
<p>

```
warning: linux-6.4.10.arch1-1 is up to date -- reinstalling
warning: linux-headers-6.4.10.arch1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Package (2)         Old Version     New Version     Net Change  Download Size

core/linux          6.4.10.arch1-1  6.4.10.arch1-1    0.00 MiB     125.53 MiB
core/linux-headers  6.4.10.arch1-1  6.4.10.arch1-1    0.00 MiB      29.79 MiB

Total Download Size:   155.32 MiB
Total Installed Size:  285.13 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] :: Retrieving packages...
 linux-6.4.10.arch1-1-x86_64 downloading...
 linux-headers-6.4.10.arch1-1-x86_64 downloading...
error: failed retrieving file 'linux-6.4.10.arch1-1-x86_64.pkg.tar.zst' from mirror.aarnet.edu.au : The requested URL returned error: 404
error: failed retrieving file 'linux-headers-6.4.10.arch1-1-x86_64.pkg.tar.zst' from mirror.aarnet.edu.au : The requested URL returned error: 404
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
:: Running pre-transaction hooks...
(1/1) Remove upgraded DKMS modules
==> dkms remove --no-depmod ipu6-drivers/r131.c3b762b27 -k 6.4.10-arch1-1
==> dkms remove --no-depmod v4l2loopback/r10.cae2954 -k 6.4.10-arch1-1
:: Processing package changes...
reinstalling linux...
reinstalling linux-headers...
:: Running post-transaction hooks...
(1/6) Arming ConditionNeedsUpdate...
(2/6) Updating module dependencies...
(3/6) Install DKMS modules
==> dkms install --no-depmod v4l2loopback/r10.cae2954 -k 6.4.10-arch1-1
==> dkms install --no-depmod ipu6-drivers/r131.c3b762b27 -k 6.4.10-arch1-1
==> depmod 6.4.10-arch1-1
(4/6) Running kernel-install...
:: kernel-install installing kernel 6.4.10-arch1-1
dracut: Executing: /usr/bin/dracut --no-hostonly --force /efi/a4eb782afbbd4fd6a2f0a115ed410757/6.4.10-arch1-1/initrd-fallback 6.4.10-arch1-1
dracut: dracut module 'dash' will not be installed, because command 'dash' could not be found!
dracut: dracut module 'mksh' will not be installed, because command 'mksh' could not be found!
dracut: dracut module 'busybox' will not be installed, because command 'busybox' could not be found!
dracut: dracut module 'dbus-broker' will not be installed, because command 'dbus-broker' could not be found!
dracut: dracut module 'rngd' will not be installed, because command 'rngd' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmand' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmanctl' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmand-wait-online' could not be found!
dracut: dracut module 'network-wicked' will not be installed, because command 'wicked' could not be found!
dracut: dracut module 'dmraid' will not be installed, because command 'kpartx' could not be found!
dracut: dracut module 'multipath' will not be installed, because command 'multipath' could not be found!
dracut: dracut module 'tpm2-tss' will not be installed, because command 'tpm2' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'dcbtool' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fipvlan' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'lldpad' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fcoemon' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fcoeadm' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'dcbtool' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'fipvlan' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'lldpad' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsi-iname' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsiadm' could not be found!
[... 145 lines cut ...]
dracut: *** Hardlinking files ***
dracut: Mode:                     real
dracut: Method:                   sha256
dracut: Files:                    774
dracut: Linked:                   1 files
dracut: Compared:                 0 xattrs
dracut: Compared:                 40 files
dracut: Saved:                    690 B
dracut: Duration:                 0.003168 seconds
dracut: *** Hardlinking files done ***
dracut: *** Generating early-microcode cpio image ***
dracut: *** Using microcode found in '/boot/intel-ucode.img' ***
dracut: *** Store current command line parameters ***
dracut: *** Stripping files ***
dracut: *** Stripping files done ***
dracut: *** Creating image file '/efi/a4eb782afbbd4fd6a2f0a115ed410757/6.4.10-arch1-1/initrd' ***
dracut: *** Creating initramfs image file '/efi/a4eb782afbbd4fd6a2f0a115ed410757/6.4.10-arch1-1/initrd' done ***
(5/6) Check if user should be informed about rebooting after certain system package upgrades.
(6/6) Checking which packages need to be rebuilt
```

</p>
</details> 

My hypothesis is that that libkmod warnings might be related but idk

striczkof, 2023-08-18:
@alba4k do you use dracut?

alba4k, 2023-08-18:
nope

alba4k, 2023-08-19:
Also during todays kernel update I got this
[image]

(Unable to get information related to the file XYZ)

I have already uninstalled using the uninstall script

striczkof, 2023-09-03:
@alba4k do you use kernel-install? (please say yes, i'm gonna jump off a cliff if you don't xd)

In EndeavourOS, reinstalling the kernel itself, triggering kernel-install is enough to break the stack yet reinstalling linux-headers just to trigger DKMS reinstalls does not.

<details><summary>sudo pacman -S linux # ran in EndeavourOS</summary>
<p>

```
warning: linux-6.4.12.arch1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Package (1)  Old Version     New Version     Net Change

core/linux   6.4.12.arch1-1  6.4.12.arch1-1    0.00 MiB

Total Installed Size:  125.56 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] 
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
:: Processing package changes...
reinstalling linux...
:: Running post-transaction hooks...
(1/5) Arming ConditionNeedsUpdate...
(2/5) Updating module dependencies...
(3/5) Running kernel-install...
:: kernel-install installing kernel 6.4.12-arch1-1
dracut: Executing: /usr/bin/dracut --no-hostonly --force /efi/0dc24d105e0c43d0a6a4e27b94fac3a4/6.4.12-arch1-1/initrd-fallback 6.4.12-arch1-1
dracut: dracut module 'dash' will not be installed, because command 'dash' could not be found!
dracut: dracut module 'mksh' will not be installed, because command 'mksh' could not be found!
dracut: dracut module 'busybox' will not be installed, because command 'busybox' could not be found!
dracut: dracut module 'dbus-broker' will not be installed, because command 'dbus-broker' could not be found!
dracut: dracut module 'rngd' will not be installed, because command 'rngd' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmand' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmanctl' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmand-wait-online' could not be found!
dracut: dracut module 'network-wicked' will not be installed, because command 'wicked' could not be found!
dracut: dracut module 'dmraid' will not be installed, because command 'kpartx' could not be found!
dracut: dracut module 'multipath' will not be installed, because command 'multipath' could not be found!
dracut: dracut module 'tpm2-tss' will not be installed, because command 'tpm2' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'dcbtool' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fipvlan' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'lldpad' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fcoemon' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fcoeadm' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'dcbtool' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'fipvlan' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'lldpad' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsi-iname' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsiadm' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsid' could not be found!
dracut: dracut module 'nbd' will not be installed, because command 'nbd-client' could not be found!
dracut: dracut module 'nvmf' will not be installed, because command 'nvme' could not be found!
dracut: dracut module 'biosdevname' will not be installed, because command 'biosdevname' could not be found!
dracut: dracut module 'memstrack' will not be installed, because command 'memstrack' could not be found!
dracut: memstrack is not available
dracut: If you need to use rd.memdebug>=4, please install memstrack and procps-ng
dracut: dracut module 'squash' will not be installed, because command 'mksquashfs' could not be found!
dracut: dracut module 'squash' will not be installed, because command 'unsquashfs' could not be found!
dracut: *** Including module: systemd ***
dracut: *** Including module: systemd-initrd ***
dracut: *** Including module: modsign ***
dracut: *** Including module: i18n ***
dracut: *** Including module: btrfs ***
dracut: *** Including module: crypt ***
dracut: *** Including module: dm ***
[... 127 lines cut ...]
dracut: *** Resolving executable dependencies ***
dracut: *** Resolving executable dependencies done ***
dracut: *** Hardlinking files ***
dracut: Mode:                     real
dracut: Method:                   sha256
dracut: Files:                    774
dracut: Linked:                   1 files
dracut: Compared:                 0 xattrs
dracut: Compared:                 34 files
dracut: Saved:                    690 B
dracut: Duration:                 0.001782 seconds
dracut: *** Hardlinking files done ***
dracut: *** Generating early-microcode cpio image ***
dracut: *** Using microcode found in '/boot/intel-ucode.img' ***
dracut: *** Store current command line parameters ***
dracut: *** Stripping files ***
dracut: *** Stripping files done ***
dracut: *** Creating image file '/efi/0dc24d105e0c43d0a6a4e27b94fac3a4/6.4.12-arch1-1/initrd' ***
dracut: *** Creating initramfs image file '/efi/0dc24d105e0c43d0a6a4e27b94fac3a4/6.4.12-arch1-1/initrd' done ***
(4/5) Check if user should be informed about rebooting after certain system package upgrades.
(5/5) Checking which packages need to be rebuilt
```

</p>
</details>

alba4k, 2023-09-03:
yeaa sorry I don't even know what that is

I will try reinstalling the kernel to confirm, but I don't think it's being used, sorryyy

striczkof, 2023-09-03:
you should try uninstalling all dkms packages you have, then nuke `/lib/modules/CURRENT_VERSION/updates/dkms`. I will try that as well in a moment after installing a simple Arch installation again

alba4k, 2023-09-03:
already tried and that sadly did nope help

hinricht, 2023-09-04:
> @alba4k do you use dracut?

I do: 

```
~ $ pacman -Q|grep drac
dracut 059-0.4
kernel-install-for-dracut 1.9.1-1
```

striczkof, 2023-09-04:
> > @alba4k do you use dracut?
> 
> I do:
> 
> ```
> ~ $ pacman -Q|grep drac
> dracut 059-0.4
> kernel-install-for-dracut 1.9.1-1
> ```

Yup, must be kernel-install. Making another arch install with kernel install enabled and lets see what happens

striczkof, 2023-10-02:
Why did I forget to answer to this question :/
kernel-install actually comes with systemd. You will see it in action whenever you install a kernel.

@alba4k and anyone else around. can you do these for me?
1. Get my latest PR: `git clone --branch fix_and_rework https://github.com/striczkof/archlinux-ipu6-webcam.git && cd archlinux-ipu6-webcam`
2. Run the uninstall script to clean any potential leftover packages: `./uninstall.sh`
3. Get rid of your GStreamer plugin blacklists: `rm ~/.cache/gstreamer-1.0/registry.x86_64.bin`
4. Install the whole thing then reboot: `./install.sh`
5. Run these tests then send me the results (through copying or redirecting outputs to log files):
```sh
./test.sh
sudo dmesg | grep ipu
systemctl status v4l2-relayd
v4l2-ctl --list-devices
gst-inspect-1.0 icamerasrc # (if this is empty, icamerasrc should be blacklisted still)
gst-inspect 1.0 # (first few lines show blacklist reasons, last line/s show why)
```

hinricht, 2023-10-02:
@striczkof Thanks for getting back to this!
Here's the ouput:

[d.md](https://github.com/stefanpartheym/archlinux-ipu6-webcam/files/12780031/d.md)

striczkof, 2023-10-02:
that was hard to read xd

But it does show that your icamerasrc is blacklisted. Assuming you already deleted the registry cache. Can I see your `groups`?

hinricht, 2023-10-02:
Sure, here you go:

<details>
  <summary>Click me</summary>

```
$ cat /etc/group
root:x:0:root
nobody:x:65534:
adm:x:999:daemon
wheel:x:998:varac
utmp:x:997:
audio:x:996:
disk:x:995:
input:x:994:
kmem:x:993:
kvm:x:992:
lp:x:991:cups,varac
optical:x:990:
render:x:989:
sgx:x:988:
storage:x:987:
tty:x:5:
uucp:x:986:
video:x:985:greeter,varac
users:x:984:
sys:x:3:bin
mem:x:8:
ftp:x:11:
mail:x:12:
log:x:19:
smmsp:x:25:
proc:x:26:polkitd
games:x:50:
lock:x:54:
network:x:90:varac
floppy:x:94:
scanner:x:96:
power:x:98:varac
systemd-journal:x:983:
rfkill:x:982:varac
dbus:x:81:
bin:x:1:daemon
daemon:x:2:bin
http:x:33:
systemd-coredump:x:981:
systemd-network:x:980:
systemd-oom:x:979:
systemd-journal-remote:x:978:
systemd-resolve:x:977:
systemd-timesync:x:976:
tss:x:975:
uuidd:x:68:
dhcpcd:x:974:
dnsmasq:x:973:
rpc:x:32:
rpcuser:x:34:
seat:x:972:
avahi:x:971:
geoclue:x:970:
git:x:969:
greeter:x:968:
polkitd:x:102:
rtkit:x:133:
usbmux:x:140:
hinricht:x:1000:
locate:x:21:
tox-bootstrapd:x:199:
node_exporter:x:967:
gitlab-runner:x:107:
colord:x:966:
saned:x:965:
cups:x:209:
systemd-journal-upload:x:964:
fwupd:x:963:
  
```
</details>

striczkof, 2023-10-02:
whoops, I mean I want to know what's your user's group, like what shows up when you type in `groups` in terminal

hinricht, 2023-10-02:
```
$ groups
network power rfkill video lp wheel hinricht
```

striczkof, 2023-10-02:
so you do have video group.

I seriously don't get this one.
Ha! time to reinstall EndeavourOS again and see what I get from gst-inspect

hinricht, 2023-10-02:
Sorry that my case is such a hard one :disappointed: 
If you want to use the same setup as I do: I'm running [Manjaro Sway Editor](https://github.com/manjaro-sway/manjaro-sway) updated to the latest package updates as of today (+ dracut, kernel-install-for-dracut).

striczkof, 2023-10-02:
FUCK

KERNEL-INSTALL + DRACUT

PERFECT

alba4k, 2023-10-02:
Hi, I will post everything this evening (~16 UTC), as I am currently at school

striczkof, 2023-10-02:
Thanks! Based on my testing, the icamerasrc blacklist issue appears to be irrelevant anyways as the kernel_install part of my kernel upgrade kills the camera stack permanently anyway by causing the firmware to fail loading. Gotta try to narrow it down furthermore. But fresh logs are still awesome to work on.

@alba4k can you send me a log of your `pacman -S linux` and let me see if you have a kernel-install set up?

alba4k, 2023-10-02:
sorry for the delay. I am currently installing from your fork, will post the outputs of everything asap

alba4k, 2023-10-02:
@striczkof Update on this

As someone already mentioned in #42, The build seems to fail with `==> ERROR: 70-ipu6-psys.rules was not found in the build directory and is not a URL.` (Related to icamerasrc-git`

striczkof, 2023-10-02:
Oh man, I forgot about that file again. It's added manually by package function but I forgot to remove it from the sources oops

alba4k, 2023-10-02:
Should be fixed now, let's see how this goes

[23 more replies]

closed 2023-11-13 by stefanpartheym