stefanpartheym/archlinux-ipu6-webcam #24: 70-ipu6-psys.rules was not found in the build directory and is not a URL. / failed to install icamerasrc-git

tkellen, 2023-06-12:
First, thank you for making this project! I just ran the install on my machine:
```sh
❯ uname -a
Linux devbox 6.3.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Mon, 01 May 2023 17:42:39 +0000 x86_64 GNU/Linux
```

I got the following failure:
```sh
# Build and install package: icamerasrc-git
~/archlinux-ipu6-webcam/icamerasrc-git ~/archlinux-ipu6-webcam
==> Making package: icamerasrc-git-fix r59.17841ab-1 (Tue 13 Jun 2023 11:26:02 AM NZST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning icamerasrc git repo...
Cloning into bare repository '/home/tkellen/archlinux-ipu6-webcam/icamerasrc-git/icamerasrc'...
remote: Enumerating objects: 696, done.
remote: Counting objects: 100% (101/101), done.
remote: Compressing objects: 100% (51/51), done.
remote: Total 696 (delta 55), reused 84 (delta 49), pack-reused 595
Receiving objects: 100% (696/696), 475.69 KiB | 2.15 MiB/s, done.
Resolving deltas: 100% (496/496), done.
==> ERROR: 70-ipu6-psys.rules was not found in the build directory and is not a URL.
~/archlinux-ipu6-webcam
ERROR: Failed to install: icamerasrc-git
```

I can't find `70-ipu6-psys.rules` anywhere in HEAD:

```sh
❯ find . -type f -name "*.rules"
./intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/config/linux/rules.d/60-intel-ipu-adlp.rules
./intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/config/linux/rules.d/60-intel-ipu-tglu.rules
```

Any suggestions on a next step? I'd love to stop carrying around an external webcam!

tkellen, 2023-06-13:
Whoops, just realized I hadn't installed the latest kernel. Same behavior with:
```sh
❯ uname -a
Linux devbox 6.3.7-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 10 Jun 2023 00:35:35 +0000 x86_64 GNU/Linux
```

Rend0e, 2023-06-13:
I had the same issue and fixed it manually by adding the 70-ipu6-psys.rules file to the icamerasrc-git folder.
Got the file from https://aur.archlinux.org/cgit/aur.git/tree/70-ipu6-psys.rules?h=icamerasrc-git
```
# This rule makes the Intel IPU6 sensor accessible to members of group 'video',
# allowing icamerasrc to be used as an unprivileged user.
SUBSYSTEM=="intel-ipu6-psys",GROUP="video",MODE="0660"
```
From the disccusions in https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/22#issuecomment-1586978451, it will be fixed when https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/23 gets merged.

tkellen, 2023-06-14:
Thanks for the help @Rend0e! I searched the repo for this error and was unable to find it. Not sure how I missed the references you shared!

closed 2023-06-14 by tkellen