stefanpartheym/archlinux-ipu6-webcam #26: Not working - manjaro 6.3.5

naps62, 2023-06-18:
kernel: `6.3.5-2-MANJARO`
laptop: `Dell XPS 13 Plus`

After going through the setup in this repo, the webcam still doesn't work

v4l2 shows me the devices:
```
❯ v4l2-ctl --list-devices
ipu6 (PCI:pci:pci0000:00):
	/dev/video1

ipu6 (pci:pci0000:00):
	/dev/media0

Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
```

chrome/firefox/obs all fail

and ffmpeg:
```
❯ ffmpeg -f v4l2 -framerate 25 -video_size 640x480 -i /dev/video0 output.mkv
ffmpeg version n6.0 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.1.1 (GCC) 20230429
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm--enable-libiec61883 --enable-libjack --enable-libjxl --enable-libmfx --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-version3 --enable-vulkan
  libavutil      58.  2.100 / 58.  2.100
  libavcodec     60.  3.100 / 60.  3.100
  libavformat    60.  3.100 / 60.  3.100
  libavdevice    60.  1.100 / 60.  1.100
  libavfilter     9.  3.100 /  9.  3.100
  libswscale      7.  1.100 /  7.  1.100
  libswresample   4. 10.100 /  4. 10.100
  libpostproc    57.  1.100 / 57.  1.100
[video4linux2,v4l2 @ 0x55da6f00c380] Not a video capture device.
/dev/video0: No such device
```

striczkof, 2023-06-30:
Can you check what's on `systemctl status v4l2-relayd`?

naps62, 2023-06-30:
```
v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: disabled)
    Drop-In: /etc/systemd/system/v4l2-relayd.service.d
             └─override.conf
     Active: active (running) since Fri 2023-06-30 09:32:54 WEST; 1h 44min ago
    Process: 591 ExecCondition=/usr/bin/test -n ${VIDEOSRC} (code=exited, status=0/SUCCESS)
    Process: 630 ExecCondition=/usr/bin/test -n $FORMAT (code=exited, status=0/SUCCESS)
    Process: 633 ExecCondition=/usr/bin/test -n $WIDTH (code=exited, status=0/SUCCESS)
    Process: 635 ExecCondition=/usr/bin/test -n $HEIGHT (code=exited, status=0/SUCCESS)
    Process: 641 ExecCondition=/usr/bin/test -n $FRAMERATE (code=exited, status=0/SUCCESS)
    Process: 643 ExecCondition=/usr/bin/test -n ${CARD_LABEL} (code=exited, status=0/SUCCESS)
   Main PID: 645 (v4l2-relayd)
      Tasks: 3 (limit: 38015)
     Memory: 12.1M
        CPU: 46ms
     CGroup: /system.slice/v4l2-relayd.service
             └─645 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink name=v4l2sink device=/dev/video0

jun 30 09:32:53 arrakis systemd[1]: Starting v4l2-relay daemon service...
jun 30 09:32:54 arrakis systemd[1]: Started v4l2-relay daemon service.
jun 30 09:32:54 arrakis v4l2-relayd[645]: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
```

striczkof, 2023-06-30:
That's literally the same error I get when I first ran the script but didn't restart. I'd say restart your PC but I assume you've done that already lol

Give me a day or two so I can install Manjaro to my XPS 13 Plus and see what's what

striczkof, 2023-07-01:
@naps62 Out of topic: how did you install Manjaro to your laptop? I used dd to copy Manjaro ISO to my flash drive as directed by Manjaro Wiki but it's not showing up on my UEFI boot options :/

naps62, 2023-07-01:
@striczkof i used to have similar issues. Nowadays I have a usb stick with Ventoy installed (https://www.ventoy.net/en/index.html)

It makes it so you can just copy iso files into it, and it handles all the boot complexities (and gives you a menu, so you can keep multiple distros ready to boot)

striczkof, 2023-07-01:
@naps62 so ye I just gave up and used ventoy because idk why ISO won't work with cp nor dd.

Funny thing now though, my webcam is working after installing and restarting, no tinkering done other than install git and linux61-header (using linux61 since it came with manjaro fresh install) so now I am totally clueless what happened to both @hinricht 's (#21 ) and your setup.
GIve me a bit and I will try with Linux 6.3.

I'm wondering, are you using 12th gen or 13th gen XPS 13 Plus? Highly doubt that should matter but dunno 

My environment:
- Laptop: `XPS 13 Plus 9320 with i7-1260P`
- OS: `Manjaro 22.1.1`
- Desktop Environment: `Xfce 4.18`
- Kernel: `6.1.31-2-MANJARO`
- Kernel packages: `linux61` `linux61-headers`

Working on Firefox and Discord after restart
Apologies for the German outputs. I forgor again :skull: 
Logs:

<details><summary>install.sh</summary>

```
Warnung: linux61-headers-6.1.31-2 ist aktuell -- Überspringe
 Es gibt nichts zu tun
# Install dependency for intel-ipu6-dkms-git
:: Löse Abhängigkeiten auf…
:: Berechne Konflikte…
:: Berechne interne Konflikte…

Aur (1) intel-ivsc-firmware-r2.764bd6e-1

:: Weiter überprüfen? [J/n]: 

:: Lade PKGBUILDs herunter…
 (1/1) Lädt herunter: intel-ivsc-firmware-r2.764bd6e-1
frage devel info ab…
==> Erstelle Paket: intel-ivsc-firmware r2.764bd6e-1 (Sat 01 Jul 2023 20:49:09)
==> Empfange Quellen...
  -> Klone das ivsc-firmware git Repo...
Klone in Bare-Repository '/home/striczkof/.cache/paru/clone/intel-ivsc-firmware/ivsc-firmware' ...
==> Überprüfe source Dateien mit sha256sums...
    ivsc-firmware ... Übersprungen
==> Erstelle Paket: intel-ivsc-firmware r2.764bd6e-1 (Sat 01 Jul 2023 20:49:11)
==> Prüfe Laufzeit-Abhängigkeiten...
==> Prüfe Buildtime-Abhängigkeiten...
==> Empfange Quellen...
  -> Aktualisiere das ivsc-firmware git Repo...
==> Überprüfe source Dateien mit sha256sums...
    ivsc-firmware ... Übersprungen
==> Entferne existierendes $srcdir/ Verzeichnis...
==> Entpacke Quellen...
  -> Erstelle Arbeitskopie des ivsc-firmware git Repos...
Klone nach 'ivsc-firmware'...
Fertig.
==> Beginne pkgver()...
==> Aktualisierte Version: intel-ivsc-firmware r8.29c5eff-1
==> Quellen sind fertig.
intel-ivsc-firmware-r2.764bd6e-1: Parse pkg Liste…
==> Erstelle Paket: intel-ivsc-firmware r8.29c5eff-1 (Sat 01 Jul 2023 20:49:12)
==> Prüfe Laufzeit-Abhängigkeiten...
==> Prüfe Buildtime-Abhängigkeiten...
==> WARNUNG: Verwende bestehenden $srcdir/ Baum
==> Beginne pkgver()...
==> Betrete fakeroot Umgebung...
==> Beginne package()...
==> Säubere Installation...
  -> Entferne libtool Dateien...
  -> Bereinige ungewollte Dateien...
  -> Entferne statische Bibliotheken...
  -> Entferne unnötige Symbole aus Binär-Dateien und Bibliotheken...
  -> Komprimiere Man-Pages und Info-Seiten...
[... 1331 lines cut ...]
Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
```
</details> 

<details><summary>systemctl status v4l2-relayd.service</summary>

```
● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: disabled)
    Drop-In: /etc/systemd/system/v4l2-relayd.service.d
             └─override.conf
     Active: active (running) since Sat 2023-07-01 20:56:03 AEST; 1h 3min ago
   Main PID: 978 (v4l2-relayd)
      Tasks: 5 (limit: 38009)
     Memory: 693.9M
        CPU: 20.460s
     CGroup: /system.slice/v4l2-relayd.service
             └─978 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2sink name=v4l2sink device=/dev/video0"

Jul 01 20:56:02 alvins-xps9320 systemd[1]: Starting v4l2-relay daemon service...
Jul 01 20:56:03 alvins-xps9320 systemd[1]: Started v4l2-relay daemon service.
Jul 01 20:59:45 alvins-xps9320 v4l2-relayd[978]: Registering meta implementation 'GstCamerasrcMeta' without init function
```
</details>

naps62, 2023-07-03:
@striczkof I just realized I failed to mention a probably important detail.
I use manjaro-sway, with wayland instead of x11 (this is so natural to me nowadays that I forgot to mention). not sure what the default desktop is in manjaro nowadays

I'm also using hyprland as wm instead of sway, although I tested booting with sway again with the same result

hinricht, 2023-07-03:
Same for me, I'm also using [manajro-sway](https://github.com/manjaro-sway/manjaro-sway) with Wayland (Kernel 6.3.5-2-MANJARO). Manjaro-sway is a rolling release so I don't have any version number for it.

striczkof, 2023-07-03:
Never used a tiling DE before, give me a bit lol

striczkof, 2023-07-06:
Uh guys, what?

Manjaro-Sway latest fresh install

<details><summary>install.sh</summary>

```
Sync Explicit (1): linux64-headers-6.4.1-5 warning: linux64-headers-6.4.1-5 is up to date skipping there is nothing to do # Install dependency for intel-ipu6-dkms-git AUR Explicit (1): intel-ivsc-firmware-r2.764bd6e-1 :: [1m(1/1) Downloaded PKGBUILD: intel-ivsc-firmware[0m 1 intel-ivsc-firmware (Build Files Exist) ==> Packages to cleanBuild? ==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4) ==> 1 intel-ivsc-firmware (Build Files Exist) ==> Diffs to show? ==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4) ==> ==> Making package: intel-ivsc-firmware r2.764bd6e-1 (Thu 06 Jul 2023 12:24:41) ==> Retrieving sources... -> Cloning ivsc-firmware git repo... Cloning into bare repository '/home/striczkof/.cache/yay/intel-ivsc-firmware/ivsc-firmware'... ==> Validating source files with sha256sums... ivsc-firmware ... Skipped :: [1m(1/1) Parsing SRCINFO: intel-ivsc-firmware[0m ==> Making package: intel-ivsc-firmware r2.764bd6e-1 (Thu 06 Jul 2023 12:24:43) ==> Checking runtime dependencies... ==> Checking buildtime dependencies... ==> Retrieving sources... -> Updating ivsc-firmware git repo... ==> Validating source files with sha256sums... ivsc-firmware ... Skipped ==> Removing existing $srcdir/ directory... ==> Extracting sources... -> Creating working copy of ivsc-firmware git repo... Cloning into 'ivsc-firmware'... done. ==> Starting pkgver()... ==> Updated version: intel-ivsc-firmware r8.29c5eff-1 ==> Sources are ready. ==> Making package: intel-ivsc-firmware r8.29c5eff-1 (Thu 06 Jul 2023 12:24:45) ==> Checking runtime dependencies... ==> Checking buildtime dependencies... ==> WARNING: Using existing $srcdir/ tree ==> Starting pkgver()... ==> Entering fakeroot environment... ==> Starting package()... ==> Tidying install... -> Removing libtool files... -> Purging unwanted files... -> Removing static library files... -> Stripping unneeded symbols from binaries and libraries... -> Compressing man and info pages... ==> Checking for packaging issues... ==> Creating package "intel-ivsc-firmware"... -> Generating .PKGINFO file... -> Generating .BUILDINFO file... -> Generating .MTREE file... -> Compressing package... ==> Leaving fakeroot environment. ==> Finished making: intel-ivsc-firmware r8.29c5eff-1 (Thu 06 Jul 2023 12:24:46) ==> Cleaning up... loading packages... resolving dependencies... looking for conflicting packages... Packages (1) intel-ivsc-firmware-r8.29c5eff-1 Total Installed Size: 9.45 MiB :: Proceed with installation? [Y/n] checking keyring... checking package integrity... loading package files... checking for file conflicts... checking available disk space... :: Processing package changes... installing intel-ivsc-firmware... :: Running post-transaction hooks... (1/1) Arming ConditionNeedsUpdate... => SUCCESS # Build and install package: intel-ipu6-dkms-git ~/archlinux-ipu6-webcam/intel-ipu6-dkms-git ~/archlinux-ipu6-webcam ==> Making package: intel-ipu6-dkms-git-fix r100.8c02a846d-1 (Thu 06 Jul 2023 12:24:47) ==> Checking runtime dependencies... ==> Installing missing dependencies... resolving dependencies... looking for conflicting packages... Packages (5) gc-8.2.4-1 guile-3.0.9-1 make-4.4.1-2 patch-2.7.6-10 dkms-3.0.11-2 Total Download Size: 8.97 MiB Total Installed Size: 55.69 MiB :: Proceed with installation? [Y/n] :: Retrieving packages... guile-3.0.9-1-x86_64 downloading... make-4.4.1-2-x86_64 downloading... gc-8.2.4-1-x86_64 downloading... patch-2.7.6-10-x86_64 downloading... dkms-3.0.11-2-any downloading... checking keyring... checking package integrity... loading package files... checking for file conflicts... checking available disk space... :: Processing package changes... installing gc... installing guile... installing make... installing patch... Optional dependencies for patch ed: for patch -e functionality installing dkms... Optional dependencies for dkms linux-headers: build modules against the Manjaro kernel [installed] :: Running post-transaction hooks... (1/2) Arming ConditionNeedsUpdate... (2/2) Updating the info directory file... ==> Checking buildtime dependencies... ==> Retrieving sources... -> Cloning ipu6-drivers git repo... Cloning into bare repository '/home/striczkof/archlinux-ipu6-webcam/intel-ipu6-dkms-git/ipu6-drivers'... ==> Validating source files with sha256sums... ipu6-drivers ... Skipped ==> Extracting sources... -> Creating working copy of ipu6-drivers git repo... Cloning into 'ipu6-drivers'... done. Switched to a new branch 'makepkg' ==> Starting prepare()... Cloning into 'ivsc-driver'... ==> Starting pkgver()... ==> Entering fakeroot environment... ==> Starting package()... ==> Tidying install... -> Removing libtool files... -> Purging unwanted files... -> Removing static library files... -> Stripping unneeded symbols from binaries and libraries... -> Compressing man and info pages... ==> Checking for packaging issues... ==> Creating package "intel-ipu6-dkms-git-fix"... -> Generating .PKGINFO file... -> Generating .BUILDINFO file... -> Generating .MTREE file... -> Compressing package... ==> Leaving fakeroot environment. ==> Finished making: intel-ipu6-dkms-git-fix r100.8c02a846d-1 (Thu 06 Jul 2023 12:25:36) ==> Installing package intel-ipu6-dkms-git-fix with pacman -U... loading packages... resolving dependencies... looking for conflicting packages... Packages (1) intel-ipu6-dkms-git-fix-r100.8c02a846d-1 Total Installed S [... cut ...] LIBC_PTHREAD -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success -- Found Threads: TRUE -- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) CMake Error at cmake/FindIA_IMAGING.cmake:28 (message): IA_IMAGING not found Call Stack (most recent call first): CMakeLists.txt:278 (find_package) -- Configuring incomplete, errors occurred! ==> ERROR: A failure occurred in build(). Aborting... ~/archlinux-ipu6-webcam [0;31mERROR:[0m Failed to install: intel-ipu6ep-camera-hal-git
```

</details>

striczkof, 2023-07-06:
So uh, @hinricht think I might need more details from you lol like the full install logs even if it is successful (don't forget to delete then clone the repo again) since i manage to get test.sh working.

Funny thing now though, I am experiencing exactly the same problem as @naps62 with Firefox not detecting my camera and v4l2-relayd showing "gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed"  Might be missing some plugins or so that somehow present on most Arch installation but not for manjaro-sway. Doing some more testing.

striczkof, 2023-07-06:
YOOO @naps62 I managed to fix it! Thanks to @stefanpartheym post in https://bbs.archlinux.org/viewtopic.php?id=277462&p=3 (hey if you see this can you merge our PR xd)

1. install gst-plugins-good and gst-plugins-bad (tbh idk which one did it lol): `sudo pacman -S gst-plugins-good gst-plugins-bad`
2. Reload v4l2loopback: `sudo modprobe -r v4l2loopback && sudo modprobe v4l2loopback`
3. Restart v4l2-relayd service: `sudo systemctl restart v4l2-relayd`
4. ????
5. PROFIT!!!

naps62, 2023-07-06:
@striczkof 🎊 🎊 🎊 

`gst-plugins-good` did it for me. I installed only that one

webcam tested on Chrome / Slack / Discord / Firefox. `--workaround` was only needed for Firefox, as far as I can tell
is it useful to remove the workaround considering I mainly use chrome? not sure if there are performance benefits

---

this next part is offtopic, but other people may end up here through google:

my last hanging fruit was the microphone, which also worked on Ubuntu, but only recorded white noise in Arch.
Just found the solution here: https://bbs.archlinux.org/viewtopic.php?pid=2058151#p2058151

> In the not working case, can you restore it  by running
> 
> `sudo alsactl init`
>
> prevent it from breaking in the first place by running
> 
> `sudo alsactl store`

striczkof, 2023-07-06:
Great! That means one less dependency to install. Updated the PR already.

I guess conversion might affect the latency a bit? The only way to find out is to compare them really. Remove it if you don't need the work around I guess. It now works without the workaround for Firefox 115 so it's not even needed for that.

Problem is my Discord clients (Regular, Arch Electron, and Canary) are still not working without the workaround. @naps62 what client are you using?

naps62, 2023-07-06:
I use he regular one, just with this config change (but I'm pretty sure it's irrelevant for this)

`Exec=/usr/bin/discord --enable-features=UseOzonePlatform --ozone-platform=wayland`

striczkof, 2023-07-07:
> I use he regular one, just with this config change (but I'm pretty sure it's irrelevant for this)
> 
> `Exec=/usr/bin/discord --enable-features=UseOzonePlatform --ozone-platform=wayland

Not working on both Arch and Manjaro and with and without options :(

PR #28 (2023-07-12): Added hibernation workaround, removed ipu6-driver-fix, used getopts for options
Added a temporary hack that should fix #11 caused by i2c_ljca module. This should be fixed upstream later on since there is already a PR waiting for merge.

I also removed intel-ipu6-dkms-git-fix since the latest one from the AUR is working perfectly. (with already merged ivsc-driver)

Used getopts to handle multiple options in the installer script. Now, you have to use '-a' flag for Firefox/Discord workarounds and '-s' for the hibernation hack, or '-as' for both.

Merged PR #27 so I can work with a cleaner install [... cut]
closed 2023-07-12 by stefanpartheym