stefanpartheym/archlinux-ipu6-webcam #31: Hibernate workaround breaks camera on wake up

zemen, 2023-08-01:
I tried to install the driver with `./install.sh -as` on Thinkpad carbon X1 gen 10. The workaround indeed helps to wake up without freeze after hibernation, then driver loads and there are no errors in journal. However neither gstreamer nor v4l2-relayd doesn't work until next reboot.

Another issue I found is that suspend(which is triggered on my laptop on lid close) is not supported in workaround.

striczkof, 2023-08-12:
that's odd. that worked fine for me. prolly because Intel pushed some new stuff. checking this morning.

striczkof, 2023-08-12:
seeing that intel has pushed the newer branches into main that has the i2c_ljca fix. I am kinda optimistic that the hibernation hack is no longer needed. Doing some testing then making a PR. Give me around 5 hours lol

striczkof, 2023-08-13:
@zemen can you give me the output of `cat /sys/power/mem_sleep`? My XPS 13 Plus is suspending as [s2idle] and can't use any other options so wondering what you have.

Also, the camera is working fine after a hibernation for me without modifications to the script. I doubt this will work but can you remove the post stage from the hibernation script? The workaround should look like this:
```shell
stage=$1
op=$2

case $stage in
  pre)
    case $op in
      hibernate|hybrid-sleep)
        /usr/bin/modprobe -r i2c_ljca
        ;;
      suspend-then-hibernate)
        [ "$SYSTEMD_SLEEP_ACTION" = "hibernate" ] && /usr/bin/modprobe -r i2c_ljca
        ;;
    esac
    ;;
esac
```

striczkof, 2023-08-13:
Wait, I just reread the original issue on ivsc-driver again. It might turn out that you are having the literally same issue as this https://github.com/intel/ivsc-driver/issues/32#issuecomment-1589730406 . Go check it out and see if the logs match or something @zemen

alba4k, 2023-11-19:
Works perfectly for me. After waking up, the camera doesn't work for a couple of seconds (which is expected as the modules are still being loaded by the workaround). Then the led flashes for about half a second and the camera starts working again.

stefanpartheym, 2024-03-01:
I'll close this one now du to inactivity and it seems to work for other people.
@zemen Please consider reopening, if necessary.