stefanpartheym/archlinux-ipu6-webcam #33: Tested on Dell Latitude 7440 with kernel Linux 6.4.8-arch1-1 x86_64 running EndeavourOS

MrTrukiny, 2023-08-08:
I was almost to write you to celebrate that I tested it on my machine and it worked successfully (three days ago)!

However, I tried to jump into a meeting today, and the camera didn't show an image.
[image]

It says, "Camera muted by the system," but I haven't changed anything.
[image]

This is what it shows when I run the _./test.sh_

➜  archlinux-ipu6-webcam git:(main) ./test.sh 
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.23] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.23] CamHAL[ERR] Parse AE gain range failed
[08-08 09:44:32.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.24] CamHAL[ERR] Parse AE eExposure time range failed
[08-08 09:44:32.24] CamHAL[ERR] Malformed ET range in exposure time range configuration
[08-08 09:44:32.24] CamHAL[ERR] Parse AE gain range failed
Setting pipeline to PAUSED ...
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov13b10.aiqb
[08-08 09:44:32.287] CamHAL[ERR] there is no aiqb file:ov13b10
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov13b10.aiqb
[08-08 09:44:32.287] CamHAL[ERR] there is no aiqb file:ov13b10
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov8856.aiqb
[08-08 09:44:32.287] CamHAL[ERR] there is no aiqb file:ov8856
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov8856.aiqb
[08-08 09:44:32.287] CamHAL[ERR] there is no aiqb file:ov8856
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[08-08 09:44:32.287] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[08-08 09:44:32.288] CamHAL[INF] aiqb file name ov01a1s.aiqb
[08-08 09:44:32.289] CamHAL[INF] aiqb file name ov01a1s.aiqb
[08-08 09:44:32.289] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
[08-08 09:44:32.291] CamHAL[ERR] MediaControl init failed
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...

I have already tried uninstalling, rebooting, installing, and rebooting again.
If you could give me some help here, I'll be glad. 
This is the work laptop, and I really don't want to use Windows.

BTW, thank you for this fantastic job!

striczkof, 2023-08-12:
perhaps related to the new commits Intel has pushed today. See #33 as well. I'm checking on it when I wake up lol

striczkof, 2023-08-13:
Okay so welcome to the club of #21 where I was a former member but somehow got fixed on a fresh install yet others have tried running the script on fresh installs but didn't work at all.

striczkof, 2023-08-13:
Okay so welcome to the club of https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/21 where I was a former member but somehow got fixed on a fresh install yet others have tried running the script on fresh installs but didn't work at all.

@MrTrukiny would it be possible for you to tell me anything system changing/updates you have done in between the instances of camera working and giving that error?

striczkof, 2023-08-15:
Ouch, I can't really think of any of any packages that can break anything here. The main issue is your IPU6 firmware is probably failing to start (go check `dmesg | grep IPU`) and it is getting another device instead of your actual device (ov13b10 is probably the IPU6 default device). Give me 5-6 hours then I'm making another PR to add in the fix for the HAL. Maybe you should try that.

MrTrukiny, 2023-08-15:
Yeah, I think your PR will do some magic! Let me know, please.

striczkof, 2023-08-16:
@MrTrukiny just submitted a PR with the fixed binary PKGBUILD. go try it out and see how it goes!

MrTrukiny, 2023-08-16:
@striczkof I tried to run it, but it failed. I left a comment in the PR. Thank you!

striczkof, 2023-08-16:
ye there was a random a in pkgbuild lol

can you send me output of `sudo dmesg | grep ipu6` if it still fails to start?

striczkof, 2023-08-17:
I THINK I JUST REPLICATED IT FUCK YEAH

MrTrukiny, 2023-08-19:
Alright, sir
I followed #21, and I'm sharing similar results.

<details><summary>sudo dmesg | grep ipu6</summary>
<p>

[    3.576270] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    3.576489] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[    3.576522] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.576524] intel-ipu6 0000:00:05.0: mapped as: 0x0000000003c8b3fd
[    3.576553] intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.576555] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.576556] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[    3.579449] intel-ipu6 0000:00:05.0: IPC reset done
[    3.579451] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[    3.582633] intel-ipu6 0000:00:05.0: FW version: 20230102
[    3.589188] intel-ipu6 0000:00:05.0: Found supported sensor OVTI02C1:00
[    3.589355] intel-ipu6 0000:00:05.0: Connected 1 cameras
[    3.590334] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[    3.605384] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[    3.674476] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[    3.674536] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[    3.798608] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[    3.800889] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
[    5.363872] intel-ipu6-isys intel-ipu6-isys0: isys port open ready failed -16
[    5.363876] intel-ipu6-isys intel-ipu6-isys0: Device close failure: -16
[    6.419123] intel-ipu6-isys intel-ipu6-isys0: isys port open ready failed -16
[    6.419131] intel-ipu6-isys intel-ipu6-isys0: Device close failure: -16
[    7.488382] intel-ipu6-isys intel-ipu6-isys0: isys port open ready failed -16
[    7.488387] intel-ipu6-isys intel-ipu6-isys0: Device close failure: -16
[   15.180057] intel-ipu6-isys intel-ipu6-isys0: bind ov02c10 1-0036 nlanes is 2 port is 1

<p>
</details> 

<details><summary>./test.sh</summary>
<p>
./test.sh 
WARNING: erroneous pipeline: no element "icamerasrc"
</p>
</details> 

<details><summary>v4l2-ctl --list-devices</summary>
<p>

ipu6 (PCI:pci:pci0000:00):
	/dev/video1
	/dev/video2
	/dev/video3
	/dev/video4
	/dev/video5
	/dev/video6
	/dev/video7
	/dev/video8

ipu6 (pci:pci0000:00):
	/dev/media0

Virtual Camera (platform:v4l2loopback-000):
	/dev/video0

</p>
</details> 

Main points and differences:
Last time, when the camera just broke up, it appeared as "Virtual Camera" but didn't show anything (black screen).
Right now, It seems the system is not recognizing the camera, as you can see in the results of "./test.sh".
Still, `v4l2-ctl --list-devices`, shows the ipu6 devices, which didn't appear before.

I also think the problem is related to updating kernels and/or headers.

I'm going to rerun a clean install, following these steps:
- Live boot USB,
- Dual boot with Windows.
- / ext4 for root, swap, and EFI shared with Windows.
- Install your PR and test in Chrome. Check camera is working.
- Run an `eos-update .-yay.` Recheck the camera.

I'll share my results after this.
PD: Do we still need the `-as` flags to include the hibernate and Firefox issues?

striczkof, 2023-08-19:
not really, also firefox now supports nv12 so workaround for thatr is no longer needed.

I managed to pinpoint it I think. Literally reinstalling/upgrading linux and linux-headers through `pacman -S linux linux-headers` is enough to permanently break all camera stack installations from now on for that particular install. This only happens on EndeavourOS and does not (normally) happen in a relatively vanilla Manjaro Arch Linux install like mine

What I can try later is to setup systemd path watchers on ALL IPU6 related files and see which ones get touched. 

Your contributions will be very much appreciated, this has been a long standing bug xd

MrTrukiny, 2023-08-19:
Bro,

I just did a clean installation, and after running the ./test.sh command, it shows this:

WARNING: erroneous pipeline: no element "icamerasrc"

And camera it's not working, neither Firefox nor Chrome. :(

striczkof, 2023-08-19:
something failed with icamerasrc?

god, if intel breaks icamerasrc again

striczkof, 2023-08-19:
@MrTrukiny can you try running the install script again, but save the output in a file and paste here?

Do this: `./install.sh 2>&1 | tee -a install.log`

MrTrukiny, 2023-09-02:
Ok, there are the steps I followed:

- I installed icamerasrc-git from AUR. Reboot.
- I installed the script from a fresh clone. Reboot.
- I tested with the ./test.sh command and in Firefox. It worked!
- I reboot and tested again. Continues working.

<details><summary>Here's the install.log</summary>
<p>

Workaround for other applications will be installed.
Hibernation workaround will be installed.
# Install headers for: linux
# Install package from the AUR/repos: linux-headers
Sync Explicit (1): linux-headers-6.4.12.arch1-1
warning: linux-headers-6.4.12.arch1-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: base-devel
Sync Explicit (1): base-devel-1-1
warning: base-devel-1-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: intel-ipu6-dkms-git
 -> intel-ipu6-dkms-git-r135.5dc00ed2a-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Build and install package: intel-ipu6ep-camera-bin
~/Linux/archlinux-ipu6-webcam/intel-ipu6ep-camera-bin ~/Linux/archlinux-ipu6-webcam
==> Making package: ipu6-camera-bin r70.c4f9e52-1 (sáb 02 sep 2023 17:00:12)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning ipu6-camera-bins git repo...
Cloning into bare repository '/home/mrtrukiny/Linux/archlinux-ipu6-webcam/intel-ipu6ep-camera-bin/ipu6-camera-bins'...
==> Validating source files with sha256sums...
    ipu6-camera-bins ... Skipped
==> Extracting sources...
  -> Creating working copy of ipu6-camera-bins git repo...
Cloning into 'ipu6-camera-bins'...
done.
==> Starting pkgver()...
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Stripping unneeded symbols from binaries and libraries...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "intel-ipu6ep-camera-bin-fix"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: ipu6-camera-bin r70.c4f9e52-1 (sáb 02 sep 2023 17:08:58)
==> Installing package intel-ipu6ep-camera-bin-fix with pacman -U...
warning: intel-ipu6ep-camera-bin-fix-r70.c4f9e52-1 is up to date -- skipping
loading packages...
 there is nothing to do
~/Linux/archlinux-ipu6-webcam
=> SUCCESS
# Build and install package: intel-ipu6ep-camera-hal-git
~/Linux/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git ~/Linux/archlinux-ipu6-webcam
==> Making package: intel-ipu6ep-camera-hal-git-fix r56.884b81a-1 (sáb 02 sep 2023 17:09:25)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
[... 523 lines cut ...]
=> SUCCESS
# Install package from the AUR/repos: gst-plugin-pipewire
Sync Explicit (1): gst-plugin-pipewire-1:0.3.79-1
warning: gst-plugin-pipewire-1:0.3.79-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: gst-plugins-good
Sync Dependency (1): gst-plugins-good-1.22.5-2
warning: gst-plugins-good-1.22.5-2 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Enable: v4l2-relayd.service
=> SUCCESS
# Start: v4l2-relayd.service
=> SUCCESS

All done.
Remember to reboot upon succesful installation!
Reboot now? [y/N] # Rebooting in 5 seconds...

</p>
</details> 

I hope it helps for future problems, if any.
Thank you again.

striczkof, 2023-09-03:
No worries. Just be warned that it's gonna break the moment linux package is updated in EOS. Yes, `pacman -S linux` is literally enough to break the entire stack.

I'm currently investigating `kernel-install` since it might be the culprit (vanilla Arch does not do that by default) since it touches the dkms modules. Here's the install log for reference.

<details><summary>sudo pacman -S linux # ran in EndeavourOS</summary>
<p>

```
warning: linux-6.4.12.arch1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Package (1)  Old Version     New Version     Net Change

core/linux   6.4.12.arch1-1  6.4.12.arch1-1    0.00 MiB

Total Installed Size:  125.56 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] 
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
:: Processing package changes...
reinstalling linux...
:: Running post-transaction hooks...
(1/5) Arming ConditionNeedsUpdate...
(2/5) Updating module dependencies...
(3/5) Running kernel-install...
:: kernel-install installing kernel 6.4.12-arch1-1
dracut: Executing: /usr/bin/dracut --no-hostonly --force /efi/0dc24d105e0c43d0a6a4e27b94fac3a4/6.4.12-arch1-1/initrd-fallback 6.4.12-arch1-1
dracut: dracut module 'dash' will not be installed, because command 'dash' could not be found!
dracut: dracut module 'mksh' will not be installed, because command 'mksh' could not be found!
dracut: dracut module 'busybox' will not be installed, because command 'busybox' could not be found!
dracut: dracut module 'dbus-broker' will not be installed, because command 'dbus-broker' could not be found!
dracut: dracut module 'rngd' will not be installed, because command 'rngd' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmand' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmanctl' could not be found!
dracut: dracut module 'connman' will not be installed, because command 'connmand-wait-online' could not be found!
dracut: dracut module 'network-wicked' will not be installed, because command 'wicked' could not be found!
dracut: dracut module 'dmraid' will not be installed, because command 'kpartx' could not be found!
dracut: dracut module 'multipath' will not be installed, because command 'multipath' could not be found!
dracut: dracut module 'tpm2-tss' will not be installed, because command 'tpm2' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'dcbtool' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fipvlan' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'lldpad' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fcoemon' could not be found!
dracut: dracut module 'fcoe' will not be installed, because command 'fcoeadm' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'dcbtool' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'fipvlan' could not be found!
dracut: dracut module 'fcoe-uefi' will not be installed, because command 'lldpad' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsi-iname' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsiadm' could not be found!
dracut: dracut module 'iscsi' will not be installed, because command 'iscsid' could not be found!
dracut: dracut module 'nbd' will not be installed, because command 'nbd-client' could not be found!
dracut: dracut module 'nvmf' will not be installed, because command 'nvme' could not be found!
dracut: dracut module 'biosdevname' will not be installed, because command 'biosdevname' could not be found!
dracut: dracut module 'memstrack' will not be installed, because command 'memstrack' could not be found!
dracut: memstrack is not available
dracut: If you need to use rd.memdebug>=4, please install memstrack and procps-ng
dracut: dracut module 'squash' will not be installed, because command 'mksquashfs' could not be found!
dracut: dracut module 'squash' will not be installed, because command 'unsquashfs' could not be found!
dracut: *** Including module: systemd ***
dracut: *** Including module: systemd-initrd ***
dracut: *** Including module: modsign ***
dracut: *** Including module: i18n ***
dracut: *** Including module: btrfs ***
dracut: *** Including module: crypt ***
dracut: *** Including module: dm ***
[... 127 lines cut ...]
dracut: *** Resolving executable dependencies ***
dracut: *** Resolving executable dependencies done ***
dracut: *** Hardlinking files ***
dracut: Mode:                     real
dracut: Method:                   sha256
dracut: Files:                    774
dracut: Linked:                   1 files
dracut: Compared:                 0 xattrs
dracut: Compared:                 34 files
dracut: Saved:                    690 B
dracut: Duration:                 0.001782 seconds
dracut: *** Hardlinking files done ***
dracut: *** Generating early-microcode cpio image ***
dracut: *** Using microcode found in '/boot/intel-ucode.img' ***
dracut: *** Store current command line parameters ***
dracut: *** Stripping files ***
dracut: *** Stripping files done ***
dracut: *** Creating image file '/efi/0dc24d105e0c43d0a6a4e27b94fac3a4/6.4.12-arch1-1/initrd' ***
dracut: *** Creating initramfs image file '/efi/0dc24d105e0c43d0a6a4e27b94fac3a4/6.4.12-arch1-1/initrd' done ***
(4/5) Check if user should be informed about rebooting after certain system package upgrades.
(5/5) Checking which packages need to be rebuilt
```

</p>
</details>

MrTrukiny, 2023-09-12:
Yes, I updated the system, and It is broken now.
I'm not using Endeavour as my main OS; I have dual-booted installed to run some tests related to this camera issue.

Let me know if you need something from my side.
Looking forward to seeing a solution for this!
Thank you for your effort.

clintlombard, 2023-10-14:
I'm on Endeavour OS as well and managed to find myself in the same situation as @MrTrukiny. webcam worked like a charm after installing on a fresh OS, but broke after a kernel update. Also only the v4l2loopback device was showing up (like in #9).

For reference my setup is as follows:

* Endeavour OS
* Dell XPS plus 13" 9320
* Kernel`linux-lts` (6.1.57-1-lts)
* drive is encrypted with LUKS.

I've managed to get things working again. The issue as @striczkof suspected is in dracut. I added my own dracut config to `/etc/dracut.conf/webcam.conf`:

```
modules+=" snd_soc_rt715_sdca snd_soc_rt1316_sdw snd_sof_pci_intel_tgl snd_hda_intel snd_soc_sof_sdw i915 "
install_items+=" /lib/firmware/intel/sof/sof-adl.ri /lib/firmware/intel/sof-tplg/sof-adl-rt1316-l12-rt714-l0.tplg /lib/firmware/vsc/soc_a1_prod/ivsc_fw_a1_prod.bin /lib/firmware/vsc/soc_a1_prod/ivsc_pkg_ovti01a0_0_a1_prod.bin /lib/firmware/vsc/soc_a1_prod/ivsc_skucfg_ovti01a0_0_1_a1_prod.bin /lib/firmware/intel/ipu6ep_fw.bin "
```

Most of the lines come from the mkinitcpio.conf shown in the [Arch 9320 guide](https://wiki.archlinux.org/title/Dell_XPS_13_Plus_\(9320\)). The critical fixes for me were adding these files:

*  /lib/firmware/vsc/soc_a1_prod/ivsc_fw_a1_prod.bin
* /lib/firmware/vsc/soc_a1_prod/ivsc_pkg_ovti01a0_0_a1_prod.bin
* /lib/firmware/vsc/soc_a1_prod/ivsc_skucfg_ovti01a0_0_1_a1_prod.bin
* /lib/firmware/intel/ipu6ep_fw.bin

I worked them out by seeing which files were being complained about in dmesg after boot.

v4l2-relay is now broken for me, but I can get the webcam to work in Firefox but just manually running:
``` shell
sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov01a10-uf ! video/x-raw,format=NV12,width=1280,height=720 ! v4l2sink device=/dev/video0
```

Hope this helps someone else.

striczkof, 2023-10-21:
so they have to be included in the initramfs. I literally have no idea how and why. I will add a dracut workaround for thank. Thank you so much @clintlombard !

striczkof, 2023-10-25:
@clintlombard I reckon it would be much better if we just drop the IPU and IVSC drivers from the dracut initramfs as it is not even included by mkinitcpio and the webcam stack loads late just fine. What do you think?

@MrTrukiny can you add this to your dracut config then either reinstall linux kernel to trigger dracut or do it manually?

```sh
sudo sh -c 'echo -e 'omit_drivers+=\" intel_ipu6 intel_ipu6_isys intel_ipu6_psys intel_vsc mei_vsc v4l2loopback \" > /etc/dracut.conf.d/ipu6.conf'
```

I would be so happy if this works for you and everyone else with dracut!

clintlombard, 2023-10-25:
@striczkof adding this worked perfectly on my side.

I also managed to get v4l2-relay working by reinstalling icamerasrc-git.

striczkof, 2023-10-25:
The v4l2loopback trying to load in the initramfs like the ipu6 might have caused it to break too, but that's just my hypothesis.

Current icamerasrc git repo is working fine but my blood is starting to boil over Intel breaking it and other packages every other commit. I will make commit-pinned PKGBUILDs that should almost always work for everyone.

fcapano, 2023-11-16:
@striczkof your fix also worked for me on 6.6.1-zen1-1-zen

striczkof, 2023-11-16:
> @striczkof your fix also worked for me on 6.6.1-zen1-1-zen

hey thanks for letting me know that Intel merged that PR!

closed 2024-10-25 by stefanpartheym