stefanpartheym/archlinux-ipu6-webcam #34: Intel have made changes in ipu6-camera-bin and ipu6-camera-hal breaking the build

aforembs, 2023-08-10:
See: [https://github.com/intel/ipu6-camera-hal/issues/67](url)

striczkof, 2023-08-12:
Ouch.

I'm making a PKGBUILD that builds up to this commit: ea976df68035252e1cb684361f241557bd4ec376 when I wake up

striczkof, 2023-08-12:
no need to use older builds, but need to make a PKDBUILD that implements the fixes mentioned in the issue. doing that

sorry was sleepy when I first read lel

striczkof, 2023-08-13:
Er that took more than 5 hours, was a bit distracted lol.

I'm kinda tired, I'm gonna take a break and play some CoD then either submit the PR if I can stay awake long enough or sleep and do it later. For now, here is the fix PKGBUILD, with workaround stolen from your comment and @hftsai256 at https://github.com/intel/ipu6-camera-hal/issues/67#issuecomment-1675109826 :
```shell
# Fixed PKGBUILD from https://github.com/stefanpartheym/archlinux-ipu6-webcam
# Fixed paths and added symlinks to fix the build
# Original PKGBUILD: https://aur.archlinux.org/packages/intel-ipu6ep-camera-bin
pkgbase=ipu6-camera-bin
pkgname=intel-ipu6ep-camera-bin-fix
_pkgname=ipu6-camera-bins
pkgver=r70.c4f9e52
pkgrel=1
pkgdesc="Intel IPU6 camera binaries (Alder Lake)"
arch=('x86_64')
url="https://github.com/intel/ipu6-camera-bins"
license=('unknown')
depends=('intel-ipu6-dkms-git' 'glibc')
makedepends=('git')
provides=(ipu6-camera-bin intel-ipu6ep-camera-bin)
conflicts=(intel-ipu6-camera-bin intel-ipu6ep-camera-bin)
source=("git+${url}.git")
sha256sums=('SKIP')

pkgver() {
    cd $_pkgname
    printf "r%s.%s" "$(git rev-list --count HEAD)" "$(git rev-parse --short HEAD)"
}

package() {
    install -dm755 "$pkgdir/usr"
    cp -rT "$_pkgname/lib" "$pkgdir/usr/lib"
    cp -rT "$_pkgname/lib/ipu_adl/pkgconfig" "$pkgdir/usr/lib/pkgconfig"
    cp -rT "$_pkgname/include" "$pkgdir/usr/include"
    # A hack, need to be removed when Intel fixes this shit
    ln -s /usr/lib/ipu_adl/{libia_cca.so,libia_log.so,libgcss.so.0} $pkgdir/usr/lib/
}

```

Also figured out that the latest icamerasrc-git now works and therefore we can use the PKGBUILD from the AUR direct.

@stefanpartheym @sigboe should we make a single PKGBUILD putting together all Intel sourced packages into a single neat packages? Intel packages breaking arbitrarily due to incomplete commits is no longer funny and then fixing themselves occasionally only to break again is no longer funny.

lukemarsden, 2023-08-14:
thank you for working on this! :raised_hands:

sigboe, 2023-08-15:
@striczkof I tried this PKGBUILD file, and updated the install.sh file to use it, and it works good!
Are you going to post a Pull Request?

striczkof, 2023-08-15:
I forgor 💀

Doing this after going back from uni in 5h. I want to do some cleanup too.

PR #35 (2023-08-27): Added IPU6 binary fix and some cleanup
Added IPU6 camera binary fix PKGBUILD, also did some cleanup on the install script. Fixes #34 

Adding in more clean up in a bit
closed 2023-08-27 by stefanpartheym