stefanpartheym/archlinux-ipu6-webcam #37: Dell Latitude 7320 - ipu6-camera-bins fail due to dir structure (i5 11th gen Tiger Lake)

usag1r, 2023-08-21:
After successfully installing majority of the components, I received following error. This attempt is on fresh EndeavourOS. I've also built on Arch Linux and had similar issue which I've fixed by changing directory structure but not sure if it's causing other issues when I do that because I couldn't get it to start on Arch yet, was hoping this would work..

On Arch, I managed to get Virtual Camera with V4l2 but it's just black image output so far.
Here is the error I got with install.sh of your solution. Thank you so much btw. This is great service to the Linux world. Unbelievable how inconsiderate Intel is moving with IPU6 devices.

```

Cloning into 'ipu6-camera-bins'...
done.
==> Starting pkgver()...
==> Updated version: ipu6-camera-bin r70.c4f9e52-1
==> Sources are ready.
==> Making package: ipu6-camera-bin r70.c4f9e52-1 (Mon 21 Aug 2023 08:37:58 PM BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Starting pkgver()...
==> Removing existing $pkgdir/ directory...
==> Entering fakeroot environment...
==> Starting package_intel-ipu6-camera-bin()...
cp: cannot stat 'ipu6-camera-bins/ipu6/lib': No such file or directory
==> ERROR: A failure occurred in package_intel-ipu6-camera-bin().
    Aborting...
 -> error making: ipu6-camera-bin-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
intel-ipu6ep-camera-bin - exit status 4
ERROR:  Failed to install: intel-ipu6ep-camera-bin

```

twouters, 2023-08-29:
The AUR packages ([intel-ipu6-camera-bin](https://aur.archlinux.org/packages/intel-ipu6-camera-bin) and [intel-ipu6ep-camera-hal-git](https://aur.archlinux.org/packages/intel-ipu6ep-camera-hal-git)) have been updated to work with the new structure.

twouters, 2023-08-29:
Oh, this has probably already been fixed by #35

usag1r, 2023-08-29:
I just wiped everything, tried again and managed to get the install script to work all the way to then without any issues. Rebooted, cheese and mpv still don't get a healthy stream from internal cameras. I was wondering, the package installs: 
- intel-ipu6ep-camera-bin-fix
- intel-ipu6ep-camera-hal-git-fix

From the official intel repos, I remember ipu6**ep** versions were for the Alder Lake chips. Maybe that's where the issue is as I'm on Tiger Lake processor.
Also the confirmed working case for Dell Latitude 7440 seems to be an Alder Lake chip, so this makes me think that's where the issue is at this stage.
It'd be great to get it working somehow!

striczkof, 2023-08-30:
Yup, this script is tailored for Adler Lake as you said but the Intel drivers/HAL allow building for IPU6 from Tiger Lake to Meteor Lake so I will try making Tiger Lake and Meteor Lake work this weekend along with other fixes

usag1r, 2023-08-30:
Thank you @striczkof . I can't wait. Thank you so much for your efforts. I'll close this as the original directory structure issue is fixed already as @twouters pointed out.

closed 2023-08-30 by usag1r

usag1r, 2023-09-28:
Is there any progress or timeline on the Tiger Lake front?

striczkof, 2023-09-28:
oh ye sorry been busy with uni stuff and shit thanks for reminding me

it should be just as easy as refactoring the script to attempt to autodetect the appropriate version by current intel CPU's generation then build then install them. I just woke up and I have nothing to do so I will work on just that.

usag1r, 2023-09-28:
Hi! 
No worries at all and thank you so much for the insight as well as potential progress and your efforts! 🙏

striczkof, 2023-09-30:
> Hi! 
> No worries at all and thank you so much for the insight as well as potential progress and your efforts! 🙏

@usag1r feel free to check #42. I just have to finish and push the hal package when I wake up tomorrow. Sorry been busy lol

striczkof, 2023-10-01:
@usag1r just wondering, what does `cat /proc/cpuinfo | grep 11th` shows up to you? I am trying to make the script autodetect the appropriate generation.

usag1r, 2023-10-01:
Hi @striczkof, here is what I get:

```
$ cat /proc/cpuinfo | grep 11th
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
model name	: 11th Gen Intel(R) Core(TM) i5-1140G7 @ 1.10GHz
```

striczkof, 2023-10-01:
awesome mate cheers!

I just realised that doing cat `/sys/devices/cpu{,_core,_atom}/caps/pmu_name` is much easier. @usag1r what does it show for you? and sorry for asking again lol. I really just want to make it less pain in the arse for everyone as possible.

usag1r, 2023-10-01:
Hi, No worries at all! Let me know if anything else comes up.

I was able to do 

```
cat /sys/devices/cpu/caps/pmu_name
icelake

```
icelake is the only output. :+1:

striczkof, 2023-10-01:
What the fuck.

You are totally using a Tiger Lake chip (https://www.intel.com/content/www/us/en/products/sku/208659/intel-core-i51140g7-processor-8m-cache-up-to-4-20-ghz-with-ipu/specifications.html) yet it says Ice Lake?

Cheers for the help, now found out that the earlier method is way reliable than this. Expect a work PR branch in 1 hour

usag1r, 2023-10-01:
Haha, I know. It seems weird. Maybe Intel based Tiger Lake on Icelake architecture since they are consecutive or something like that. Looking forward to testing the new PR! You are most welcome. Thank you also

striczkof, 2023-10-01:
cant make it. intel broke something either in icamerasrc or in the HAL and I'm way too tired to figure it out.

Basically the icamerasrc gst plugin keeps getting blacklisted, something like "can't open sem lock error permission denied" so i've been digging through the source code. job for tomorrow ig

striczkof, 2023-10-01:
@usag1r would it be possible for you to try it right now? I managed to get it to the state of being able to easily install and uninstall. It should be able to detect your tiger lake then install only ipu6 stuff.
Just do:
```sh
git clone --branch fix_and_rework https://github.com/striczkof/archlinux-ipu6-webcam.git
cd archlinux-ipu6-webcam
./install.sh
```
then restart just to be sure

Fuck me it's 12:19am already. I will fix whatever comes up tomorrow

usag1r, 2023-10-01:
Absolutely! Testing asap. I'll keep you posted.

usag1r, 2023-10-01:
```
$./install.sh 
# Detected Tiger Lake (11th generation).
# Install headers for: linux
# Install package from the AUR/repos: linux-headers
warning: linux-headers-6.5.4.arch2-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Installing dependencies.
# Install package from the AUR/repos: base-devel
warning: base-devel-1-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: gst-plugin-pipewire
warning: gst-plugin-pipewire-1:0.3.80-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: gst-plugins-good
warning: gst-plugins-good-1.22.6-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Installing the camera stack.
rm: cannot remove 'ipu6-pkgs.txt': No such file or directory
```

I guess, extra line or check fail?

usag1r, 2023-10-01:
```
uninstall.sh:INSTALLED_PKG_LIST=ipu6-pkgs.txt
uninstall.sh:for pkg in $(tac "${INSTALLED_PKG_LIST}"); do
install.sh:INSTALLED_PKG_LIST=ipu6-pkgs.txt
install.sh:# These packages will be written down to $INSTALLED_PKG_LIST which will be then used by uninstall.sh
install.sh:rm "${INSTALLED_PKG_LIST}"
install.sh:  echo "${pkg}" >> "${INSTALLED_PKG_LIST}"
```

Must be related to one of these.

usag1r, 2023-10-01:
I commented out 

`rm "${INSTALLED_PKG_LIST}"`

from install.sh and the script is continuing right now. I'll write the outcomes here later. Thank you so much for your effort and have a nice day at work tomorrow!

usag1r, 2023-10-01:
I just rebooted and checked via 3 methods but they are all showing black screen.

Methods tried for testing:

1. mpv av://v4l2:/dev/video0
2. online test via firefox
3. cheese

[screenshot](https://pasteboard.co/cLDPIKu9h0eq.png)

striczkof, 2023-10-01:
rip.

Can you try rebuilding the reinstalling icamerasrc-git using `paru -S --rebuild icamerasrc-git` or your equivalent in your helper (should be no need to reboot after but just in case)?

Then, I'd love to see what you get out of these:
`./test.sh` (try running the test script first)
`sudo dmesg | grep ipu`
`systemctl status v4l2-relayd`
`v4l2-ctl --list-devices`
`gst-inspect-1.0 icamerasrc` (if this is empty should be blacklisted still)
`gst-inspect 1.0` (first few lines show blacklist reasons, last line/s show why)

usag1r, 2023-10-02:
Hi! Sure.

- Just rebuilt **icamerasrc** using paru and rebooted just in case.

```$ ./test.sh
WARNING: erroneous pipeline: no element "icamerasrc"
```

Although `pacman -Q | grep cam` shows `icamerasrc-git r65.931412a-1` installed...

```
$ sudo dmesg | grep ipu
[    3.093690] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    3.093960] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    3.093980] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.093982] intel-ipu6 0000:00:05.0: mapped as: 0x000000007b5ed7c4
[    3.115910] intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.115915] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.115917] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.128704] intel-ipu6 0000:00:05.0: IPC reset done
[    3.128712] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    3.135902] intel-ipu6 0000:00:05.0: FW version: 20210414
[    3.148249] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    3.148269] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.148271] intel-ipu6 0000:00:05.0: mapped as: 0x00000000bd45f556
[    3.148559] intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.148562] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.148564] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.150693] intel-ipu6 0000:00:05.0: IPC reset done
[    3.150695] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    3.153638] intel-ipu6 0000:00:05.0: FW version: 20210414
[    3.156482] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    3.156500] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.156501] intel-ipu6 0000:00:05.0: mapped as: 0x000000007b5ed7c4
[    3.157755] intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.157758] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.157759] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.159611] intel-ipu6 0000:00:05.0: IPC reset done
[    3.159613] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    3.162435] intel-ipu6 0000:00:05.0: FW version: 20210414
[    3.204032] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    3.204049] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.204050] intel-ipu6 0000:00:05.0: mapped as: 0x000000007b5ed7c4
[    3.204126] intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.204127] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.204128] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.205982] intel-ipu6 0000:00:05.0: IPC reset done
[    3.205984] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    3.208716] intel-ipu6 0000:00:05.0: FW version: 20210414
[    3.267824] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    3.267837] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.267839] intel-ipu6 0000:00:05.0: mapped as: 0x000000007b5ed7c4
[    3.269651] intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.269654] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.269656] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.271593] intel-ipu6 0000:00:05.0: IPC reset done
[    3.271596] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    3.274889] intel-ipu6 0000:00:05.0: FW version: 20210414
[    3.278196] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    3.278211] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.278213] intel-ipu6 0000:00:05.0: mapped as: 0x000000007b5ed7c4
[    3.278287] intel-ipu6 0000:00:05.0: IPU in secure mode
[    3.278288] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    3.278290] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    3.280758] intel-ipu6 0000:00:05.0: IPC reset done
[    3.280761] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    3.284028] intel-ipu6 0000:00:05.0: FW version: 20210414
[    3.285022] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    3.285035] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    3.285037] intel-ipu6 0000:00:05.0: mapped as: 0x000000007b5ed7c4
[... 516 lines cut ...]
$  v4l2-ctl --list-devices
Virtual Camera (platform:v4l2loopback-000):
	/dev/video0

```

```
$ gst-inspect-1.0 icamerasrc
No such element or plugin 'icamerasrc'
```

```
...
x265:  x265enc: x265enc
ximagesink:  ximagesink: Video sink
ximagesrc:  ximagesrc: Ximage video source
xingmux:  xingmux: MP3 Xing muxer
xvimagesink:  xvimagesink: Video sink
y4mdec:  y4mdec: YUV4MPEG demuxer/decoder
y4menc:  y4menc: YUV4MPEG video encoder
zbar:  zbar: Barcode detector
zxing:  zxing: Barcode detector

Total count: 249 plugins (1 blacklist entry not shown), 1047 features
```

striczkof, 2023-10-02:
Oh yeah right, I just got that piece of shit fixed by symlinking the missing files from /usr/lib/ipu_* to /usr/lib.

Delete your current repo then reclone it again. run ./uninstall.sh then /install.sh. These will uninstall then reinstall everything. I managed to fix the hiccups earlier.

Love you man for testing xd

usag1r, 2023-10-02:
No worries! Absolutely love working with you on this.

deleted everything, rebooted, recloned. Building right now. I'll report in shortly.

usag1r, 2023-10-02:
So I got these. Issue with icamerasrc pops up again.

```
$ ./test.sh 

(gst-plugin-scanner:4851): GStreamer-WARNING **: 04:20:42.787: Failed to load plugin '/usr/lib/gstreamer-1.0/libgsticamerasrc.so': libia_cca.so: cannot open shared object file: No such file or directory
WARNING: erroneous pipeline: no element "icamerasrc"
```

```
$ sudo pacman -Q | grep cam

icamerasrc-git r65.931412a-1
intel-ipu6-camera-bin-fix r72.ff21b55-1
intel-ipu6-camera-hal-git-fix r93.9fa05a9-1
libcamera 0.1.0-2
libcamera-ipa 0.1.0-2
```

striczkof, 2023-10-02:
oh ye right it's still blacklisted. get rid of the blacklist by `rm ~/.cache/gstreamer-1.0/registry.x86_64.bin`

usag1r, 2023-10-02:
I removed it but it's immediately created again after running ./test.sh

```
$ ./test.sh 

(gst-plugin-scanner:4894): GStreamer-WARNING **: 04:30:32.497: Failed to load plugin '/usr/lib/gstreamer-1.0/libgsticamerasrc.so': libia_cca.so: cannot open shared object file: No such file or directory
WARNING: erroneous pipeline: no element "icamerasrc"
```

striczkof, 2023-10-02:
> I removed it but it's immediately created again after running ./test.sh
>
> ```
> $ ./test.sh
>
> (gst-plugin-scanner:4894): GStreamer-WARNING **: 04:30:32.497: Failed to load plugin [...]

Why is it not detecting/installing? try this.

`sudo ln -sf /usr/lib/ipu_tgl/{libia_cca.so,libia_log.so,libgcss.so.0} "/usr/lib/"`

Fuck, that's why, it's using ipu_adl

usag1r, 2023-10-02:
Wow, yes! ./test.sh worked for the first time. Here's the output:

```
$ ./test.sh 

Setting pipeline to PAUSED ...
[10-02 04:41:28.653] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 04:41:28.653] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 04:41:28.655] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.655] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-02 04:41:28.655] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-02 04:41:28.656] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.656] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 04:41:28.656] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 04:41:28.657] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.657] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.657] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.657] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.657] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.658] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.658] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.658] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 04:41:28.658] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.658] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.658] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.658] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.658] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.658] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 04:41:28.658] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.658] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.658] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.658] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.658] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 04:41:28.659] CamHAL[ERR] invalid media format, default value used.
[10-02 04:41:28.659] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-02 04:41:28.659] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-02 04:41:28.659] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-02 04:41:28.659] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: hm11b1 , devname: /dev/v4l-subdev1
[10-02 04:41:28.662] CamHAL[ERR] MediaControl init failed
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
[10-02 04:41:28.750] CamHAL[WAR] Seems camera device 0 is not closed properly (pid 5586).
```

usag1r, 2023-10-02:
If I run anything that tries to access the camera cache registry (blacklist?) appears again. I can then remove it again and run the test and it gives the same output...

striczkof, 2023-10-02:
Awesome mate. Now I gotta figure out how I broke the relax then fix the camera in apps.

> If I run anything that tries to access the camera cache registry (blacklist?) appears again. I can then remove it again and run the test and it gives the same output...

Oh no need. It won't get blacklisted anymore. You only remove it whenever it gets blacklisted but right now should be never.

usag1r, 2023-10-02:
Awesome, thank you!

I mentioned blacklist because if I run for example cheese or sth like that...

```
$ cheese 

[10-02 04:53:11.675] CamHAL[ERR] failed to open sem lock, errno: Permission denied
[10-02 04:53:11.889] CamHAL[ERR] failed to open sem lock, errno: Permission denied
```

Then icamerasrc is inaccessible again...

```
$ ./test.sh 

WARNING: erroneous pipeline: no element "icamerasrc"
```

Then if I remove the registry, I can run the test script again.

striczkof, 2023-10-02:
Wait, when you clear the blacklist registry, does `./test.sh` even work (show your video camera) or it just outputs that long list of accessed files as you have shown earlier?

usag1r, 2023-10-02:
I meant it just runs and gives the same output. Can't access the camera at all. If I try to get image from the camera by any app, then registry is recreated and can't even run the test.sh as in the recent screenshot..

striczkof, 2023-10-02:
@usag1r  I literally just found out about this. Apparently Tiger Lakes only use YUY12. So try this out and see if you get any output:
Replace `waylandsink` with `ximagesink` if you are not on Wayland.
`sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=YUY2,width=1280,height=720 ! videoconvert ! waylandsink`

usag1r, 2023-10-02:
Great discovery! but it must be something else?

Maybe something is blocking the camera or not letting it start from the start up? Btw, I have triple boot and Camera is 100% functional on Windows 11, so we're sure it's not a hardware fault.

Please see output below. I'm on xorg so replaced `waylandsink` with `ximagesink` as instructed.

```
$ echo $XDG_SESSION_TYPE
x11
```

```
$ sudo -E gst-launch-1.0 icamerasrc buffer-count=7 ! video/x-raw,format=YUY2,width=1280,height=720 ! videoconvert ! ximagesink
 
Setting pipeline to PAUSED ...
[10-02 11:26:27.175] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 11:26:27.177] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 11:26:27.178] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.178] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-02 11:26:27.179] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-02 11:26:27.179] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.179] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 11:26:27.180] CamHAL[INF] aiqb file name hm11b1.aiqb
[10-02 11:26:27.181] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.181] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.181] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.181] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.181] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.181] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.182] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.182] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.182] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.182] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.182] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.182] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.183] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.184] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.184] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.184] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[10-02 11:26:27.184] CamHAL[ERR] invalid media format, default value used.
[10-02 11:26:27.184] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-02 11:26:27.184] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-02 11:26:27.184] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-02 11:26:27.184] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: hm11b1 , devname: /dev/v4l-subdev1
[10-02 11:26:27.187] CamHAL[ERR] MediaControl init failed
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
[10-02 11:26:27.265] CamHAL[WAR] Seems camera device 0 is not closed properly (pid 5553).
```

usag1r, 2023-10-02:
And also:

```
$ sudo systemctl status | grep v4l2-relayd

           │ ├─v4l2-relayd.service
           │ │ └─1868 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! v4l2sink name=v4l2sink device=/dev/video0"
               │ ├─6122 grep --color=auto v4l2-relayd
```

striczkof, 2023-10-02:
Most likely will not fix your issue but I have just fixed the icamerasrc not displaying to application by pinning it down to a july commit. Might as well try it out by redoing the install again with a fresh clone of my branch. Also, let's move this to #44 as closed issues are kinda pain in the arse to track xd

I have also modified the ./test.sh so it tries to detect if you are using Tiger Lake so only use ./test.sh.