stefanpartheym/archlinux-ipu6-webcam #39: Webcam does not work after upgrading Linux to 6.5.2

elhossary, 2023-09-10:
Hi,

I had my webcam working on my alder lake X1 carbon for monthes but today, I upgraded to the latest linux 6.5.2.
I noticed this during the installation:
```
==> dkms install --no-depmod xmm7360-pci-git/r231.107b6ba -k 6.5.2-arch1-1
==> dkms install --no-depmod ipu6-drivers/r100.8c02a846d -k 6.5.2-arch1-1
Error! Bad return status for module build on kernel: 6.5.2-arch1-1 (x86_64)
Consult /var/lib/dkms/ipu6-drivers/r100.8c02a846d/build/make.log for more information.
==> WARNING: `dkms install --no-depmod ipu6-drivers/r100.8c02a846d -k 6.5.2-arch1-1' exited 10
==> dkms install --no-depmod v4l2loopback/r10.f94def5 -k 6.5.2-arch1-1
==> depmod 6.5.2-arch1-1
```
When I rebooted, the camera stopped working
Can anyone help me to fix it?

Thanks

ParriauxMaxime, 2023-09-10:
Hi, I got the same problem tonigh after a clean reinstall and update to kernel `6.5.2`.

First of all, thanks to contributors and maintainers of this repository :heart:

I've got a Dell XPS 13 Plus 9320, and I've been able to make my camera work tonight (after a lot of trial and errors).
My solution is sub-optimal and I hope that it may work for you too.  

In the `install.sh`, comment the installation of the following module :  `intel-ipu6ep-camera-bin`,  `intel-ipu6ep-camera-hal-git` and `icamerasrc-git`  

_install.sh_
```
#!/usr/bin/env bash

set -euo pipefail

RED='\033[0;31m'
ORANGE='\033[0;33m'
NC='\033[0m'
# Insert your AUR package manager and equivalent commands here if any.
# The script will try to use either yay or paru if not specified.
PKGMAN=()
MAKEPKG=(makepkg -si --noconfirm --needed)
# Script flags, best to leave them unless you want to 'hardcode' behaviours.
FLAG_YUY2_WA=false
FLAG_S2DISK_HACK=false
FLAG_EXPLICIT_WAYLAND=false
FLAG_REBOOT_AFTER_INSTALL=false
FLAG_QUIET_MODE=false

SUPPORTED_KERNELS=(linux linux-lts linux-zen linux-hardened)
# All packages installed, in order.
PKGS=(base-devel
      intel-ipu6-dkms-git
      # intel-ipu6ep-camera-bin
      # intel-ipu6ep-camera-hal-git
      v4l2loopback-dkms-git
      v4l2-relayd
      # icamerasrc-git # Will build from repos, old fix PKGBUILD is renamed to icamerasrc-git.old
      gst-plugin-pipewire
      gst-plugins-good
)
[...]
```

It should fail after installing  ipu6-drivers with dkms (Equivallent to [this repo](https://github.com/intel/ipu6-drivers), which is "nice".

Then, install binary from `ipu6-camera-bins` [this repo](https://github.com/intel/ipu6-camera-bins)
Moving to `ipu6-camera-hal` + the adaptor, follow [this instruction](https://github.com/intel/ipu6-camera-hal). Note that you will also need the `icamerasrc` in [this repo](https://github.com/intel/icamerasrc/tree/icamerasrc_slim_api) to operate.
**THIS MAY FAIL** because build is done without optimization flag (`-O`). Open `ipu6-camera-hal` repository in your IDE, lookup for `-Werror` references in `CMakeLists.txt` files. Prepend with the `-O` flag. Or just remove the `-Werror` :shrug: .

Last step, reexecute the inital `install.sh` from this repo and go with `v4l2*` and `gst*` related packages installation.

Reboot, try with `test.sh`, it should work, and cherry on the top, it should also work in Firefox/Chromium.

Désolé for the fuzzy explanation, I just come home from vacation, will be happy to help further later if needed :sleeping:

elhossary, 2023-09-13:
Hi @ParriauxMaxime ,
Thanks for the suggestions, it does not work for me too

iosonofabio, 2023-09-21:
FWIW, it works on my machine on `linux 6.5.3.arch1-1`.

All I did was clone the repo, `pacman -R` a few packages of the stack that were installed before, and run `install.sh`.

At the end the machine was rebooted and after that the camera worked fine.

fbrinker, 2023-09-21:
I had a few problems before with newer kernels, but running `./uninstall.sh`, deleting the cloned directory and doing a fresh clone solved my build problems.

Works fine for me with kernel `6.5.3-arch1-1` on the newest thinkpad carbon x1, too :)

elhossary, 2023-09-21:
Worked again in Kernel 6.5.3

closed 2023-09-21 by elhossary