stefanpartheym/archlinux-ipu6-webcam #40: not working with 6.5 kernel series

noskoski, 2023-09-21:
Reset branch 'makepkg'
==> Iniciando pkgver()...
==> Removendo diretório $pkgdir/ existente...
==> Iniciando build()...
CMake Deprecation Warning at CMakeLists.txt:17 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- install prefix: /usr
INFOadd definition -DIPU_SYSVER_ipu6v5 for ipu6ep
Install camera configure files foripu6ep
-- Configuring done (0.1s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
IA_BCOMP_LIB
    linked by target "camhal_static" in directory /home/lan/dev/arch/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal
    linked by target "camhal" in directory /home/lan/dev/arch/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal

-- Generating done (0.0s)
CMake Generate step failed.  Build files cannot be regenerated correctly.
==> ERRO: Uma falha ocorreu em build().
    Abortando...
ERROR: Failed to build/install: intel-ipu6ep-camera-hal-git

jjeziorny, 2023-09-22:
not sure why the error, I had the same and after deleting `intel-ipu6ep-camera-hal-git/src/build/CMakeCache.txt` and trying the install again it worked. 

Mine contained  `-notfound` for the `IA_BCOMP_LIB`
Probably from the previous build where this lib wasn't there and/or required.

You can also try and cloning the repo again and running the install.

Nevertheless, it will compile and install after this (at least for me it did) but unfortunately camera did not work on 6.5 for me

stefanpartheym, 2023-09-27:
I can confirm this won't work with kernels `v6.5.4` and `v6.5.3`. I downgraded to kernel `6.4.12-arch1-1` to make it work again.

mkobel, 2023-09-27:
As written in https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/39#issuecomment-1729242173 I could solve the issue by uninstalling, doing a fresh clone and installing again.

stefanpartheym, 2023-09-28:
@mkobel Whoops, thanks for the hint. I think this is how I fixed the build for an earlier 6.5.X kernel a few weeks ago. I'll try this later and report back ;)

stefanpartheym, 2023-09-28:
I reinstalled from a fresh clone of the repo, but unfortunately doesn't work either. Also after downgrading to kernel `6.4.12-arch1-1` now also doesn't work anymore with installing from a fresh clone.
I keep getting the following error: `WARNING: erroneous pipeline: no element "icamerasrc"`
Need some time to figure this out I guess...

stefanpartheym, 2023-09-28:
Ok, never mind. It was `icamerasrc-git` and `intel-ipu6-dkms-git` that I had to reinstall from AUR – sorry.
Everything now works fine for me on kernel `6.5.5-arch1-1`.

@noskoski Please try as described by @mkobel in https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/40#issuecomment-1738095665 and also make a fresh reinstall of `icamerasrc-git` and `intel-ipu6-dkms-git` from AUR. Thanks.

striczkof, 2023-09-28:
Ye prolly my oversight on icamerasrc-git-fix not being removed in the uninstaller script.

I will work on fixes today or so. Plus still trying to get that permanently installation-breaking bug sorted out (narrowed it down to maybe kernel-install/dracut)

jjeziorny, 2023-10-02:
There's a typo on the install.sj

`./install.sh: line 206: echp: command not found
`

stefanpartheym, 2023-11-13:
I'll close this issue now, as we have found a "workaround" for the 6.5.x kernel versions and @noskoski gave positive feedback on this.

closed 2023-11-13 by stefanpartheym