stefanpartheym/archlinux-ipu6-webcam #41: install script fails doe to IA_BCOMP_LIB not set

naps62, 2023-09-28:
trying to install on a kernel 6.5 (I'm aware of #40 ). towards the end I get the following error:

```
Install camera configure files foripu6ep
-- Configuring done (0.0s)
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
IA_BCOMP_LIB
    linked by target "camhal_static" in directory /home/naps62/contrib/stefanpartheym/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal
    linked by target "camhal" in directory /home/naps62/contrib/stefanpartheym/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal
```

striczkof, 2023-09-28:
On it,  going to run a fresh Arch Linux install to see if I can replicate this one.

Maybe I should try Manjaro again. You're still using Manjaro right?

naps62, 2023-09-29:
yes, sorry for not mentioning (I often forget it myself 😅 )

striczkof, 2023-09-30:
The AUR package for intel-ipu6ep-camera-hal-git is now fixed and updated to latest. I believe it's working now. To install it together with the script, just delete the 'ntel-ipu6ep-camera-hal-git' directory and the script will then install the one from the AUR. See if it works.

I'm making a PR that will incorporate that plus allowing to build tiger lake and meteor lake. should be done by tomorrow.

striczkof, 2023-10-01:
@naps62 got my PR to a working state. Feel free to test it!

stefanpartheym, 2023-11-19:
Anyone experiencing this issue installing the drivers with the latest commits?
Cause I didn't. Would close this one, if it's not the case.

stefanpartheym, 2024-03-01:
Closing this one now.

closed 2024-03-01 by stefanpartheym