stefanpartheym/archlinux-ipu6-webcam #44: No output for camera sensors in Tiger Lake systems. Need more testers

striczkof, 2023-10-02:
Using this issue to figure out why the camera stack does not work on Tiger Lake. More Tiger Lake testers are welcome!
Please use PR #42 branch as it supports building for Tiger Lake. Grab it then install it by:
```sh
git clone --branch fix_and_rework https://github.com/striczkof/archlinux-ipu6-webcam.git
cd archlinux-ipu6-webcam
./install.sh # then restart after
```

What to test:
- Does the script install everything successfully? (if not, send the logs!)
- Does the `./test.sh` even run it at all? 
- Is the firmware being loaded at all? `sudo dmesg | grep ipu`
- Is it being blacklisted? `sudo gst-inspect-1.0 icamerasrc` (If you inspect it as a normal user, then it will show permission denied then somehow get blacklisted, but for some reason still works for me)
- Why is it being blacklisted (the inspect command above and `sudo gst-inspect` should be able to give out the reason)
- Does `v4l2-ctl --list-devices` show something?
- What does `systemctl status v4l2-relayd` say? (not gonna lie, this never helped me so far)

Big thanks to @usag1r for doing most of the test!

usag1r, 2023-10-02:
Hi, thanks for transferring this to open issue. Great idea!

./install.sh yielded in ended towards the end. You can see at the very end of the output below regarding building icamerasrc-git and something to do with the rules...

```
$ ./install.sh 

# Detected Tiger Lake (11th generation).
# Install headers for: linux
# Install package from the AUR/repos: linux-headers
warning: linux-headers-6.5.5.arch1-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Installing dependencies.
# Install package from the AUR/repos: base-devel
warning: base-devel-1-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: gst-plugin-pipewire
warning: gst-plugin-pipewire-1:0.3.80-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: gst-plugins-good
warning: gst-plugins-good-1.22.6-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Installing the camera stack.
# Install package from the AUR/repos: intel-ipu6-dkms-git
:: Resolving dependencies...
:: Calculating conflicts...
:: Calculating inner conflicts...

Aur (2) intel-ivsc-firmware-r2.764bd6e-1  intel-ipu6-dkms-git-r78.f83b0747b-1

:: Proceed to review? [Y/n]: 

:: Downloading PKGBUILDs...
 PKGBUILDs up to date
fetching devel info...
==> Making package: intel-ivsc-firmware r2.764bd6e-1 (Mon 02 Oct 2023 01:49:22 PM BST)
==> Retrieving sources...
  -> Updating ivsc-firmware git repo...
==> Validating source files with sha256sums...
    ivsc-firmware ... Skipped
==> Making package: intel-ivsc-firmware r2.764bd6e-1 (Mon 02 Oct 2023 01:49:24 PM BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating ivsc-firmware git repo...
==> Validating source files with sha256sums...
    ivsc-firmware ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of ivsc-firmware git repo...
Cloning into 'ivsc-firmware'...
done.
==> Starting pkgver()...
==> Updated version: intel-ivsc-firmware r10.10c214f-1
==> Sources are ready.
intel-ivsc-firmware-r2.764bd6e-1: parsing pkg list...
:: intel-ivsc-firmware-r2.764bd6e-1 is up to date -- skipping build
loading packages...
resolving dependencies...
looking for conflicting packages...

Packages (1) intel-ivsc-firmware-r10.10c214f-1

Total Installed Size:  10.97 MiB

:: Proceed with installation? [Y/n] 
[... 702 lines cut ...]
(1/1) checking available disk space                                                                      [##############################################################] 100%
:: Processing package changes...
(1/1) installing v4l2-relayd                                                                             [##############################################################] 100%
:: Running post-transaction hooks...
(1/2) Reloading system manager configuration...
(2/2) Arming ConditionNeedsUpdate...
/tmp/archlinux-ipu6-webcam
=> SUCCESS
# Build and install package: icamerasrc-git
/tmp/archlinux-ipu6-webcam/package-fix/icamerasrc-git /tmp/archlinux-ipu6-webcam
==> Making package: icamerasrc-git-fix r63.feeea6e-1 (Mon 02 Oct 2023 01:52:50 PM BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning icamerasrc git repo...
Cloning into bare repository '/tmp/archlinux-ipu6-webcam/package-fix/icamerasrc-git/icamerasrc'...
remote: Enumerating objects: 717, done.
remote: Counting objects: 100% (191/191), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 717 (delta 170), reused 168 (delta 158), pack-reused 526
Receiving objects: 100% (717/717), 438.14 KiB | 2.08 MiB/s, done.
Resolving deltas: 100% (510/510), done.
==> ERROR: 70-ipu6-psys.rules was not found in the build directory and is not a URL.
ERROR: Failed to build/install: icamerasrc-git
```

striczkof, 2023-10-02:
Ye I was kinda stupid and forgot to remove that from the source.

Now try it out again and see what happens.

usag1r, 2023-10-02:
Hi, no worries.
Tried again. Failed building icamerasrc again. Seems like a slightly different but related issue...

```
# Build and install package: icamerasrc-git

/tmp/archlinux-ipu6-webcam/package-fix/icamerasrc-git /tmp/archlinux-ipu6-webcam
==> Making package: icamerasrc-git-fix r63.feeea6e-1 (Tue 03 Oct 2023 12:25:08 AM BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning icamerasrc git repo...
Cloning into bare repository '/tmp/archlinux-ipu6-webcam/package-fix/icamerasrc-git/icamerasrc'...
remote: Enumerating objects: 717, done.
remote: Counting objects: 100% (191/191), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 717 (delta 170), reused 168 (delta 158), pack-reused 526
Receiving objects: 100% (717/717), 438.14 KiB | 351.00 KiB/s, done.
Resolving deltas: 100% (510/510), done.
  -> Found 70-ipu6-psys.rules
==> Validating source files with sha256sums...
    icamerasrc ... Skipped
    70-ipu6-psys.rules ... FAILED
==> ERROR: One or more files did not pass the validity check!
ERROR: Failed to build/install: icamerasrc-git
```

striczkof, 2023-10-03:
Another oversight again. I forgot about correcting the sha256sum. Fixed it. Sorry about that lol
This time I test run all the packages and they seem to work. I gotta do more testing after I walk to my classes today.

usag1r, 2023-10-03:
Haha, it happens!
Just re-ran the install script. Rebooted. All went smoothly. ./test.sh fails similar to before however..

striczkof, 2023-10-04:
Hey apologies, was quite a wild night earlier, trying to unfuck my system after a mate and I slapped liquid metal on the CPU/GPU dies so now I can't be doing much testing myself until weekend or in a month.

I saw your dmesg logs and I don't even know. Looks like your firmware is loading repeatedly instead of just one time like everyone else. Can you send you do `sudo dmesg | grep ipu` just to make sure I'm not hallucinating?

usag1r, 2023-10-04:
No worries! Here it is:
```
$ sudo dmesg | grep ipu

[    4.134940] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[    4.135599] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.135634] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.135638] intel-ipu6 0000:00:05.0: mapped as: 0x0000000096674dac
[    4.150717] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.150727] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.150731] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.166811] intel-ipu6 0000:00:05.0: IPC reset done
[    4.166831] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    4.182218] intel-ipu6 0000:00:05.0: FW version: 20210414
[    4.183900] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.183933] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.183937] intel-ipu6 0000:00:05.0: mapped as: 0x0000000096674dac
[    4.190511] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.190518] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.190521] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.192366] intel-ipu6 0000:00:05.0: IPC reset done
[    4.192371] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    4.201512] intel-ipu6 0000:00:05.0: FW version: 20210414
[    4.284941] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.284962] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.284965] intel-ipu6 0000:00:05.0: mapped as: 0x0000000096674dac
[    4.285107] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.285110] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.285113] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.300624] intel-ipu6 0000:00:05.0: IPC reset done
[    4.300633] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    4.307873] intel-ipu6 0000:00:05.0: FW version: 20210414
[    4.313396] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.313447] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.313451] intel-ipu6 0000:00:05.0: mapped as: 0x000000002fea463e
[    4.314164] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.314170] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.314172] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.319046] intel-ipu6 0000:00:05.0: IPC reset done
[    4.319051] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    4.325202] intel-ipu6 0000:00:05.0: FW version: 20210414
[    4.326982] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.327011] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.327016] intel-ipu6 0000:00:05.0: mapped as: 0x0000000096674dac
[    4.327207] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.327210] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.327213] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.329044] intel-ipu6 0000:00:05.0: IPC reset done
[    4.329047] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    4.335990] intel-ipu6 0000:00:05.0: FW version: 20210414
[    4.473745] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.473775] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.473779] intel-ipu6 0000:00:05.0: mapped as: 0x0000000096674dac
[    4.473928] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.473932] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.473934] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.475761] intel-ipu6 0000:00:05.0: IPC reset done
[    4.475762] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    4.485617] intel-ipu6 0000:00:05.0: FW version: 20210414
[    4.494659] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.494704] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[    4.494708] intel-ipu6 0000:00:05.0: mapped as: 0x000000002fea463e
[    4.501414] intel-ipu6 0000:00:05.0: IPU in secure mode
[    4.501421] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[    4.501425] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[    4.503445] intel-ipu6 0000:00:05.0: IPC reset done
[    4.503449] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[    4.509926] intel-ipu6 0000:00:05.0: FW version: 20210414
[    4.513831] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[    4.513870] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[... 541 lines cut ...]
[  477.265286] intel-ipu6 0000:00:05.0: IPU in secure mode
[  477.265292] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[  477.265295] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[  477.281476] intel-ipu6 0000:00:05.0: IPC reset done
[  477.281481] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[  477.287015] intel-ipu6 0000:00:05.0: FW version: 20210414
[  477.338417] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[  477.338479] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[  477.338483] intel-ipu6 0000:00:05.0: mapped as: 0x00000000a2478e96
[  477.338914] intel-ipu6 0000:00:05.0: IPU in secure mode
[  477.338920] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[  477.338923] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[  477.360184] intel-ipu6 0000:00:05.0: IPC reset done
[  477.360195] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[  477.365842] intel-ipu6 0000:00:05.0: FW version: 20210414
[  477.498992] intel-ipu6 0000:00:05.0: Device 0x9a19 (rev: 0x1)
[  477.499051] intel-ipu6 0000:00:05.0: physical base address 0x6052000000
[  477.499057] intel-ipu6 0000:00:05.0: mapped as: 0x00000000bedc471a
[  477.499302] intel-ipu6 0000:00:05.0: IPU in secure mode
[  477.499310] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[  477.499316] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[  477.516087] intel-ipu6 0000:00:05.0: IPC reset done
[  477.516101] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6_fw.bin
[  477.524543] intel-ipu6 0000:00:05.0: FW version: 20210414
```

Pietrell, 2024-01-18:
System: Microsoft Surface Pro 8
OS: Arch Linux x86_64
Kernel: 6.6.10-arch1-1-surface
DE: GNOME 45.3 Wayland

Installation
I had to run the script 2 times, becouse the first one i had a problem with a package version that paru updated, second time everything ok.

Test
I had to change the script: remove the -E flag of sudo because of this error: 
`WARNING: erroneous pipeline: no element "icamerasrc"`
Then i run again the script and got this output:
  ```
Setting pipeline to PAUSED ...
  error: XDG_RUNTIME_DIR is invalid or not set in the environment.
  Failed to set pipeline to PAUSED.
  Setting pipeline to NULL ...
  Freeing pipeline ...
```
Then changed the script again to force `ximagesink` and got this output:
```
Setting pipeline to PAUSED ...
[01-18 13:29:04.965] CamHAL[INF] aiqb file name hm11b1.aiqb
[01-18 13:29:04.966] CamHAL[INF] aiqb file name hm11b1.aiqb
[01-18 13:29:04.967] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.967] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-18 13:29:04.967] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-18 13:29:04.967] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.967] CamHAL[INF] aiqb file name hm11b1.aiqb
[01-18 13:29:04.967] CamHAL[INF] aiqb file name hm11b1.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL_2.aiqb
[01-18 13:29:04.968] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.968] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.969] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.969] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.969] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.969] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[01-18 13:29:04.969] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.969] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-18 13:29:04.969] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-18 13:29:04.969] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.969] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-18 13:29:04.969] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-18 13:29:04.969] CamHAL[ERR] invalid media format, default value used.
[01-18 13:29:04.969] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-18 13:29:04.969] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-18 13:29:04.969] CamHAL[ERR] invalid media format, default value used.
[... 103 lines cut ...]
Virtual Camera (platform:v4l2loopback-000):
	/dev/video0
```
output of `systemctl status v4l2-relayd`
```
4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: disabled)
     Active: active (running) since Thu 2024-01-18 10:02:10 CET; 3h 32min ago
    Process: 508 ExecCondition=/usr/bin/test -n ${VIDEOSRC} (code=exited, status=0/SUCCESS)
    Process: 528 ExecCondition=/usr/bin/test -n $FORMAT (code=exited, status=0/SUCCESS)
    Process: 530 ExecCondition=/usr/bin/test -n $WIDTH (code=exited, status=0/SUCCESS)
    Process: 533 ExecCondition=/usr/bin/test -n $HEIGHT (code=exited, status=0/SUCCESS)
    Process: 535 ExecCondition=/usr/bin/test -n $FRAMERATE (code=exited, status=0/SUCCESS)
    Process: 537 ExecCondition=/usr/bin/test -n ${CARD_LABEL} (code=exited, status=0/SUCCESS)
   Main PID: 539 (v4l2-relayd)
      Tasks: 5 (limit: 9312)
     Memory: 24.4M (peak: 27.9M)
        CPU: 123ms
     CGroup: /system.slice/v4l2-relayd.service
             └─539 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1>

Jan 18 10:02:10 archlinux systemd[1]: Starting v4l2-relay daemon service...
Jan 18 10:02:10 archlinux systemd[1]: Started v4l2-relay daemon service.
Jan 18 10:02:10 archlinux (sh)[539]: v4l2-relayd.service: Referenced but unset environment variable evaluates to an empty string: SPLASHSRC
```