stefanpartheym/archlinux-ipu6-webcam #46: Firefox, cheese, pipewire issues

mikeshultz, 2023-10-09:
Hey there, thanks for this repo.  

I've got an XPS 13 (9320) with one of these cams.  I'm having some issues with getting things working with browsers and Cheese.  `test.sh` works and looks clear.

When trying to run Cheese (unable to run with sudo), I get this in the pipewire logs:

```
spa.v4l2: '/dev/video0' can't allocate enough buffers (2 < 16)
pw.port: 0x555c70a593f0: negotiate buffers on node: -95 (Operation not supported)
pw.port: 0x555c70a593f0: state ready -> error (can't negotiate buffers on port)
pw.link: (49.0.0 -> 75.0.0) allocating -> error (can't negotiate buffers on port) (error-re>
pw.core: 0x555c7095ccd0: error -32 for resource 33: can't negotiate buffers on port
mod.client-node: 0x555c70bca250: error seq:218 -32 (can't negotiate buffers on port)
pw.core: 0x555c7095ccd0: error -32 for resource 33: can't negotiate buffers on port
mod.client-node: 0x555c70bca250: error seq:218 -32 (can't negotiate buffers on port)
spa.v4l2: '/dev/video0' can't allocate enough buffers (2 < 16)
pw.port: 0x555c70a593f0: negotiate buffers on node: -95 (Operation not supported)
pw.port: 0x555c70a593f0: state ready -> error (can't negotiate buffers on port)
pw.link: (49.0.0 -> 75.0.0) allocating -> error (can't negotiate buffers on port) (error-re>
pw.core: 0x555c7095ccd0: error -32 for resource 33: can't negotiate buffers on port
mod.client-node: 0x555c70be9ea0: error seq:261 -32 (can't negotiate buffers on port)
pw.core: 0x555c7095ccd0: error -32 for resource 33: can't negotiate buffers on port
mod.client-node: 0x555c70be9ea0: error seq:261 -32 (can't negotiate buffers on port)
```

I did do some manual installation before coming to this repo, so maybe there's some conflicts happening.  I tried to revert most of that and start over, but I could've missed something.

Running kernel `6.5.6-arch2-1` with Wayland. I wasn't able to find much information about these errors, could you provide me some direction?

striczkof, 2023-10-13:
Hey, the latest commit to icamerasrc broke it once again. For now, building the icamerasrc from the last working commit 'feeea6eed1bfd9baefa4a6ace2798e2eae3d3cce' is the work around.

Check: https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/42

stefanpartheym, 2023-11-19:
Closing this one due to inactivity.

closed 2023-11-19 by stefanpartheym