stefanpartheym/archlinux-ipu6-webcam #47: Thank you - Dell precision 5480

sysmso, 2023-10-11:
Thank you for all the work, just runned install.sh on my Dell precision 5480 on Manjaro 6.5 kernel and it works like a charm !

stefanpartheym, 2023-10-19:
Hi @sysmso,
thanks for your feedback. I added Dell Precision 5480 to the list of working devices in the README :)
Best regards 
Stefan

closed 2023-10-19 by stefanpartheym

wenchiching, 2023-11-10:
Hi @sysmso 

Does your camera on Dell precision 5480 works good after runned install.sh?

I am using precision 5680 and my camera not work, I tried https://github.com/intel/ipu6-camera-hal/issues/86#issuecomment-1804878367 but not success, maybe I did something wrong. Will try archlinux-ipu6-webcam later

sysmso, 2023-11-11:
Yes that's it. On a fresh manjaro install, just launched install.sh and rebooted and it works.