stefanpartheym/archlinux-ipu6-webcam #48: Dell Precision 5480

treiher, 2023-10-23:
Unfortunately, in contrast to #47 the webcam still does not work for me after installation. I'm using Arch Linux with kernel version `6.5.8-arch1-1` on a `Dell Precision 5480`. I used the state of today's `main` branch (698d5ad) for the installation. The camera works when executing `test.sh`. `install.sh` ran completely without any errors. After a reboot, the webcam still doesn't work. A "Virtual Camera" is detected, but there is no output (the screen just keeps being black when trying the WebRTC sample).

```
❯ pacman -Q | grep "ipu6\|ivsc\|hal"
intel-ipu6-dkms-git r149.3f8135801-1
intel-ipu6ep-camera-bin-fix r72.ff21b55-1
intel-ipu6ep-camera-hal-git-fix r93.9fa05a9-1
intel-ivsc-firmware r10.10c214f-1
```

```
❯ lsmod | grep "ivsc\|ipu\|ov"
ov01a10                28672  1
intel_vsc              16384  3 mei_ace,mei_csi,ov01a10
intel_ipu6_psys       126976  0
intel_ipu6_isys       167936  0
videobuf2_dma_contig    28672  1 intel_ipu6_isys
v4l2_fwnode            32768  2 ov01a10,intel_ipu6_isys
v4l2_async             28672  3 v4l2_fwnode,ov01a10,intel_ipu6_isys
videobuf2_v4l2         40960  1 intel_ipu6_isys
videodev              389120  8 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov01a10,v4l2loopback,intel_ipu6_isys
videobuf2_common       94208  4 videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,videobuf2_memops
mc                     90112  6 v4l2_async,videodev,videobuf2_v4l2,ov01a10,intel_ipu6_isys,videobuf2_common
intel_ipu6            143360  2 intel_ipu6_isys,intel_ipu6_psys
```

```
❯ dkms status                                        
ipu6-drivers/r149.3f8135801, 6.5.8-arch1-1, x86_64: installed (original_module exists)
v4l2loopback/r10.cae2954, 6.5.8-arch1-1, x86_64: installed
```

```
❯ modprobe -c | grep v4l2loopback
options v4l2loopback exclusive_caps=1 card_label="Virtual Camera"
```

```
❯ v4l2-ctl --list-devices
ipu6 (PCI:pci:pci0000:00):
        /dev/video0
        /dev/video1
        /dev/video2
        /dev/video3
        /dev/video4
        /dev/video5
        /dev/video6
        /dev/video7

ipu6 (pci:pci0000:00):
        /dev/media0

Virtual Camera (platform:v4l2loopback-000):
        /dev/video8
```

```
❯ journalctl -k | grep ipu            
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: physical base address 0x628e000000
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000da2268ed
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: IPC reset done
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: FW version: 20230102
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: CSE authenticate_run done
Okt 23 15:23:39 pyrite kernel: intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
Okt 23 15:23:39 pyrite kernel: intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
Okt 23 15:23:39 pyrite kernel: intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
Okt 23 15:23:50 pyrite kernel: intel-ipu6-isys intel-ipu6-isys0: bind ov01a10 18-0036 nlanes is 1 port is 2
Okt 23 15:23:50 pyrite kernel: intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
Okt 23 15:37:03 pyrite kernel: intel-ipu6-isys intel-ipu6-isys0: stream on ov01a10 18-0036
Okt 23 15:37:08 pyrite kernel: intel-ipu6-isys intel-ipu6-isys0: stream off ov01a10 18-0036
Okt 23 15:37:57 pyrite kernel: intel-ipu6-isys intel-ipu6-isys0: stream on ov01a10 18-0036
Okt 23 15:37:59 pyrite kernel: intel-ipu6-isys intel-ipu6-isys0: stream off ov01a10 18-0036
```

```
❯ journalctl -b -u v4l2-relayd.service
Okt 23 15:23:41 pyrite systemd[1]: Starting v4l2-relay daemon service...
Okt 23 15:23:41 pyrite systemd[1]: Started v4l2-relay daemon service.
Okt 23 15:23:41 pyrite (sh)[876]: v4l2-relayd.service: Referenced but unset environment variable evaluates to an empt>
Okt 23 15:25:40 pyrite v4l2-relayd[876]: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
Okt 23 15:25:41 pyrite v4l2-relayd[876]: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
```

`v4l2-relayd[876]: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed` pops up every time an attempt is made to use the webcam.

Any ideas why it doesn't work for me?

treiher, 2023-10-25:
The problem seems to be caused by v4l2-relayd. I'm able to use the webcam in Chromium after running

`sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov01a10-uf ! video/x-raw,format=NV12,width=1280,height=720 ! v4l2sink device=/dev/video8`

(as described in https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/33#issuecomment-1762967139).

stefanpartheym, 2023-11-19:
Hi @treiher,
are you still having issues? If so, try to update to the latest kernel version and use a fresh clone of this repo to install the drivers.

treiher, 2023-11-23:
Hi @stefanpartheym,

Unfortunately, the issue still persists with the latest kernel version and a fresh installation of the drivers using the current `main` (d3e96251bee388a40d863eac68b81ddc2c11016c). Even worse, the workaround mentioned in https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/48#issuecomment-1778778160 does not work anymore. :frowning_face: 

```console
❯ sudo -E gst-launch-1.0 icamerasrc buffer-count=7 device-name=ov01a10-uf ! video/x-raw,format=NV12,width=1280,height=720 ! v4l2sink device=/dev/video8
[11-23 10:34:32.376] CamHAL[ERR] Malformed ET range in exposure time range configuration
[11-23 10:34:32.376] CamHAL[ERR] Parse AE eExposure time range failed
[11-23 10:34:32.376] CamHAL[ERR] Malformed ET range in exposure time range configuration
[11-23 10:34:32.376] CamHAL[ERR] Parse AE gain range failed
Setting pipeline to PAUSED ...
[11-23 10:34:32.453] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-23 10:34:32.453] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-23 10:34:32.454] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[11-23 10:34:32.454] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev16 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3153): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.000228848
Setting pipeline to NULL ...
Freeing pipeline ...
```

fiskhest, 2024-04-10:
FWIW I am unable to get my webcam running on `6.8.4-arch1-1` with the [latest commit](https://github.com/stefanpartheym/archlinux-ipu6-webcam/commit/8aed83fc36beb6085b6717ca38cdaac9bb8d59de).

stefanpartheym, 2024-05-24:
> FWIW, also on a 5480 and I am unable to get my webcam running on `6.8.4-arch1-1` with the [latest commit](https://github.com/stefanpartheym/archlinux-ipu6-webcam/commit/8aed83fc36beb6085b6717ca38cdaac9bb8d59de).

Yep, kernel `v6.8.x` and `v6.9.x` won't work. You need an LTS kernel, or wait for `v6.10.x` release (as mentioned in #80 and #82).

fiskhest, 2024-07-30:
Are there any hints as to what else would be missing given a `v6.10.x` release installed  on the system?
I'm on `6.10.2-arch1-1` but was thus far unable to get a working internal camera setup.

stefanpartheym, 2024-10-11:
> Are there any hints as to what else would be missing given a `v6.10.x` release installed on the system? I'm on `6.10.2-arch1-1` but was thus far unable to get a working internal camera setup.

As far as I know, no one got it working on `v6.10.x` so far :(

Closing this one for now.

closed 2024-10-11 by stefanpartheym