stefanpartheym/archlinux-ipu6-webcam #49: Not working on Thinkpad X1 Carbon Gen 11

wille, 2023-10-27:
Hi, I run Thinkpad X1 Carbon Gen 11 not working with the OLED display and the only option available for the camera was the MIPI one :joy: 

Tried both 6.5.8-hardened1-1-hardened and the regular 6.5.8 kernel but the camera is all black, and during some attempts I got the LED to light up, but it will not go away until a reboot.

I will be happy to help in any way I can.

Here are the logs from `./test.sh`

```
Setting pipeline to PAUSED ...
[10-27 15:55:39.495] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-27 15:55:39.496] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-27 15:55:39.496] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.496] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-27 15:55:39.496] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-27 15:55:39.497] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.497] CamHAL[INF] aiqb file name ov8856.aiqb
[10-27 15:55:39.497] CamHAL[INF] aiqb file name ov8856.aiqb
[10-27 15:55:39.497] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.497] CamHAL[INF] aiqb file name ov8856.aiqb
[10-27 15:55:39.497] CamHAL[INF] aiqb file name ov8856.aiqb
[10-27 15:55:39.497] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.497] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-27 15:55:39.498] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-27 15:55:39.499] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.499] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-27 15:55:39.499] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-27 15:55:39.499] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.499] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-27 15:55:39.500] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-27 15:55:39.500] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.500] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-27 15:55:39.500] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-27 15:55:39.501] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.501] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-27 15:55:39.501] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-27 15:55:39.501] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.501] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-27 15:55:39.501] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-27 15:55:39.501] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.501] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-27 15:55:39.501] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-27 15:55:39.502] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.502] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-27 15:55:39.502] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-27 15:55:39.502] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.502] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-27 15:55:39.502] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-27 15:55:39.503] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.503] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-27 15:55:39.503] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-27 15:55:39.503] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.503] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-27 15:55:39.504] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-27 15:55:39.504] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.504] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-27 15:55:39.504] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-27 15:55:39.504] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.504] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-27 15:55:39.504] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-27 15:55:39.504] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.504] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[10-27 15:55:39.505] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[10-27 15:55:39.505] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.505] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[10-27 15:55:39.505] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[10-27 15:55:39.506] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.506] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-27 15:55:39.506] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-27 15:55:39.507] CamHAL[ERR] invalid media format, default value used.
[... 65 lines cut ...]
[10-27 15:55:39.512] CamHAL[ERR] invalid media format, default value used.
[10-27 15:55:39.512] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.512] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.512] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.512] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-27 15:55:39.512] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[10-27 15:55:39.520] CamHAL[ERR] Get entity fail for calling getEntityById
[10-27 15:55:39.520] CamHAL[ERR] Get entity fail for calling getEntityById
[10-27 15:55:39.520] CamHAL[ERR] setup Link ov13b10  [-1:0] ==> Intel IPU6 CSI-2  [-1x0] enable 1 failed.
[10-27 15:55:39.520] CamHAL[ERR] set MediaCtlConf McLink failed: ret = -1
[10-27 15:55:39.520] CamHAL[ERR] set up mediaCtl failed
[10-27 15:55:39.520] CamHAL[ERR] @configure Device Configure failed
[10-27 15:55:39.520] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3153): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.004024563
Setting pipeline to NULL ...
Freeing pipeline ...
```

jannec22, 2023-11-02:
having similiar issue, on Dell Precision 5480. install.sh passed without errors, after reboot LED lights up but test.sh reports the following:

```
Setting pipeline to PAUSED ...
[11-02 16:14:37.331] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-02 16:14:37.332] CamHAL[INF] aiqb file name ov01a10.aiqb
[11-02 16:14:37.332] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[11-02 16:14:37.332] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev16 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[11-02 16:14:40.413] CamHAL[ERR] Open: Failed to open device node /dev/video0 Device or resource busy
[11-02 16:14:40.413] CamHAL[ERR] Open device(Generic) failed:-16
[11-02 16:14:40.413] CamHAL[ERR] Create devices failed:-16
[11-02 16:14:40.413] CamHAL[ERR] @configure Device Configure failed
[11-02 16:14:40.413] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3153): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:03.077702853
Setting pipeline to NULL ...
[11-02 16:14:40.414] CamHAL[ERR] Close: Device node /dev/video0 is not opened! No such file or directory
[11-02 16:14:40.414] CamHAL[ERR] Close: Device node /dev/video0 is not opened! No such file or directory
Freeing pipeline ...
```

```
❯  uname -a
Linux archbook 6.5.9-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 26 Oct 2023 00:52:20 +0000 x86_64 GNU/Linux

❯  pacman -Q | grep "ipu6\|ivsc\|hal"
intel-ipu6-dkms-git r149.3f8135801-1
intel-ipu6ep-camera-bin-fix r73.af5ba0c-1
intel-ipu6ep-camera-hal-git-fix r93.9fa05a9-1
intel-ivsc-firmware r10.10c214f-1

❯  lsmod | grep "ivsc\|ipu\|ov"
intel_ipu6_isys       167936  0
videobuf2_dma_contig    28672  1 intel_ipu6_isys
videobuf2_v4l2         40960  2 intel_ipu6_isys,uvcvideo
videobuf2_common       94208  6 videobuf2_vmalloc,videobuf2_dma_contig,videobuf2_v4l2,intel_ipu6_isys,uvcvideo,videobuf2_memops
ov01a10                28672  1
intel_vsc              16384  3 mei_ace,mei_csi,ov01a10
v4l2_fwnode            32768  2 ov01a10,intel_ipu6_isys
intel_ipu6_psys       126976  0
v4l2_async             28672  3 v4l2_fwnode,ov01a10,intel_ipu6_isys
videodev              389120  8 v4l2_async,v4l2_fwnode,videobuf2_v4l2,ov01a10,v4l2loopback,intel_ipu6_isys,uvcvideo
intel_ipu6            143360  2 intel_ipu6_isys,intel_ipu6_psys
mc                     90112  7 v4l2_async,videodev,videobuf2_v4l2,ov01a10,intel_ipu6_isys,uvcvideo,videobuf2_common

❯  dkms status
ipu6-drivers/r149.3f8135801, 6.5.9-arch2-1, x86_64: installed (original_module exists)
v4l2loopback/r10.cae2954, 6.5.9-arch2-1, x86_64: installed

❯  modprobe -c | grep v4l2loopback
options v4l2loopback exclusive_caps=1 card_label="Virtual Camera"

❯  v4l2-ctl --list-devices
ipu6 (pci:pci0000:00):
    /dev/media0

Virtual Camera (platform:v4l2loopback-000):
    /dev/video8

Failed to open /dev/video0: Device or resource busy

❯  journalctl -k | grep ipu
Nov 02 16:13:57 archbook kernel: intel_ipu6: loading out-of-tree module taints kernel.
Nov 02 16:13:57 archbook kernel: intel_ipu6: module verification failed: signature and/or required key missing - tainting kernel
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x0)
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: physical base address 0x628e000000
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: mapped as: 0x00000000acb292c9
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: IPU in secure mode
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: IPC reset done
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: FW version: 20230102
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: CSE authenticate_run done
Nov 02 16:13:57 archbook kernel: intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
Nov 02 16:13:57 archbook kernel: intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
Nov 02 16:13:57 archbook kernel: intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
Nov 02 16:13:58 archbook kernel: intel-ipu6-isys intel-ipu6-isys0: isys port open ready failed -16
Nov 02 16:13:58 archbook kernel: intel-ipu6-isys intel-ipu6-isys0: Device close failure: -16
Nov 02 16:14:00 archbook kernel: intel-ipu6-isys intel-ipu6-isys0: isys port open ready failed -16
Nov 02 16:14:00 archbook kernel: intel-ipu6-isys intel-ipu6-isys0: Device close failure: -16
```

fbrinker, 2023-11-22:
It's working for me on my thinkpad carbon x1 Gen 11 aigain since https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/53

You may need to uninstall it, delete the cloned directory and do it all over again :)

wille, 2023-12-05:
I'm all over to 6.6.4 now (and linux-hardened 6.5.13-hardened) so I don't have it working (6.6.2, 6.6.3 didn't work either)

wille, 2023-12-19:
It's now working on 6.6.7-hardened1-1-hardened and 6.6.7.arch1-1

I uninstalled, deleted the repo clone, forgot about it for a couple of weeks, cloned a fresh copy then installed it.

wille, 2023-12-20:
Works once after installation, triggered this twice by uninstalling, installing and rebooting. After another reboot camera will go black and the led will not light up.

JanusDC, 2023-12-20:
I confirm the exact same behaviour on the ThinkPad X1 Yoga Gen 8. It worked with the kernel 6.6.6 although, and stopped working again with 6.6.7. I will try latter to downgrade the kernel and install it again to see if I get the webcam working again.

wille, 2023-12-20:
Did uninstall, install and reboot a couple of times cannot get it to work at all now. Did get it working once more earlier today.

JanusDC, 2023-12-20:
Just uninstalled, removed the yay cache packages, removed the git repository and cloned again, reinstalled, and everything is working. The test does not work, but the webcam is working on Chrome (I tested it with Google Meet) and with Zoom. It takes forever for the first time to start, but after it does, then it starts faster the next time.

Lenovo ThinkPad X1 Yoga Gen 8 (same hardware as the X1 Carbon Gen 11).
Arch Linux
Kernel 6.6.7-zen1-1-zen

JanusDC, 2023-12-20:
> Just uninstalled, removed the yay cache packages, removed the git repository and cloned again, reinstalled, and everything is working. The test does not work, but the webcam is working on Chrome (I [...]

Just restarted, and it is not working again. It shows a black square, and that's all.

wille, 2023-12-20:
You just described again what I am seeing. If you have time, try to redo the process again and see if you can get it working again. If you do, when you reboot, it will break.

JanusDC, 2023-12-20:
OK! I just run the install.sh script again (without uninstalling previously), the webcam still was not working, I restarted the computer, and the webcam worked again. 

I guess we have to try to understand what part of the script is making it work. The packages were already installed, so it skipped those.

wille, 2023-12-21:
Will try that to confirm and see if we can reliably reproduce it

wille, 2023-12-22:
Got it working temporarily and actually joined my first video call with the camera working, the mic was not working but I'm debugging what happened

Run `sudo systemctl restart v4l2-relayd.service` until it works. It helps when the led refuses to turn of as well

JanusDC, 2024-01-15:
> OK! I just run the install.sh script again (without uninstalling previously), the webcam still was not working, I restarted the computer, and the webcam worked again.
>
> I guess we have to try to [...]

I just got my laptop repaired, they had to change the motherboard because there were some fails. The change came with a change on the CPU because there was no stock of mine. After changing it, now the webcam is working properly with kernel 6.6.10-zen and this script.

So, I confirm it works for the ThinkPad X1 Yoga Gen 8, with processor i7-1365U, using kernel 6.6.10-zen.

stefanpartheym, 2024-03-01:
Hi @wille,

is it working now reliably?
Also, the mic issue is most likely not related to this.

wille, 2024-03-01:
Hi @stefanpartheym 

I've been running this for a couple of months now. It's not reliable, but I've learnt what to do to make it work.

Some boots it will never work and the logs from v4l2relayd.service looks similar to what I posted in the original post above, I will have to restart my computer and if I get it working then, that error will not occur again until another restart.

Now if I pass the problem above, then it's very common that the v4l2 service mentions some kind of timeout error.
I have to restart the service, everything between 1-30 times, with google meet open, until the webcam appears. If I leave a meeting or shut of the camera, I'll have to repeat this process.

isimluk, 2024-03-09:
Got the thing working on X1 Gen 11 Intel Core i7 1355U Raptor Lake.

Steps:
 - stare at `CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1` about 60 times during the evening
 - mess around with this repo and AUR packages and various kernels
 - learn about ipu6 and all adjacent technologies
 - unload and load kernel modules manually (including ov2740) 
 - turn the camera off and on in bios
 - give up, order an usb camera, go to bed
 - wake up, tidy up the house, boot, and accidentally hit the last `gst-launch-1.0` command
 - look at your tired face on the screen, it works :partying_face: 

While it works let me capture the software components on the system.

Packages
 - 6.6.21-1-lts
 - intel-ipu6-camera-bin r73.af5ba0c-1
 - intel-ipu6-dkms-git r158.fb4c17019-1
 - intel-ipu6ep-camera-hal-git r95.da2e282-1
 - intel-ipu6ep-camera-hal-git-debug r95.da2e282-1
 - intel-ivsc-firmware r10.10c214f-1
 - intel-ucode 20231114-2
 - pipewire-v4l2 1:1.0.3-1
 - v4l2-relayd-debug r42.6fd6b6a-1
 - v4l2loopback-dkms 0.12.7-2
 - v4l2loopback-utils 0.12.7-2
 - gst-plugins-base 1.24.0-1
 - gst-plugins-base-libs 1.24.0-1
 - gst-plugins-good 1.24.0-1
 - gstreamer 1.24.0-1

last dmesg snippet during the boot
```
[   12.422659] intel-ipu6 0000:00:05.0: Device 0xa75d (rev: 0x1)
[   12.422684] intel-ipu6 0000:00:05.0: physical base address 0x603c000000
[   12.422687] intel-ipu6 0000:00:05.0: mapped as: 0x00000000b60cdeec
[   12.422802] intel-ipu6 0000:00:05.0: IPU in secure mode
[   12.422807] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   12.422810] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[   12.424697] intel-ipu6 0000:00:05.0: IPC reset done
[   12.424702] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[   12.425383] intel-ipu6 0000:00:05.0: FW version: 20230102
[   12.428910] intel-ipu6 0000:00:05.0: Found supported sensor INT3474:01
[   12.429106] intel-ipu6 0000:00:05.0: Connected 1 cameras
[   12.430405] intel-ipu6 0000:00:05.0: Sending BOOT_LOAD to CSE
[   12.439503] intel-ipu6 0000:00:05.0: Sending AUTHENTICATE_RUN to CSE
[   12.508365] intel-ipu6 0000:00:05.0: CSE authenticate_run done
[   12.508398] intel-ipu6 0000:00:05.0: IPU6-v3 driver version 1.0
[   12.511101] intel-ipu6-isys intel-ipu6-isys0: bind ov2740 18-0036 nlanes is 2 port is 1
[   12.511625] intel-ipu6-isys intel-ipu6-isys0: All sensor registration completed.
[   12.513266] intel-ipu6-psys intel-ipu6-psys0: pkg_dir entry count:8
[   12.514544] intel-ipu6-psys intel-ipu6-psys0: psys probe minor: 0
```

isimluk, 2024-03-09:
After reboot, it wouldn't work again. But cycling the sensor initialization fixed it.

```
modprobe -r ov2740 && modprobe ov2740
```

stefanpartheym, 2024-10-25:
Will close this issue now.

closed 2024-10-25 by stefanpartheym