stefanpartheym/archlinux-ipu6-webcam #50: Conflicts packages in v1.1.0 on Latitude 7440

nottatdat, 2023-10-30:
Got this error when trying to run `install.sh`

```sh
➜  archlinux-ipu6-webcam git:(v1.1.0) ./install.sh       
# Build and install package: intel-ipu6-dkms-git
~/tmp/archlinux-ipu6-webcam/intel-ipu6-dkms-git ~/tmp/archlinux-ipu6-webcam
==> Making package: intel-ipu6-dkms-git-fix r78.f83b0747b-1 (Mon 30 Oct 2023 03:37:17 PM +07)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning ipu6-drivers git repo...
Cloning into bare repository '/home/dat/tmp/archlinux-ipu6-webcam/intel-ipu6-dkms-git/ipu6-drivers'...
remote: Enumerating objects: 81226, done.
remote: Counting objects: 100% (849/849), done.
remote: Compressing objects: 100% (408/408), done.
remote: Total 81226 (delta 406), reused 696 (delta 312), pack-reused 80377
Receiving objects: 100% (81226/81226), 213.80 MiB | 2.64 MiB/s, done.
Resolving deltas: 100% (9431/9431), done.
==> Validating source files with sha256sums...
    ipu6-drivers ... Skipped
==> Extracting sources...
  -> Creating working copy of ipu6-drivers git repo...
Reset branch 'makepkg'
==> Starting prepare()...
Cloning into 'ivsc-driver'...
remote: Enumerating objects: 330, done.
remote: Counting objects: 100% (126/126), done.
remote: Compressing objects: 100% (74/74), done.
remote: Total 330 (delta 63), reused 75 (delta 39), pack-reused 204
Receiving objects: 100% (330/330), 141.56 KiB | 1.68 MiB/s, done.
Resolving deltas: 100% (121/121), done.
==> Starting pkgver()...
==> Removing existing $pkgdir/ directory...
==> Entering fakeroot environment...
==> Starting package()...
==> Tidying install...
  -> Removing libtool files...
  -> Purging unwanted files...
  -> Removing static library files...
  -> Stripping unneeded symbols from binaries and libraries...
  -> Compressing man and info pages...
==> Checking for packaging issues...
==> Creating package "intel-ipu6-dkms-git-fix"...
  -> Generating .PKGINFO file...
  -> Generating .BUILDINFO file...
  -> Generating .MTREE file...
  -> Compressing package...
==> Leaving fakeroot environment.
==> Finished making: intel-ipu6-dkms-git-fix r78.f83b0747b-1 (Mon 30 Oct 2023 03:39:23 PM +07)
==> Installing package intel-ipu6-dkms-git-fix with pacman -U...
loading packages...
resolving dependencies...
looking for conflicting packages...
:: intel-ipu6-dkms-git-fix and intel-ipu6-dkms-git are in conflict. Remove intel-ipu6-dkms-git? [y/N] 
error: unresolvable package conflicts detected
error: failed to prepare transaction (conflicting dependencies)
:: intel-ipu6-dkms-git-fix and intel-ipu6-dkms-git are in conflict
==> WARNING: Failed to install built package(s).
~/tmp/archlinux-ipu6-webcam
ERROR: Failed to install: intel-ipu6-dkms-git
```

striczkof, 2023-11-04:
Huh, I thought it auto replaces the conflicted packages. Can you uninstall the old AUR packages first?

stefanpartheym, 2024-03-01:
Closing due to inactivity. Please reopen, if necessary.

closed 2024-03-01 by stefanpartheym