stefanpartheym/archlinux-ipu6-webcam #52: Test works but apps only have black output

lmaksymenko, 2023-11-05:
Hey guys. I've had the drivers work just fine for me in the past. I haven't used the webcam much so I'm not sure when exactly the issue started. 

Currently, running test.sh will open a window with a live camera stream. Everything works great. However, applications which worked in the past  (Discord, Firefox, Chromium, etc) all find the camera but only show a black screen. 

I've tried uninstalling and reinstalling so far, but not too much more. 

I suspect this is something related to v4l2 virtual camera, but I honestly don't have a clue as to where to start debugging it. 

Any help for be appreciated!

stefanpartheym, 2023-11-13:
Which kernel version are you running on? As described in issue #40 , it is currently not working on kernel versions `6.6.x`.
On kernel versions `6.5.x` please make sure to follow the instructions *Solving error WARNING: erroneous pipeline: no element "icamerasrc"* in the [README](https://github.com/stefanpartheym/archlinux-ipu6-webcam#solving-error-warning-erroneous-pipeline-no-element-icamerasrc).

benediktmag, 2023-12-08:
Same here, test.sh works but no black screen on all other applications. 
Running as root does not seem to be the problem. 
I'm running Endeavour with kernel 6.6.4.

stefanpartheym, 2024-10-11:
You might want to try a new LTS kernel like `v6.6.56-1-lts`.

stefanpartheym, 2024-10-25:
As already found out by #71, this probably is an issue with the service file of `v4l2-relayd` on Wayland.
It is now fixed with commit d1c471caf4099e25d392ad23a3410706692a8701.
Please consider updating to the latest LTS kernel and use a fresh clone of this repo.