stefanpartheym/archlinux-ipu6-webcam #53: Does not work after upgrading to Kernel 6.6.1

elhossary, 2023-11-09:
Hi All,

I upgraded the kernel to 6.6.1 and reinstalled this package without errors.
It does not work (renders black image) after I restarted the machine.

BTW, I also knew that IVSC driver was merged into Kernel 6.6:
https://www.phoronix.com/news/Intel-IVSC-Media-Linux-6.6

striczkof, 2023-11-09:
at the first glance to dmesg, IPU drivers dont even load at all.

striczkof, 2023-11-10:
i used https://github.com/intel/ipu6-drivers/pull/175 PR but now I am getting different sets of errors. Ill troubleshoot this on weekeends

fbrinker, 2023-11-16:
I have no deeper knowledge on this topic, but I'm wondering why kernel updates keep breaking it. Is there any pattern we could look into to prevent this?

striczkof, 2023-11-16:
@elhossary okay so Intel has already merged that PR so no need to do some workarounds, see if it works for you now.

elhossary, 2023-11-16:
@striczkof thanks,
Excuse me for the ignorance, test what exactly?
I uninstalled this repo, and it does not work, re-installed and does not work too

i-oliva, 2023-11-17:
I uninstalled and re-installed the repo and it did not work either

arcan1s, 2023-11-17:
work for me on 6.6.x with the latest intel commits

elhossary, 2023-11-17:
how? can you describe roughly what you have done?

arcan1s, 2023-11-17:
hmm, nothing special, i just updated all -*git packages to the latest commits (which is performed automatically during rebuild). I have carbon x1 gen10. I didn't try yet kernel module mentioned above

stefanpartheym, 2023-11-18:
Hey @arcan1s, thanks for the hint. I can confirm, drivers work fine for me on my Dell XPS 13 Plus 9320.

@elhossary You can try the following steps to get the drivers working on kernel `v6.6.1` (at least this worked for me):
- (on previous kernel version) uninstall drivers via `./uninstall.sh` script
- Update all system packages including linux kernel and headers
- Reboot
- Clone this repo into a fresh directory (to make sure, no build cache from previous and probably outdated builds is being used)
- Install the drivers using the `./install.sh` script
- Reboot
- You'll probably need a fresh reinstallation of the `icamerasrc-git` package via `yay --nocleanmenu --nodiffmenu --noeditmenu --rebuild -S icamerasrc-git`
- Finally, test if it works via the `./test.sh` script

Just for clarification: If you run the install script, `yay` will automatically update the `-fix` packages (like `intel-ipu6ep-camera-bin` and `intel-ipu6ep-camera-hal-git`) to the latest commits.

Hope this helps :)
Let me know!

elhossary, 2023-11-18:
@stefanpartheym Thank you
I went exactly as you mentioned in the steps, but the install step fails
Here is the trace 
```
./install.sh 
# Install headers for: linux
# Install package from the AUR/repos: linux-headers
Sync Explicit (1): linux-headers-6.6.1.arch1-1
[sudo] password for muhoss: 
warning: linux-headers-6.6.1.arch1-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: base-devel
Sync Explicit (1): base-devel-1-1
warning: base-devel-1-1 is up to date -- skipping
 there is nothing to do
=> SUCCESS
# Install package from the AUR/repos: intel-ipu6-dkms-git
AUR Explicit (1): intel-ipu6-dkms-git-r78.f83b0747b-1
AUR Dependency (1): intel-ivsc-firmware-r2.764bd6e-1
:: (1/2) Downloaded PKGBUILD: intel-ivsc-firmware
:: (2/2) Downloaded PKGBUILD: intel-ipu6-dkms-git
  2 intel-ipu6-dkms-git                      (Build Files Exist)
  1 intel-ivsc-firmware                      (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
  2 intel-ipu6-dkms-git                      (Build Files Exist)
  1 intel-ivsc-firmware                      (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
==> Making package: intel-ivsc-firmware r2.764bd6e-1 (Sat 18 Nov 2023 07:28:54 PM CET)
==> Making package: intel-ipu6-dkms-git r78.f83b0747b-1 (Sat 18 Nov 2023 07:28:54 PM CET)
==> Retrieving sources...
==> Retrieving sources...
  -> Cloning ivsc-firmware git repo...
Cloning into bare repository '/home/muhoss/.cache/yay/intel-ivsc-firmware/ivsc-firmware'...
  -> Cloning ipu6-drivers git repo...
Cloning into bare repository '/home/muhoss/.cache/yay/intel-ipu6-dkms-git/ipu6-drivers'...
remote: Enumerating objects: 56, done.
remote: Counting objects: 100% (56/56), done.
remote: Compressing objects: 100% (30/30), done.
remote: Enumerating objects: 81243, done.
remote: Counting objects: 100% (867/867), done.
remote: Compressing objects: 100% (419/419), done.
remote: Total 56 (delta 38), reused 39 (delta 26), pack-reused 0
Receiving objects: 100% (56/56), 1.67 MiB | 14.13 MiB/s, done.
Resolving deltas: 100% (38/38), done.
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    ivsc-firmware ... Skipped
remote: Total 81243 (delta 416), reused 706 (delta 317), pack-reused 80376
Receiving objects: 100% (81243/81243), 213.85 MiB | 22.55 MiB/s, done.
Resolving deltas: 100% (9437/9437), done.
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    ipu6-drivers ... Skipped
:: (1/2) Parsing SRCINFO: intel-ipu6-dkms-git
:: (2/2) Parsing SRCINFO: intel-ivsc-firmware
==> Making package: intel-ivsc-firmware r2.764bd6e-1 (Sat 18 Nov 2023 07:29:06 PM CET)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating ivsc-firmware git repo...
==> Validating source files with sha256sums...
    ivsc-firmware ... Skipped
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of ivsc-firmware git repo...
[... 479 lines cut ...]
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_aiqb_parser.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_bcomp.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_cca.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_ccat.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_cmc_parser.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_coordinate.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_dvs.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_emd_decoder.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_exc.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_isp_bxt.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_lard.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_log.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_ltm.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_mkn.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_nvm.so exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libia_p2p_ipu6.a exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/libipu6.a exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/pkgconfig/ia_imaging.pc exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/pkgconfig/libgcss.pc exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/pkgconfig/libiacss.pc exists in filesystem
intel-ipu6ep-camera-bin-fix: /usr/lib/ipu_tgl/pkgconfig/libipu6.pc exists in filesystem
Errors occurred, no packages were upgraded.
==> WARNING: Failed to install built package(s).
ERROR: Failed to build/install: intel-ipu6ep-camera-bin
```

arcan1s, 2023-11-18:
@elhossary  looks like there is a conflicting package installed (you can check by `pacman -Qo /usr/lib/ipu_tgl/pkgconfig/ia_imaging.pc`). Either you installed it without using makepkg before

elhossary, 2023-11-18:
@arcan1s Thanks for noticing the error, There were no package owning them, so I removed them manually then the installation works.
@stefanpartheym Now I can confirm that it works.

Many thanks for both of you

closed 2023-11-18 by elhossary

fbrinker, 2023-11-22:
Thank you, it's working again :)