stefanpartheym/archlinux-ipu6-webcam #57: Rebuilding the DKMS in 6.6.2 is problematic

elhossary, 2023-11-21:
Hi,
I get these errors while upgrading the Kernel to 6.6.2
```
:: Running post-transaction hooks...
(1/7) Creating system user accounts...
(2/7) Reloading system manager configuration...
(3/7) Arming ConditionNeedsUpdate...
(4/7) Rebuilding certificate stores...
(5/7) Updating module dependencies...
(6/7) Install DKMS modules
==> dkms install --no-depmod ipu6-drivers/0.0.0 -k 6.6.2-arch1-1
Error! Bad return status for module build on kernel: 6.6.2-arch1-1 (x86_64)
Consult /var/lib/dkms/ipu6-drivers/0.0.0/build/make.log for more information.
==> WARNING: `dkms install --no-depmod ipu6-drivers/0.0.0 -k 6.6.2-arch1-1' exited 10
==> dkms install --no-depmod v4l2loopback/r10.cae2954 -k 6.6.2-arch1-1
==> dkms install --no-depmod ipu6-drivers/r153.067270ff0 -k 6.6.2-arch1-1
==> dkms install --no-depmod xmm7360-pci-git/r231.107b6ba -k 6.6.2-arch1-1
Error! Bad return status for module build on kernel: 6.6.2-arch1-1 (x86_64)
Consult /var/lib/dkms/xmm7360-pci-git/r231.107b6ba/build/make.log for more information.
==> WARNING: `dkms install --no-depmod xmm7360-pci-git/r231.107b6ba -k 6.6.2-arch1-1' exited 10
==> depmod 6.6.2-arch1-1
(7/7) Updating linux initcpios...
```
and here is the log content
```
cat /var/lib/dkms/ipu6-drivers/0.0.0/build/make.log
DKMS make.log for ipu6-drivers-0.0.0 for kernel 6.6.2-arch1-1 (x86_64)
Tue Nov 21 02:20:10 PM CET 2023
make -C /usr/lib/modules/6.6.2-arch1-1/build M=/var/lib/dkms/ipu6-drivers/0.0.0/build modules
make[1]: Entering directory '/usr/lib/modules/6.6.2-arch1-1/build'
make[3]: *** No rule to make target '/var/lib/dkms/ipu6-drivers/0.0.0/build/drivers/mfd/ljca.o', needed by '/var/lib/dkms/ipu6-drivers/0.0.0/build/ljca.o'.  Stop.
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/lib/modules/6.6.2-arch1-1/build/Makefile:1913: /var/lib/dkms/ipu6-drivers/0.0.0/build] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.6.2-arch1-1/build'
make: *** [Makefile:78: all] Error 2
```

stefanpartheym, 2023-11-22:
Hey @elhossary,

I think, I had the same or a similar issue when I was upgrading to kernel `v6.6.1`.
Will try this on my machine in a few days.

iosonofabio, 2023-11-22:
same here, thanks for taking a look @stefanpartheym

stefanpartheym, 2023-11-24:
Hey @elhossary,

did you upgrade from kernel `v6.5.x` to `v6.6.2`?
I just upgraded from `v6.6.1` to `v6.6.2` and didn't experience any issues regarding the DKMS module.

elhossary, 2023-11-24:
@stefanpartheym no,
from 6.6.1 to 6.6.2

GorgetGuilhem, 2023-11-28:
Same error here upgrading from 6.6.1 to 6.6.2

elhossary, 2023-11-29:
Also, the issue remains when upgrading from 6.6.2 to 6.6.3

hinricht, 2023-11-30:
I also fail with `6.6.2-1-MANJARO`, see this 
[make.log](https://github.com/stefanpartheym/archlinux-ipu6-webcam/files/13512452/make.log) from installing `linux66-headers-6.6.2-1` (Manjaro sway edition). Is this related or should I create a seperate issue for this ?

GorgetGuilhem, 2023-12-11:
Hey, it works again with `6.6.5-arch1-1`.

elhossary, 2023-12-11:
@GorgetGuilhem Thanks, the issue still persists in 6.6.5 for my setup

GorgetGuilhem, 2023-12-11:
@elhossary sorry I updated my previous answer, I did uninstall everything and reinstalled again after kernel upgrade. It may be worth trying for you too !

sysmso, 2023-12-13:
works fine with `6.6.6-2-MANJARO` here. Before upgrading i uninstalled everything, then reinstalled with new kernel.

elhossary, 2024-01-17:
Disappeared in Kernel 6.7.0, but camera does not work though

stefanpartheym, 2024-01-21:
@elhossary Yep, same here. Kernel v6.7.0 breaks a lot of stuff on my Dell XPS 13 Plus 9320, not only the webcam. Had to downgrade to one of the v6.6.x kernel versions.
Uff, this really shit really starting to piss me off at some point... 
Guess we have to patiently wait for fix.

elhossary, 2024-01-21:
I am pissed too, that went too far with breaking things with each release, gotta have some better testing. What I also don't understand is, why it is not working out of the box if they already integrated the drivers in the Kernel.

elhossary, 2024-01-24:
@stefanpartheym 
I got a slightly different output with Kernel 6.7.1, but still does not work

stefanpartheym, 2024-01-24:
@elhossary Thanks for the update. I think I'll have a look at this next week or so.

elhossary, 2024-01-29:
@stefanpartheym 
Still, no luck in 6.7.2. Although, different errors!

elhossary, 2024-03-01:
Well, eventually it works, but the mentioned error are still showing.
I believe that's an issue outside this repo scope. So, yes should be closed

stefanpartheym, 2024-03-01:
Hmm, good point.
Even though, this might be an issue outside of the scope of this repo. I would leave it open, in case someone knows how to get rid of this errors.
Unfortunately, I can't reproduce it on my install though.

stefanpartheym, 2024-10-11:
Ok, closing this one now. :saluting_face:

closed 2024-10-11 by stefanpartheym