stefanpartheym/archlinux-ipu6-webcam #60: Camera is not working - multiple problems

vavra7, 2023-12-04:
Hello, sometime during the summer, I still had a working camera **thanks to this repository, so I am really grateful for your work**! Sometimes when I updated the operating system the camera stopped working, however it was always possible to solve it by running `sh ./uninstall.sh`, cloning the latest repository and running `sh ./install.sh` again. Once it stopped helping and because I'm lazy I decided to wait for the next update that will solve my problem, which still hasn't happened and that's why I decided to read through this repository and find solution for my problem.

So having cloned latest repository, installing and restarting the system:

<details>
  <summary>sh ./test.sh</summary>

  ```
 WARNING: erroneous pipeline: no element "icamerasrc"
  ```
</details>

Following the instructions in README.md I run sh `./uninstall.sh` then delete cache `rm -rf ~/.cache/yay/icamerasrc-git` and reinstall. After rebooting I run the test again.

<details>
  <summary>sh ./test.sh</summary>

  ```
 WARNING: erroneous pipeline: no element "icamerasrc"
  ```
</details>

I verify that the package exists

<details>
  <summary>pacman -Q icamerasrc-git</summary>

  ```
 icamerasrc-git r67.528a6f1-1
  ```
</details>

Just to be sure, I run installation of the package manually. 

<details>
  <summary>yay -S icamerasrc-git</summary>

  ```
AUR Explicit (1): icamerasrc-git-r57.2d36ade-1
:: PKGBUILD up to date, skipping download: icamerasrc-git
  1 icamerasrc-git                           (Installed) (Build Files Exist)
==> Packages to cleanBuild?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
  1 icamerasrc-git                           (Installed) (Build Files Exist)
==> Diffs to show?
==> [N]one [A]ll [Ab]ort [I]nstalled [No]tInstalled or (1 2 3, 1-3, ^4)
==> 
==> Making package: icamerasrc-git r57.2d36ade-1 (Mon 04 Dec 2023 10:29:28 AM CET)
==> Retrieving sources...
  -> Updating icamerasrc git repo...
  -> Found 70-ipu6-psys.rules
==> WARNING: Skipping verification of source file PGP signatures.
==> Validating source files with sha256sums...
    icamerasrc ... Skipped
    70-ipu6-psys.rules ... Passed
:: (1/1) Parsing SRCINFO: icamerasrc-git
==> Making package: icamerasrc-git r57.2d36ade-1 (Mon 04 Dec 2023 10:29:29 AM CET)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Updating icamerasrc git repo...
  -> Found 70-ipu6-psys.rules
==> Validating source files with sha256sums...
    icamerasrc ... Skipped
    70-ipu6-psys.rules ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Creating working copy of icamerasrc git repo...
Cloning into 'icamerasrc'...
[... 199 lines cut ...]
  <summary>neofetch</summary>

  ```
                   -`                    vavra@Arch-XPS 
                  .o+`                   -------------- 
                 `ooo/                   OS: Arch Linux x86_64 
                `+oooo:                  Host: XPS 9320 
               `+oooooo:                 Kernel: 6.6.3-arch1-1 
               -+oooooo+:                Uptime: 28 mins 
             `/:-:++oooo+:               Packages: 1329 (pacman) 
            `/++++/+++++++:              Shell: fish 3.6.1 
           `/++++++++++++++:             Resolution: 1920x1200 
          `/+++ooooooooooooo/`           DE: GNOME 45.2 
         ./ooosssso++osssssso+`          WM: Mutter 
        .oossssso-````/ossssss+`         WM Theme: Adwaita 
       -osssssso.      :ssssssso.        Theme: Adwaita [GTK2/3] 
      :osssssss/        osssso+++.       Icons: Adwaita [GTK2/3] 
     /ossssssss/        +ssssooo/-       Terminal: gnome-terminal 
   `/ossssso+/:-        -:/+osssso+-     CPU: 12th Gen Intel i7-1260P (16) @ 4.700GHz 
  `+sso+:-`                 `.-/+oso:    GPU: Intel Alder Lake-P GT2 [Iris Xe Graphics] 
 `++:.                           `-/+/   Memory: 4200MiB / 31706MiB 
 .`                                 `/

  ```
</details>

stefanpartheym, 2024-03-01:
Hi @vavra7,

specifically regarding the issue _WARNING: erroneous pipeline: no element "icamerasrc"_ (also mentioned in #74), I updated the section in the readme on how to fix this error.
Please consider trying the "new" approach.
Not sure, if this will fix your issue though. But always consider testing via `test.sh` **and** other applications like browsers (via the webrtc URL provided in the section _Chromium and Firefox-based browsers_).

stefanpartheym, 2024-05-24:
Closing due to inactivity

closed 2024-05-24 by stefanpartheym