stefanpartheym/archlinux-ipu6-webcam #65: Missing Dependency

thamma, 2023-12-21:
Hello folks,

I am building this projects from a relatively fresh Arch install on `6.6.7-arch1-1`. After executing `install.sh`, the installation aborted when building `v42l-relayd`, missing dependencies
```
gstreamer-1.0 >= 1.0.0
gstreamer-app-1.0 >= 1.0.0
gstreamer-video-1.0 >= 1.0.0
```

I previously installed `gstreamer`, unsure if it was because I ran into the same issue. However, upon installing (1) `gstreamer-plugins-base` or (2) `gstreamer-plugins-good`, the installation succeeded successfully.

I am not sure if I had to manually install `gstreamer` nor wether installing (1) or (1)+(2) fixed the issue, but you probably know which packages should be added to the dependencies of the PKGBUILD.

JanusDC, 2023-12-21:
Hi, check in `/etc/pacman.conf` if you have the [extra] repository enabled, since gstreamer is there.
On the other hand, I do not see that gstreamer-app or gstreamer-video are intalled on my system, and while streamer is installed, it does not seem to be a dependency for v4l2-relayed:

```
Name            : v4l2-relayd
Version         : r42.6fd6b6a-1
Description     : V4L2-relayd (Alder Lake)
Architecture    : x86_64
URL             : https://git.launchpad.net/v4l2-relayd
Licenses        : unknown
Groups          : None
Provides        : None
Depends On      : None
Optional Deps   : None
Required By     : None
Optional For    : None
Conflicts With  : None
Replaces        : None
Installed Size  : 19.37 KiB
Packager        : Unknown Packager
Build Date      : Wed 20 Dec 2023 03:07:05 PM CET
Install Date    : Wed 20 Dec 2023 03:07:19 PM CET
Install Reason  : Explicitly installed
Install Script  : No
Validated By    : None
```

d-passaro, 2023-12-27:
Solved for me by installing gstreamer-vaapi

thamma, 2023-12-28:
> Hi, check in `/etc/pacman.conf` if you have the [extra] repository enabled, since gstreamer is there. On the other hand, I do not see that gstreamer-app or gstreamer-video are intalled on my system, and while streamer is installed, it does not seem to be a dependency for v4l2-relayed: [...]

Hi! I had `extra` enabled and managed to install the required dependencies before opening this issue. I created this issue to raise awareness to the missing dependencies from the `v4l2-relayd` PKGBUILD.

alba4k, 2024-10-25:
this can probably be closed @thamma

stefanpartheym, 2024-11-17:
Closing this one.

closed 2024-11-17 by stefanpartheym