stefanpartheym/archlinux-ipu6-webcam #66: Pipeline could not be constructed: syntax error

thamma, 2023-12-21:
Hello,

I successfully ran through `install.sh` on a Lenovo ThinkPad X1 Yoga Gen8. `./test.sh` asks for sudo privileges and subsequently crashes with
```
(gst-launch-1.0:18145): GStreamer-CRITICAL **: 18:10:46.398: gst_element_link_pads_filtered: assertion 'GST_IS_BIN (parent)' failed
ERROR: pipeline could not be constructed: syntax error.
```

The same error was mentioned previously in [this MR](https://github.com/stefanpartheym/archlinux-ipu6-webcam/pull/22), but not discussed any further. Being not acquainted with gstreamer, I don't see what is causing a syntax error.

Happy to hear about any pointers

d-passaro, 2023-12-27:
I'm experiencing the same issue, thinkpad carbon x1 gen10 6.6.8-arch1-1

gfarrell, 2024-01-05:
Experiencing the same -- installation worked fine, but webcam does not (either with test.sh, cheese, or the webrtc test site).

* 6.6.8 kernel
* Dell XPS 9320

```
~/projects/archlinux-ipu6-webcam > ./test.sh                                    (main)

(gst-launch-1.0:6375): GStreamer-CRITICAL **: 17:30:03.409: gst_element_link_pads_filtered: assertion 'GST_IS_BIN (parent)' failed
ERROR: pipeline could not be constructed: syntax error.
```

```
~/projects/archlinux-ipu6-webcam > cheese -d "Virtual Camera"                   (main)

(cheese:5613): cheese-WARNING **: 17:28:43.924: Device '/dev/video1' does not support 2:0:0:0 colorimetry: ../gstreamer/subprojects/gst-plugins-good/sys/v4l2/gstv4l2object.c(4213): gst_v4l2_object_set_format_full (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin36/GstV4l2Src:v4l2src1:
Device wants 2:0:0:0 colorimetry
```

Test site just gives me a black screen

gfarrell, 2024-01-05:
IDK if this has anything to do with the fact that the `v4l2-relayd` service failed to start -- when I start it I don't get any errors anymore with Cheese, but still get a black screen.

oles, 2024-01-17:
Experiencing the same with Dell XPS 13 9315 here on 6.7.0-arch3-1

The install went fine, but the camera isn't working, and I get the same error from running the test:

```
(gst-launch-1.0:2729): GStreamer-CRITICAL **: 11:02:15.599: gst_element_link_pads_filtered: assertion 'GST_IS_BIN (parent)' failed
ERROR: pipeline could not be constructed: syntax error.
```

gfarrell, 2024-01-18:
Webcam is now working for me on 6.6.10 kernel but the test script and Cheese do not -- Zoom, Slack, Firefox (incl. the webrtc test site) all work.

i-oliva, 2024-01-20:
@gfarrell did you install any other specific package? I downgraded to 6.6.10 but I'm still getting the same error

```
[i-oliva@archlinux archlinux-ipu6-webcam]$ uname -r
6.6.10-arch1-1
[i-oliva@archlinux archlinux-ipu6-webcam]$ ./test.sh

(gst-launch-1.0:2396): GStreamer-CRITICAL **: 14:43:41.950: gst_element_link_pads_filtered: assertion 'GST_IS_BIN (parent)' failed
ERROR: pipeline could not be constructed: syntax error.
[i-oliva@archlinux archlinux-ipu6-webcam]$
```

gfarrell, 2024-01-23:
@i-oliva I did not -- the test script never worked for me. Now on 6.7.0 everything is broken again (and I can't keep the drivers installed because my laptop crashes when I close the lid if they are installed). Back to square 1.

david-imola, 2024-02-08:
Same issue on acer switft 3, kernel `6.7.4`

stefanpartheym, 2024-02-29:
As already mentioned in #69 kernel versions `6.7.x` are currently not supported. Please try using an LTS kernel.
However, this probably won't fix the error discussed in this issue.

closed 2024-10-25 by stefanpartheym