stefanpartheym/archlinux-ipu6-webcam #67: Dell Precision 5470 - camera led is always on when the laptop is plugged

allanger, 2024-01-02:
First, I've user this repo to setup my camera, and I need to confirm that it's working. And before I've tried enabling hibernation everything was fine. 

Then I've decided to start using hibernation, and reinstalled the driver with `-s` flag. And since then if the laptop charger is plugged, the camera led is also turned on. I've disabled hibernation, reinstalled drivers with different options, but nothing helps. The led is turned of once I close the lid, but if I re-plug, it's again on.

The only thing that helped was to uninstall drivers, but then obviously camera doesn't work. 

Maybe somebody has an idea how I can fix it?

alba4k, 2024-01-24:
do you have anything running that might be using the webcam (e.g. obs)?

maybe try switching to a different tty and stopping your display manager, so that there is as little running stuff as possible

allanger, 2024-01-25:
Unfortunately, I've switched to Fedora, so I can't test it anymore. 

I don't think I had something accessing the camera directly, because the LED was on already on the login screen, and it was only happening when it was charging, and it could be turned on only by closing and opening the lid. I've got a feeling that after I've played around with hibernation, I've done something that would enable the virtual device when the laptop is plugged in (though I'm not sure, it's even possible).

I think I can close it now, because I don't have this problem anymore (after distro switch)

closed 2024-01-25 by allanger