stefanpartheym/archlinux-ipu6-webcam #69: Camera does not work after upgrading to Kernel 6.7.0

elhossary, 2024-01-17:
Camera stopped working after upgrading to Kernel 6.7.0 in my X1C10
Uninstalled then re-installed it, and no issues during reinstallation.
test.sh output
the camera's led does not light up as well.
```
[muhoss@MuHoss-PC archlinux-ipu6-webcam]$ ./test.sh 
[sudo] password for muhoss: 
Setting pipeline to PAUSED ...
[01-17 15:06:04.757] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-17 15:06:04.757] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-17 15:06:04.758] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.758] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-17 15:06:04.758] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-17 15:06:04.758] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.758] CamHAL[INF] aiqb file name ov8856.aiqb
[01-17 15:06:04.758] CamHAL[INF] aiqb file name ov8856.aiqb
[01-17 15:06:04.759] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.759] CamHAL[INF] aiqb file name ov8856.aiqb
[01-17 15:06:04.759] CamHAL[INF] aiqb file name ov8856.aiqb
[01-17 15:06:04.759] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.759] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-17 15:06:04.759] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-17 15:06:04.760] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.760] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-17 15:06:04.760] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-17 15:06:04.761] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.761] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-17 15:06:04.761] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-17 15:06:04.761] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.761] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-17 15:06:04.761] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-17 15:06:04.761] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.761] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-17 15:06:04.761] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-17 15:06:04.762] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.762] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-17 15:06:04.762] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-17 15:06:04.762] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.762] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-17 15:06:04.762] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-17 15:06:04.762] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.762] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-17 15:06:04.762] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-17 15:06:04.763] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.763] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-17 15:06:04.763] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-17 15:06:04.763] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.763] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-17 15:06:04.763] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-17 15:06:04.764] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.764] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-17 15:06:04.764] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-17 15:06:04.764] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.764] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-17 15:06:04.764] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-17 15:06:04.764] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.764] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-17 15:06:04.764] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-17 15:06:04.764] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.764] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[01-17 15:06:04.765] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[01-17 15:06:04.765] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.765] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[01-17 15:06:04.765] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[01-17 15:06:04.765] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.765] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-17 15:06:04.765] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-17 15:06:04.765] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.765] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[01-17 15:06:04.765] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[... 48 lines cut ...]
[01-17 15:06:04.769] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.769] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-17 15:06:04.770] CamHAL[ERR] invalid media format, default value used.
[01-17 15:06:04.770] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.770] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.770] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-17 15:06:04.770] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
Failed to set pipeline to PAUSED.
Setting pipeline to NULL ...
Freeing pipeline ...
```
Any help?

neilking01, 2024-01-18:
same issue for me.

renatocan, 2024-01-20:
Same issue for me.

rclovis, 2024-01-21:
Same issue on my dell xps 13 9315

sreedevk, 2024-01-21:
same issue on Dell Precision 5470. slightly different output when running `test.sh` (added below)

Kernel: Linux 6.7.0-arch3-1

```
[01-21 14:03:05.321] CamHAL[ERR] Malformed ET range in exposure time range configuration
[01-21 14:03:05.321] CamHAL[ERR] Parse AE eExposure time range failed
Setting pipeline to PAUSED ...
[01-21 14:03:05.915] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-21 14:03:05.915] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-21 14:03:05.916] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.916] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-21 14:03:05.916] CamHAL[INF] aiqb file name ov13b10.aiqb
[01-21 14:03:05.916] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.916] CamHAL[INF] aiqb file name ov8856.aiqb
[01-21 14:03:05.916] CamHAL[INF] aiqb file name ov8856.aiqb
[01-21 14:03:05.917] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.917] CamHAL[INF] aiqb file name ov8856.aiqb
[01-21 14:03:05.917] CamHAL[INF] aiqb file name ov8856.aiqb
[01-21 14:03:05.917] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.917] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 14:03:05.917] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 14:03:05.917] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.917] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 14:03:05.917] CamHAL[INF] aiqb file name ov01a10.aiqb
[01-21 14:03:05.918] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 14:03:05.918] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[01-21 14:03:05.918] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[01-21 14:03:05.918] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.918] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-21 14:03:05.918] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-21 14:03:05.919] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-21 14:03:05.919] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[01-21 14:03:05.919] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[01-21 14:03:05.919] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[01-21 14:03:05.919] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[01-21 14:03:05.919] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[01-21 14:03:05.919] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.920] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-21 14:03:05.920] CamHAL[INF] aiqb file name ov01a1s.aiqb
[01-21 14:03:05.920] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.920] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[01-21 14:03:05.920] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[... 62 lines cut ...]
[01-21 14:03:05.922] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
[01-21 14:03:05.922] CamHAL[ERR] invalid media format, default value used.
[01-21 14:03:05.922] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.922] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.922] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.922] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[01-21 14:03:05.923] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
[... cut ...]
Caught SIGSEGV
#0  0x00007fcc24139f6f in poll () at /usr/lib/libc.so.6
#1  0x00007fcc243332b6 in  () at /usr/lib/libglib-2.0.so.0
#2  0x00007fcc242d5b97 in g_main_loop_run () at /usr/lib/libglib-2.0.so.0
#3  0x00005db5f6dadc15 in  ()
#4  0x00007fcc2405ecd0 in  () at /usr/lib/libc.so.6
#5  0x00007fcc2405ed8a in __libc_start_main () at /usr/lib/libc.so.6
#6  0x00005db5f6dab055 in  ()
Spinning.  Please run 'gdb gst-launch-1.0 12284' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

gfarrell, 2024-01-22:
Same here -- I get the following in the kernel buffer when I try to use the webcam:

```
[  540.357065] camerasrc0:src[11829]: segfault at 118 ip 00007ebbf2877edd sp 00007ebbebdff608 error 4 in libc.so.6[7ebbf2743000+15a000] likely on CPU 4 (core 8, socket 0)
[  540.357078] Code: 00 00 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 89 f8 48 89 fa c5 f9 ef c0 25 ff 0f 00 00 3d e0 0f 00 00 0f 87 33 01 00 00 <c5> fd 74 0f c5 fd d7 c1 85 c0 74 57 f3 0f bc c0 c5 f8 77 c3 66 66
```

gfarrell, 2024-01-23:
And, would note that now sending the laptop to sleep (e.g. closing the lid) crashes it so I've uninstalled the drivers.

daveemu, 2024-01-23:
> And, would note that now sending the laptop to sleep (e.g. closing the lid) crashes it so I've uninstalled the drivers.

Can confirm this issue. Laptop hangs on sleep and reboot. Had do go back to kernel 6.6.10.

elhossary, 2024-01-23:
@gfarrell @daveemu 
It was always crashing with sleep/hibernate. This disappeared after upgrading to 6.7.0. Though, not because it was fixed, but because the driver ain't functional anymore.

hecrogon, 2024-01-23:
Hello @elhossary , I detected the same sleeping issue with the kernel 6.7.0 + archlinux-ipu6-webcam. Actually I noticed that problem before realizing the webcam wasn't working anymore. After downgrading to .6.6.10 too, everything works (sleep and webcam)

elhossary, 2024-01-24:
I got a slightly different output with Kernel 6.7.1, but still does not work

renatocan, 2024-01-26:
> Hello @elhossary , I detected the same sleeping issue with the kernel 6.7.0 + archlinux-ipu6-webcam. Actually I noticed that problem before realizing the webcam wasn't working anymore. After downgrading to .6.6.10 too, everything works (sleep and webcam). I have a DELL XPS 13 9315

@hecrogon How have you downgraded? I tried downgrading the "linux" and "linux-headers" packages to 6.6.10 and 6.6.8 but the camera still did not work, even when I uninstalled and reinstalled the archlinux-ipu6-webcam script. Thanks!

neilking01, 2024-01-26:
Try 6.5.9. It's working for me on that kernel.

hecrogon, 2024-01-26:
Hey @renatocan,
Probably you are having the same issue I had. After downgrading the `linux` and `linux-headers` packages, the `archlinux-ipu6-webcam` `install.sh` automatically install the last `linux-headers` available (no matter if you added to the IgnorePkg.

I did the follow dirty trick, just comment the line 125
https://github.com/stefanpartheym/archlinux-ipu6-webcam/blob/main/install.sh#L125

from
```
build_and_install "${kernel}-headers"
```
to
```
# build_and_install "${kernel}-headers"
```

gfarrell, 2024-01-26:
If you want to rollback arch you should really set the repo to an archive
URL so that your archlinux system is consistent (otherwise you might break
stuff if you only pin a few things). If you do this, then the install
script will only know about the pinned version of the headers anyway. I
kept docs for this (for myself) here:
https://gtf21.notion.site/Package-Management-b683024fee6944088db5517ecbf4d9c2#69069a47e1fa44818d183113410d6735
--
*Gideon Farrell*
Co-founder / CTO

www.converge.io

renatocan, 2024-01-27:
> Hey @renatocan, Probably you are having the same issue I had. After downgrading the `linux` and `linux-headers` packages, the `archlinux-ipu6-webcam` `install.sh` automatically install the last `linux-headers` available (no matter if you added to the IgnorePkg.
> 
> I did the follow dirty trick, just comment the line 125 before running the `install.sh` https://github.com/stefanpartheym/archlinux-ipu6-webcam/blob/main/install.sh#L125
> 
> from
> 
> ```
> build_and_install "${kernel}-headers"
> ```
> 
> to
> 
> ```
> # build_and_install "${kernel}-headers"
> ```

Thanks! This was the problem.

renatocan, 2024-01-27:
> If you want to rollback arch you should really set the repo to an archive URL so that your archlinux system is consistent (otherwise you might break stuff if you only pin a few things). If you do [...]

Thanks for the tip.

neilking01, 2024-01-27:
> Thanks! This was the problem.

is this the solution for newer kernels or just for downgrading?

renatocan, 2024-01-27:
> > Thanks! This was the problem.
> 
> is this the solution for newer kernels or just for downgrading?

If we downgrade and install the script, it re-updates linux-headers. I downgraded linux-headers after installing the script and the camera worked.

fbrinker, 2024-01-30:
Slightly offtopic, but required for the webcam to work: If I want to downgrade my entire system, any idea how to properly downgrade dbus after I've installed dbus-broker-units? I get file conflicts in /usr/lib/systemd/system/dbus.service, even when replacing broker-units with daemon-units before the downgrade.

Is there any fix for 6.7 kernels on the horizon?

elhossary, 2024-01-30:
Yes, that's what I eventually did.
Recently, Kernel 6.6 went LTS officially, I switched to it (6.6.14), and everything just works straightforward.
@stefanpartheym 
Maybe using lts kernel should be announced in the README

gfarrell, 2024-01-30:
@elhossary ah fantastic -- I've switched to 6.6.14-lts and now I have a webcam again :sparkling_heart: great tip

fbrinker, 2024-01-30:
Good idea. Instead of downgrading the entire system, I've installed linux-lts in parallel and added a default boot entry for it.
Now people have to see my face again in meetings. :laughing:

arcan1s, 2024-01-31:
> Is there any fix for 6.7 kernels on the horizon?
I guess it has to be addressed to intel instead. Didn't really see bug reports there though except for kernel panic https://github.com/intel/ipu6-drivers/issues/205

i-oliva, 2024-01-31:
I downgraded to 6.6.14-lts and still having the same issue

elhossary, 2024-01-31:
It works with many people, so it is unlike in kernel 6.7 that doesn't work with everyone, this means you probably have an individual issue.
Try to fresh clone, uninstall, reboot, install, reboot, then test

i-oliva, 2024-01-31:
Yep, did that again, no luck. I will try removing `pacman` cache since it is installing packages with pacman.

i-oliva, 2024-01-31:
The only "weird" thing I see is an error in-between in the `./uninstall` script, but it doesn't stop the script from getting executed.

```
:: Do you want to remove these packages? [Y/n] 
:: Running pre-transaction hooks...
(1/1) Remove DKMS modules
==> dkms remove --no-depmod ipu6-drivers/r156.07f0612ea -k 6.6.14-1-lts
==> depmod 6.6.14-1-lts
:: Processing package changes...
(1/2) removing intel-ipu6-dkms-git                                                                                                           [--------------------------------------------------------------------------------------] 100%
(2/2) removing intel-ivsc-firmware                                                                                                           [--------------------------------------------------------------------------------------] 100%
:: Running post-transaction hooks...
(1/2) Arming ConditionNeedsUpdate...
(2/2) Refreshing py3status arch_updates module...
error: target not found: intel-ivsc-firmware
 -> exit status 1
checking dependencies...

Package (1)  Old Version    Net Change

v4l2-relayd  r42.6fd6b6a-1   -0.02 MiB

Total Removed Size:  0.02 MiB

:: Do you want to remove these packages? [Y/n] 
:: Processing package changes...
(1/1) removing v4l2-relayd                                                                                                                   [--------------------------------------------------------------------------------------] 100%
:: Running post-transaction hooks...
(1/3) Reloading system manager configuration...
(2/3) Arming ConditionNeedsUpdate...
(3/3) Refreshing py3status arch_updates module...
checking dependencies...

Package (2)                Old Version    Net Change

dkms                       3.0.12-1        -0.11 MiB
v4l2loopback-dkms-git-fix  r10.cae2954-1   -0.23 MiB
```

daveemu, 2024-02-01:
> Yep, did that again, no luck. I will try removing `pacman` cache since it is installing packages with pacman.
> 
> PS: actually `yay -Scc` does that already. Not sure how to debug what's wrong 😞
> 
> I understand that many people got it working, it would be nice to know if they did other thing other than downgrading to lts, because when you downgrade you do actually keep the previous headers installed in the pacman cache which are then used in the ./install script, which requires to be removed, so they must have done something other than just installing it.

I got it to work by downgrading the kernel and kernel-header to 6.6.* (no LTS). I used AUR package "downgrade" for this. Afterwards (after a restart) I had to rebuild DKMS modules. However, a few days later i switched to LTS kernel, to be save for future upgrades.

aplund, 2024-02-02:
I'm not sure if this is related, but after updating I found that the icamerasrc plugin is blacklisted.  It's unclear to me why, or how to fix it.

```
$ gst-inspect-1.0 libgsticamerasrc.so
Plugin Details:
  Name                     libgsticamerasrc.so
  Description              Plugin for blacklisted file
  Filename                 /usr/lib/gstreamer-1.0/libgsticamerasrc.so
  Version                  0.0.0
  License                  BLACKLIST
  Source module            BLACKLIST
  Binary package           BLACKLIST
  Origin URL               BLACKLIST

  0 features:
```

aplund, 2024-02-02:
To answer my own question, the blacklisting is due to the plugin raising a SIGSEGV. I managed to get this backtrace, and the culprit seems to be the 'sem' variable being 0x0.

```
#0  __new_sem_wait_fast (definitive_result=0, sem=0x0)
    at /home/lund/.cache/debuginfod_client/1bc67512a1f89c5a034f03e066cbd11fc7905cb4/source##usr##src##debug##glibc##glibc##nptl##sem_waitcommon.c:136
#1  ___sem_timedwait64 (sem=0x0, abstime=0x7ffd0ef92da0)
    at /home/lund/.cache/debuginfod_client/1bc67512a1f89c5a034f03e066cbd11fc7905cb4/source##usr##src##debug##glibc##glibc##nptl##sem_timedwait.c:37
#2  0x00007f8c4f92db14 in icamera::CameraSharedMemory::lock (this=0x5a1e44b243c0)
    at /home/lund/src/Upstream/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/src/iutils/CameraShm.cpp:263
#3  0x00007f8c4f92ddc4 in icamera::CameraSharedMemory::acquireSharedMemory (this=0x5a1e44b243c0)
    at /home/lund/src/Upstream/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/src/iutils/CameraShm.cpp:92
#4  0x00007f8c4f92e073 in icamera::CameraSharedMemory::CameraSharedMemory (this=this@entry=0x5a1e44b243c0)
    at /home/lund/src/Upstream/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/src/iutils/CameraShm.cpp:41
#5  0x00007f8c4f9232fc in icamera::CameraHal::CameraHal (this=this@entry=0x5a1e44b23ea0)
    at /home/lund/src/Upstream/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/src/hal/CameraHal.cpp:45
#6  0x00007f8c4f89918c in icamera::initCameraHAL ()
    at /home/lund/src/Upstream/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/src/hal/ICamera.cpp:340
```

aplund, 2024-02-04:
I have done a bisection of the kernel code for this issue.

```
34a674e99acd6ec4b541cd4c630e126ba1a4c22f is the first bad commit
commit 34a674e99acd6ec4b541cd4c630e126ba1a4c22f
Author: Vitaly Lubart <vitaly.lubart@intel.com>
Date:   Sun Oct 15 11:05:40 2023 +0300

    mei: me: emit error only if reset was unexpected

    GSC devices perform legal firmware initiated resets due to state transition
    that may appear as unexpected to the driver. Lower the log level for those
    devices to debug level and save the firmware status registers.
    When the device comes out of the reset it is possible to check whether the
    resets was due to a firmware error or an exception
    and only than produce a warning.

    Signed-off-by: Vitaly Lubart <vitaly.lubart@intel.com>
    Signed-off-by: Tomas Winkler <tomas.winkler@intel.com>
    Link: https://lore.kernel.org/r/20231015080540.95922-1-tomas.winkler@intel.com
    Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>

 drivers/misc/mei/hw-me-regs.h |  3 +++
 drivers/misc/mei/hw-me.c      | 61 ++++++++++++++++++++++++++++++++++++++++---
 drivers/misc/mei/init.c       | 27 +++++++++++++++++--
 drivers/misc/mei/mei_dev.h    | 47 +++++++++++++++++++++++++++++++++
 4 files changed, 133 insertions(+), 5 deletions(-)
```

fbrinker, 2024-02-05:
Interesting work. Did you forward this to the official firmware project we rely on as well?

aplund, 2024-02-05:
> Interesting work. Did you forward this to the official firmware project we rely on as well?

No, but that's only because I'm not actually sure who this bug belongs to.  Is it the kernel?  Is it the v4l2loopback or the intel modules?  Or is the icamerasrc gst plugin?  My test for bisection was: does the ./test.sh script run.

benlodz, 2024-02-17:
I was getting this exact same issue with the plugin being blacklisted. I was able to get it working though kind of haphazardly. Here's exactly what I did:

- Install the lts kernel and it's headers
- ran the install script and rebooted
- no dice ( got the issue involving the plugin)
- ran the uninstall script and then the reinstall script
- rebooted and it works! 

Not really sure what changed exactly, just kind of winged it.

stefanpartheym, 2024-02-29:
Added a note in the readme to recommend using a LTS kernel.

alba4k, 2024-03-25:
has this changed?

elhossary, 2024-03-25:
I have not tested it in kernel 6.8

stefanpartheym, 2024-03-26:
I have not tested it either in 6.8.x. Will have a look on the weekend probably.

aplund, 2024-03-27:
I tried to install on 6.8.1, but `icamerasrc-git` doesn't compile for me now:

(this is most likely unrelated to this bug)

```
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -std=c++11 -Wall -Werror -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-6 -pthread -Wall -I/usr/include/libcamhal -I/usr/include/libcamhal/api -I/usr/include/libcamhal/utils -DCHROME_SLIM_CAMHAL -fstack-protector-all -fPIC -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/home/lund/.cache/paru/clone/icamerasrc-git/src=/usr/src/debug/icamerasrc-git -flto=auto -MT libgsticamerainterface_1.0_la-gstcameraispinterface.lo -MD -MP -MF .deps/libgsticamerainterface_1.0_la-gstcameraispinterface.Tpo -c gstcameraispinterface.cpp  -fPIC -DPIC -o .libs/libgsticamerainterface_1.0_la-gstcameraispinterface.o
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -std=c++11 -Wall -Werror -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/sysprof-6 -pthread -Wall -I/usr/include/libcamhal -I/usr/include/libcamhal/api -I/usr/include/libcamhal/utils -DCHROME_SLIM_CAMHAL -fstack-protector-all -fPIC -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -Wp,-D_GLIBCXX_ASSERTIONS -g -ffile-prefix-map=/home/lund/.cache/paru/clone/icamerasrc-git/src=/usr/src/debug/icamerasrc-git -flto=auto -MT libgsticamerainterface_1.0_la-gstcameradewarpinginterface.lo -MD -MP -MF .deps/libgsticamerainterface_1.0_la-gstcameradewarpinginterface.Tpo -c gstcameradewarpinginterface.cpp  -fPIC -DPIC -o .libs/libgsticamerainterface_1.0_la-gstcameradewarpinginterface.o
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1plus: all warnings being treated as errors
make[3]: *** [Makefile:567: libgsticamerainterface_1.0_la-gstcamerawfovinterface.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1plus: all warnings being treated as errors
make[3]: *** [Makefile:546: libgsticamerainterface_1.0_la-gstcamera3ainterface.lo] Error 1
cc1plus: all warnings being treated as errors
make[3]: *** [Makefile:553: libgsticamerainterface_1.0_la-gstcameraispinterface.lo] Error 1
cc1plus: all warnings being treated as errors
make[3]: *** [Makefile:560: libgsticamerainterface_1.0_la-gstcameradewarpinginterface.lo] Error 1
make[3]: Leaving directory '/home/lund/.cache/paru/clone/icamerasrc-git/src/icamerasrc/src/interfaces'
make[2]: *** [Makefile:697: all-recursive] Error 1
make[2]: Leaving directory '/home/lund/.cache/paru/clone/icamerasrc-git/src/icamerasrc/src'
make[1]: *** [Makefile:526: all-recursive] Error 1
make[1]: Leaving directory '/home/lund/.cache/paru/clone/icamerasrc-git/src/icamerasrc'
make: *** [Makefile:435: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
error: failed to build 'icamerasrc-git-r57.2d36ade-1':
error: packages failed to build: icamerasrc-git-r57.2d36ade-1
error: package 'icamerasrc-git' was not found
ERROR: Couldn't find/unable to install: icamerasrc-git
```

arcan1s, 2024-03-28:
there is an opened mr [here](https://github.com/intel/ipu6-drivers/pull/213), personally I didn't test it, but it might help with the issue

DutchEllie, 2024-04-04:
I am still getting a lot of those segfault issues. I have tried everything so far. I was using the LTS kernel, then I downgraded my normal kernel to 6.5.9 which resulted in the same issues, and then to 6.4.9 which also has again the same issues. 

Dmesg:
```
[  287.505904] camerasrc0:src[3705]: segfault at 60 ip 00007f5352e9eb25 sp 00007f5343ffeb30 error 4 in libgstreamer-1.0.so.0.2401.0[7f5352e85000+c1000] likely on CPU 9 (core 16, socket 0)
[  287.505914] Code: 00 00 00 48 85 c0 74 0d 48 89 c7 e9 65 d7 03 00 90 0f 1f 40 00 c3 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 f3 0f 1e fa 53 <48> 8b 47 60 48 89 fb 48 8d 78 10 ff 15 7a c7 10 00 48 8b 7b 60 8b
[  290.070280] camerasrc0:src[5796]: segfault at 60 ip 00007f9161176b25 sp 00007f915ddadb30 error 4 in libgstreamer-1.0.so.0.2401.0[7f916115d000+c1000] likely on CPU 9 (core 16, socket 0)
[  290.070304] Code: 00 00 00 48 85 c0 74 0d 48 89 c7 e9 65 d7 03 00 90 0f 1f 40 00 c3 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 f3 0f 1e fa 53 <48> 8b 47 60 48 89 fb 48 8d 78 10 ff 15 7a c7 10 00 48 8b 7b 60 8b
[  292.509834] camerasrc0:src[5833]: segfault at 60 ip 00007f199206db25 sp 00007f198ac06b30 error 4 in libgstreamer-1.0.so.0.2401.0[7f1992054000+c1000] likely on CPU 4 (core 8, socket 0)
[  292.509846] Code: 00 00 00 48 85 c0 74 0d 48 89 c7 e9 65 d7 03 00 90 0f 1f 40 00 c3 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 f3 0f 1e fa 53 <48> 8b 47 60 48 89 fb 48 8d 78 10 ff 15 7a c7 10 00 48 8b 7b 60 8b
[  304.533870] gst-plugin-scan[5920]: segfault at 0 ip 00007f63abd932bf sp 00007ffec13eba20 error 4 in libc.so.6[7f63abd24000+15b000] likely on CPU 4 (core 8, socket 0)
[  304.533879] Code: 66 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 4a 55 48 89 f5 53 48 89 fb 48 83 ec 08 e8 01 f4 ff ff <48> 8b 03 85 c0 74 1a 48 8d 50 ff f0 48 0f b1 13 75 0f 48 83 c4 08
[  304.723767] gst-plugin-scan[5928]: segfault at 0 ip 00007fce735902bf sp 00007ffd28b86ec0 error 4 in libc.so.6[7fce73521000+15b000] likely on CPU 6 (core 12, socket 0)
[  304.723781] Code: 66 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 4a 55 48 89 f5 53 48 89 fb 48 83 ec 08 e8 01 f4 ff ff <48> 8b 03 85 c0 74 1a 48 8d 50 ff f0 48 0f b1 13 75 0f 48 83 c4 08
```

test.sh

```
$ ./test.sh

(gst-plugin-scanner:14850): GStreamer-WARNING **: 12:57:08.959: static caps 0x7f2ceffb7070 string is NULL

(gst-plugin-scanner:14850): GStreamer-CRITICAL **: 12:57:08.960: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(gst-launch-1.0:14849): GStreamer-WARNING **: 12:57:08.993: static caps 0x7f1dc9805070 string is NULL

(gst-launch-1.0:14849): GStreamer-CRITICAL **: 12:57:08.994: gst_mini_object_unref: assertion 'mini_object != NULL' failed

(gst-launch-1.0:14849): GStreamer-CRITICAL **: 12:57:08.994: gst_caps_can_intersect: assertion 'GST_IS_CAPS (caps2)' failed
WARNING: erroneous pipeline: could not link camerasrc0 to videoconvert0, camerasrc0 can't handle caps video/x-raw, format=(string)NV12, width=(int)1280, height=(int)720
```

alexhulbert, 2024-04-04:
I'm also seeing the same issues. I'm on kernel 6.9.0-rc1 with a Dell XPS 9640, which has the new meteor lake chip on it.

DutchEllie, 2024-04-04:
I'm going to be super honest here, but I am too drunk to check right now, but I do know I have a ThinkPad X1 Carbon Gen 11 with that OLED 2.5k or something screen. And even more embarrassing is that even while sober I've never figured out what the sensor model number in my device is. Honestly all I want is just a working webcam. How hard can that be? Why does Intel need to make it so difficult?  ---- On Thu, 04 Apr 2024 21:10:52 +0200  ***@***.***  wrote ----
I'm also seeing the same issues as DutchEllie. I'm on kernel 6.9.0-rc1 with a Dell XPS 9640, which has the new meteor lake chip on it.

madsl, 2024-04-09:
> I have done a bisection of the kernel code for this issue.
>
> ```
> 34a674e99acd6ec4b541cd4c630e126ba1a4c22f is the first bad commit
> commit 34a674e99acd6ec4b541cd4c630e126ba1a4c22f
> Author: Vitaly Lubart [...]

Thank you for this. With https://github.com/intel/ipu6-drivers/pull/213, https://github.com/intel/ipu6-drivers/pull/214 and the reverting of the commit you mention here, I have camera working again on my Dell XPS 9320.

stefanpartheym, 2024-10-25:
Will close this one now. It probably won't ever work on kernel v6.7.x to v6.9.x.

closed 2024-10-25 by stefanpartheym