stefanpartheym/archlinux-ipu6-webcam #70: no output on dell xps

quentinalbertone, 2024-01-29:
Hello everyone,

I just made a fresh endeavourOS install on my dell xps 9320. The `install.sh -as` seems to have no errors but the `test.sh` doesn't work

<details>
  <summary>./test.sh</summary>

  ```Bash
  [sudo] password for my_user: 
  [01-29 10:16:38.169] CamHAL[ERR] Malformed ET range in exposure time range configuration
  [01-29 10:16:38.169] CamHAL[ERR] Parse AE eExposure time range failed
  [01-29 10:16:38.169] CamHAL[ERR] Malformed ET range in exposure time range configuration
  [01-29 10:16:38.169] CamHAL[ERR] Parse AE gain range failed
  [...]
  [01-29 10:16:38.174] CamHAL[ERR] Malformed ET range in exposure time range configuration
  [01-29 10:16:38.174] CamHAL[ERR] Parse AE gain range failed
  Setting pipeline to PAUSED ...
  [01-29 10:16:38.844] CamHAL[INF] aiqb file name ov13b10.aiqb
  [01-29 10:16:38.844] CamHAL[INF] aiqb file name ov13b10.aiqb
  [01-29 10:16:38.845] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.845] CamHAL[INF] aiqb file name ov13b10.aiqb
  [01-29 10:16:38.845] CamHAL[INF] aiqb file name ov13b10.aiqb
  [01-29 10:16:38.845] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.845] CamHAL[INF] aiqb file name ov8856.aiqb
  [01-29 10:16:38.846] CamHAL[INF] aiqb file name ov8856.aiqb
  [01-29 10:16:38.846] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.846] CamHAL[INF] aiqb file name ov8856.aiqb
  [01-29 10:16:38.846] CamHAL[INF] aiqb file name ov8856.aiqb
  [01-29 10:16:38.846] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.846] CamHAL[INF] aiqb file name ov01a10.aiqb
  [01-29 10:16:38.846] CamHAL[INF] aiqb file name ov01a10.aiqb
  [01-29 10:16:38.847] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.847] CamHAL[INF] aiqb file name ov01a10.aiqb
  [01-29 10:16:38.847] CamHAL[INF] aiqb file name ov01a10.aiqb
  [01-29 10:16:38.847] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.847] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
  [01-29 10:16:38.847] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
  [01-29 10:16:38.848] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.848] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
  [01-29 10:16:38.848] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
  [01-29 10:16:38.848] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.848] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
  [01-29 10:16:38.848] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
  [01-29 10:16:38.848] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.848] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
  [01-29 10:16:38.848] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
  [01-29 10:16:38.848] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.848] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
  [01-29 10:16:38.849] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
  [01-29 10:16:38.849] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.849] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
  [01-29 10:16:38.849] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
  [01-29 10:16:38.849] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.849] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
  [01-29 10:16:38.849] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
  [01-29 10:16:38.849] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.849] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
  [01-29 10:16:38.849] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
  [01-29 10:16:38.850] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[... 70 lines cut ...]
  [01-29 10:16:38.855] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
  [01-29 10:16:38.855] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
  [01-29 10:16:38.855] CamHAL[ERR] invalid media format, default value used.
  [01-29 10:16:38.855] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.855] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.855] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.855] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
  [01-29 10:16:38.855] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
  [01-29 10:16:38.858] CamHAL[ERR] MediaControl init failed
  Failed to set pipeline to PAUSED.
  Setting pipeline to NULL ...
  Freeing pipeline ...
  [01-29 10:16:39.90] CamHAL[WAR] Seems camera device 0 is not closed properly (pid 2569).
  ```
</details>

I'm on ondeavourOS
```Bash
$> uname -a
Linux my_computer 6.7.1-arch1-1 #1 SMP PREEMPT_DYNAMIC Sun, 21 Jan 2024 22:14:10 +0000 x86_64 GNU/Linux
```

When I launch the [website to test](https://webrtc.github.io/samples/src/content/devices/input-output/) on firefox I can see the `virtual camera` in the settings but I have no output

I hope you can help me

arcan1s, 2024-01-29:
the camera doesn't work with 6.7.1 kernels #69

stefanpartheym, 2024-02-29:
Will close this one for now

closed 2024-02-29 by stefanpartheym