stefanpartheym/archlinux-ipu6-webcam #74: WARNING: erroneous pipeline: no element "icamerasrc"

jjeziorny, 2024-02-29:
Just moved from 6.7 to lts kernel in order to have the camera working again, after fresh install I keep getting this error message when testing the camera
`WARNING: erroneous pipeline: no element "icamerasrc"`

Even after removing `icamerasrc-git` from yum cache.

Any other sugestions?

`Linux jj 6.6.18-1-lts #1 SMP PREEMPT_DYNAMIC Fri, 23 Feb 2024 14:58:40 +0000 x86_64 GNU/Linux`

stefanpartheym, 2024-03-01:
> ...
> Even after removing `icamerasrc-git` from yum cache.
> ...

Weird... Have you reinstalled the `icamerasrc -git` package after removing it from the cache?

Also maybe try this:
```sh
yay \
  --cleanmenu=false \
  --diffmenu=false \
  --editmenu=false \
  --rebuild \
  -S icamerasrc-git
```

It's basically the same as it is described in the readme, but you might want to give it a try anyways.

jjeziorny, 2024-03-01:
> > ...
> > Even after removing `icamerasrc-git` from yum cache.
> > ...
> 
> Weird... Have you reinstalled the `icamerasrc -git` package after removing it from the cache?
> 
> Also maybe try this:
> 
> ```shell
> yay \
>   --cleanmenu=false \
>   --diffmenu=false \
>   --editmenu=false \
>   --rebuild \
>   -S icamerasrc-git
> ```
> 
> It's basically the same as it is described in the readme, but you might want to give it a try anyways.

Yes, I had tried to remove it manually after cleaning up the yum cache.

I've tried the suggested yay command now and I'm happy to say it did fix the issue.

jjeziorny, 2024-03-01:
Fixed with
```
yay \
  --cleanmenu=false \
  --diffmenu=false \
  --editmenu=false \
  --rebuild \
  -S icamerasrc-git
```

closed 2024-03-01 by jjeziorny

stefanpartheym, 2024-03-01:
Updated section _Solving error WARNING: erroneous pipeline: no element "icamerasrc"_ in the readme accordingly.