stefanpartheym/archlinux-ipu6-webcam #75: Is it possible to modify the resolution

elhossary, 2024-02-29:
My Camera supports FHD resolution, but the default running resolution is 1280*720.

Is it possible to change some configuration to make "1920*1080" the default output resolution?

stefanpartheym, 2024-03-01:
Hi @elhossary,

are you referring to general output resolution (i.e. resolution also in Chromium, Firefox, etc.)? Or are you referring to the output resolution from the test script?

elhossary, 2024-03-01:
@stefanpartheym I see the resolution mentioned in the test script, can I make it permanent config for the cam output which Zoom for example will use

stefanpartheym, 2024-03-01:
Hi @elhossary,
sorry to disappoint you, I have no clue, if that's possible. I guess, however, that it depends on the application accessing the camera. But as already mentioned, I have no clue, this is just a guess.
The only thing I can say for sure is, you can specify the output resolution in the test script, which will not influence the output resolution used in other applications.

elhossary, 2024-03-01:
Okay, thanks

closed 2024-03-01 by elhossary