stefanpartheym/archlinux-ipu6-webcam #77: Error while building intel-ipu6ep-camera-hal-git (Dell XPS 13 9315)

kotturi-mahipal, 2024-03-19:
[image]

treiher, 2024-03-22:
I have the same problem with the current `main` branch and an up-to-date Arch Linux:

```
==> Starting build()...
CMake Deprecation Warning at CMakeLists.txt:17 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- install prefix: /usr
INFOadd definition -DIPU_SYSVER_ipu6v5 for ipu6ep
Install camera configure files foripu6ep
-- Configuring done (0.0s)
-- Generating done (0.0s)
-- Build files have been written to: /home/tr/admin/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/build
[  1%] Building C object CMakeFiles/camhal_static.dir/modules/ia_css/ipu6ep/src/ia_css_terminal.c.o
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/camhal_static.dir/build.make:76: CMakeFiles/camhal_static.dir/modules/ia_css/ipu6ep/src/ia_css_terminal.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:356: CMakeFiles/camhal_static.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
ERROR: Failed to build/install: intel-ipu6ep-camera-hal-git
```

Milihhard, 2024-03-25:
I have the same error but on the `icamerasrc` installation.

kotturi-mahipal, 2024-03-28:
Hope someone solves this quickly.

Heikmike, 2024-03-31:
Same problem on Arch Linux

Tropicao, 2024-04-01:
Hello. I stumbled upon this issue too, on both `intel-ipu6ep-camera-hal-git` and `icamerasrc`.
This kind of dirty hotfix allowed me to go through the install:
* for `intel-ipu6ep-camera-hal-git`, edit `intel-ipu6ep-camera-hal-git/PKGBUILD` and add the following method:
```sh
prepare() {
    cd $_pkgname
    sed -i "s|-D_FORTIFY_SOURCE=2| |" CMakeLists.txt
}
```
then replay `install.sh`
* for `imcamerasrc`, the trick is similar but on different files, and since the AUR package is directly used, you have to call your package manager with relevant options to allow editing the PKGBUILD during install. For me it was:
  * `yay  -S --editmenu icamerasrc`
  * then add a similar method in PKGBUILD:
```sh
prepare() {
    cd $_pkgname
    sed -i "s|-D_FORTIFY_SOURCE=2| |" src/Makefile.am
    sed -i "s|-D_FORTIFY_SOURCE=2| |" src/interfaces/Makefile.am
}
```

From there you can again replay `install.sh` which should finish installing missing packages.

Tropicao, 2024-04-07:
For the record, despite the hotfix I mentioned above, I struggled like hell to make my webcam work after an upgrade (`v4l2-relayd` crashing systematically on stream start). I eventually narrowed down the issue to icamera-src current version, and circumvented it by targeting the same version as the `icamerasrc-git.old` still present in the repository. I plan to check whether this issue has been identified upstream, and if not, open the corresponding issue

Milihhard, 2024-04-08:
Thanks for the information!. Cleaning the repository and the package manager, then switching to the old fix for icamerasrc seems to work.

ragu-manjegowda, 2024-04-09:
Thank you! Switching to old fix for icamerasrc worked for me as well. 

If anyone needs exact steps,
* Open `install.sh` and change `icamerasrc-git` to `icamerasrc-git-fix.old`
* Follow [this instruction](https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/77#issuecomment-2029508628) to get around compilation errors
    -  for intel-ipu6ep-camera-hal-git, edit intel-ipu6ep-camera-hal-git/PKGBUILD and add the following method:
        ```
        prepare() {
            cd $_pkgname
            sed -i "s|-D_FORTIFY_SOURCE=2| |" CMakeLists.txt
        }
        ```
    - for icamerasrc-git-fix.old, edit icamerasrc-git-fix.old/PKGBUILD and add the following method:
        ```
        prepare() {
            cd $_pkgname
            sed -i "s|-D_FORTIFY_SOURCE=2| |" src/Makefile.am
            sed -i "s|-D_FORTIFY_SOURCE=2| |" src/interfaces/Makefile.am
        }
        ```
* Make sure to uninstall `icamerasrc-git` if you previously installed with workaround. Also, ensure clean build by manually deleting cache or using appropriate build arg like `--rebuild=yes` (for paru)

kotturi-mahipal, 2024-04-11:
> Thank you! Switching to old fix for icamerasrc worked for me as well.
>
> If anyone needs exact steps,
>
>     * Open `install.sh` and change `icamerasrc-git` to `icamerasrc-git-fix.old`
>
>     * Follow [...]

Newbie here. Where should put the prepare() block in the PKGBUILD?

Milihhard, 2024-04-11:
> Newbie here. Where should put the prepare() block in the PKGBUILD?

```
prepare() {
    cd $_pkgname
    sed -i "s|-D_FORTIFY_SOURCE=2| |" src/Makefile.am
    sed -i "s|-D_FORTIFY_SOURCE=2| |" src/interfaces/Makefile.am
}
pkgver() {
    cd $_pkgname
    printf "r%s.%s" "$(git rev-list --count HEAD)" "$(git rev-parse --short HEAD)"
}

build() {
    cd "$srcdir/$_pkgname"
    export CHROME_SLIM_CAMHAL=ON
    export STRIP_VIRTUAL_CHANNEL_CAMHAL=ON
    ./autogen.sh
    make
}
```

danielksato, 2024-04-13:
I needed to pin some package versions, in addition to following the steps @kotturi-mahipal outlined. More details in #79.

elhossary, 2024-04-14:
@Tropicao 
I did exactly as you mentioned, but the compilation fails in my case:
```
/home/muhoss/git_repos/archlinux-ipu6-webcam/intel-ipu6ep-camera-hal-git/src/ipu6-camera-hal/src/core/CameraBuffer.cpp:23:10: fatal error: drm/i915_drm.h: No such file or directory
   23 | #include <drm/i915_drm.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/camhal_static.dir/build.make:580: CMakeFiles/camhal_static.dir/src/core/CameraBuffer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:356: CMakeFiles/camhal_static.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
ERROR: Failed to build/install: intel-ipu6ep-camera-hal-git
```
Any idea, what I can do?

Tropicao, 2024-04-14:
> @Tropicao I did exactly as you mentioned, but the compilation fails in my case:
>
> ``` [...]

Could you be missing `libdrm` on your system ?

elhossary, 2024-04-14:
No, It is installed already, I reinstalled it and still get that error

danielksato, 2024-04-14:
@elhossary I ran into the same issue.

I was able to make it compile by modifying the source to point to `libdrm/i915_drm.h` instead of `drm` - I suspect this might be a distro-specific issue; it could be fixed in the PKGBUILD.

HOWEVER, even though it compiled, that built version didn't work.

I resolved the problem by pointing the PKGBUILD to a specific commit as described in #79.

elhossary, 2024-04-14:
@danielksato, Thank you for the clarification

stefanpartheym, 2024-10-25:
I recently merged PR #92 which makes camera usable again on LTS kernels. At least for most or some people. For me it still only works partially. However, a workaround was provided by the PR which makes the camera also usable in the browser.
Please refer to #92 and #90 for further details.

closed 2024-10-25 by stefanpartheym