stefanpartheym/archlinux-ipu6-webcam #79: New versions of upstream dependencies break the driver.

danielksato, 2024-04-13:
I just recently tried to update to some of the latest upstream releases that purport to include support for kernels 6.7 and 6.8. All I succeeded in doing was breaking everything!

I was able to get my webcam working again by following the steps in #77 to use `icamerasrc-git-fix.old`. I added the `prepare` section to its PKGBUILD, as well as `intel-ipu6ep-hal-git`.

In addition, I had to pin the version of `intel-ipu6ep-hal-git` to commit `da2e2821244f21b95bcb37a1271bf73360c4669e`, and the version of `intel-ipu6-dkms-git` to commit `cfb7af1e50a9f04e06d6bb77ea2340eb78350859`.

This got me up and running again too. As a bonus, Cheese works for me now, where it didn't before.

elhossary, 2024-04-14:
Kindly, Can you make a pull request for it?

danielksato, 2024-04-14:
I'm not sure if codifying this workaround is the way to go, since there's been a lot of activity upstream and I'm hoping working kernel 6.7+ support is right around the corner. But that's a question for the maintainers, I suppose; I'll try to make a PR tonight. I'll need to figure out how to use a custom PKGBUILD for `intel-ipu6-dkms-git` since right now we're using the one from the AUR.

elhossary, 2024-04-14:
In principle, this repo itself a wokaround for IPU6 cameras, as these cams are not working out of the box. It is crazy how fragile the drivers and the setup are on each update.
I suggested that to have at least something known to work, may be the maintainers make another branch out of your pull request.
I will personally clone the PR to have my cam working.
Thanks in advance BTW

danielksato, 2024-04-14:
@elhossary I submitted a PR if you wanna take a peek. I copied the edited PKGBUILD from my AUR helper and edited it to reflect the conflict it creates between my `intel-ipu6-dkms-git` package and the AUR version. However, I don't really want test it locally since it might mean breaking a working install :D

elhossary, 2024-04-15:
@danielksato Thank you for the PR. I will test that

derkrasseleo, 2024-04-17:
I also ran into this build issue 
```
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<command-line>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
make[2]: *** [CMakeFiles/camhal_static.dir/build.make:76: CMakeFiles/camhal_static.dir/modules/ia_css/ipu6ep/src/ia_css_terminal.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:356: CMakeFiles/camhal_static.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
ERROR: Failed to build/install: intel-ipu6ep-camera-hal-git

```

marmistrz, 2024-04-17:
I also needed to add `PKG_CONFIG_PATH=/usr/lib/ipu_adl/pkgconfig/`, otherwise `ia_imaging` was not found by CMake.

wille, 2024-05-07:
Running `danielksato/fixes-for-recent-releases-77-79` on `6.6.29-1-lts` on my Thinkpad X1 Carbon gen 11 without problems.

closed 2024-05-20 by danielksato