stefanpartheym/archlinux-ipu6-webcam #84: usage of mainline driver?

thepiwo, 2024-05-28:
Hello, I was wondering if there is any plan (if even possible?) to adapt this to work with the mainline driver available from 6.10? https://github.com/intel/ipu6-drivers/issues/22#issuecomment-2115692411

stefanpartheym, 2024-05-28:
Hi @thepiwo, not sure if I fully understood your question.
From what I read, if you're running kernel v6.10.x you won't need this repository anyways, as the driver is already mainlined. If you're running a LTS kernel however, you can use this repo. The mainline driver won't be compatible with the LTS kernel as far is I know.

stefanpartheym, 2024-05-28:
I just updated the README and added information about the mainline IPU6 drivers

closed 2024-05-28 by stefanpartheym

thepiwo, 2024-05-28:
Thank you for the feedback, I'm sorry I'm not to much into the details and inner working of everything thats needed to get it to work end-to-end.

As far as I understood this repo does both, provide the driver and necessary patches, plus also the integration to expose the feed to be used in applications like browsers.

Whereas the kernel only provides the drivers and distributions will have to provide the integrations to expose it.

So I was wondering if this repository could also provide guidance to whoever intends to use the mainline drivers before this integration has happened.

If I'm wrong with my understanding of this I'm sorry. Anyways I appreciate the time and effort that has been put into this.

stefanpartheym, 2024-05-29:
Hi @thepiwo,

I think you know more than I do, tbh. I have no idea about the inner working of everything too. 
I'm sure however, that the patches in this repo won't be relevant to the `6.10.x` kernels.
In my understanding, the integration part primarily lies in the `v4l2` service. I assumed this would then work out of the box on a `6.10.x` kernel. I could be completely wrong though, as I didn't have the chance yet to try it on a new kernel.

danielksato, 2024-05-29:
@stefanpartheym I'm not sure this repo won't still be relevant, as icamerasrc doesn't seem to play nicely with the latest versions that supposedly add support for 6.7+. I hope I'm wrong.

stefanpartheym, 2024-05-29:
@danielksato You could be right. I actually don't know. I think it is best to wait for a 6.10.x kernel release to test against.

danielksato, 2024-05-29:
Absolutely. I've tested the latest versions against 6.9.x and I haven't been able to get them to work at all, but I'm also very much not a kernel developer. I *thought* the problem was icamerasrc though.

wille, 2024-06-02:
@stefanpartheym You can install the  `linux-mainline` AUR package to run 6.10-rc1, I have but with no positive results, just acts like on any 6.x kernel without this repo installed.

catwell, 2024-06-21:
Same here, I tried with `6.10.0-rc4-1-mainline` from [miffe's repository](https://wiki.archlinux.org/title/Unofficial_user_repositories#miffe) and I could not make the webcam work. It is not detected by v4l2.

```
$ sudo v4l2-ctl --list-devices
Cannot open device /dev/video0, exiting.
```

marmistrz, 2024-06-29:
`linux-mainline` doesn't include `CONFIG_VIDEO_INTEL_IPU6` by default yet. I enabled it manually by patching the `config` file from `linux-mainline`, but - unfortunately - the ipu6 devices remain unusable.