stefanpartheym/archlinux-ipu6-webcam #85: Uninstall script not working and installation is broken after manual cleanup

wille, 2024-06-02:
Might have ended up in a bad state.
Removing all of the mentioned packages manually works, but you need to carefully remove them in order, because some packages depends on another which depends on another etc.

```
Removed "/etc/systemd/system/multi-user.target.wants/v4l2-relayd.service".
error: target not found: icamerasrc-git
 -> exit status 1
checking dependencies...
error: failed to prepare transaction (could not satisfy dependencies)
:: removing intel-ipu6ep-camera-hal-git-fix breaks dependency 'ipu6-camera-hal' required by icamerasrc-git-fix
 -> exit status 1
checking dependencies...
error: failed to prepare transaction (could not satisfy dependencies)
:: removing intel-ipu6ep-camera-bin-fix breaks dependency 'intel-ipu6ep-camera-bin' required by intel-ipu6ep-camera-hal-git-fix
 -> exit status 1
error: target not found: intel-ipu6-dkms-git
 -> exit status 1
checking dependencies...
error: failed to prepare transaction (could not satisfy dependencies)
:: removing intel-ivsc-firmware breaks dependency 'intel-ivsc-firmware' required by intel-ipu6-dkms-git-fix
 -> exit status 1
checking dependencies...

Packages (1) v4l2-relayd-r42.6fd6b6a-1
```

After manually removing the packages, I try to install and reboot, but not getting v4l2 to start again

I've tried 6.6.31-lts, 6.6.32-lts and finally downgraded to 6.6.29-lts which I know was working, 

```
Jun 02 11:58:06 x11 (sh)[1483]: v4l2-relayd.service: Referenced but unset environment variable evaluates to an empty string: SPLASHSRC
Jun 02 11:58:06 x11 sh[1485]: grep: /sys/devices/virtual/video4linux/*/name: No such file or directory
Jun 02 11:58:06 x11 (sh)[1483]: g_source_remove: assertion 'tag > 0' failed
Jun 02 11:58:06 x11 (sh)[1483]: g_source_remove: assertion 'tag > 0' failed
Jun 02 11:58:06 x11 (sh)[1483]: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
Jun 02 11:58:06 x11 (sh)[1483]: gst_object_unref: assertion 'object != NULL' failed
Jun 02 11:58:06 x11 (sh)[1483]: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed
Jun 02 11:58:06 x11 (sh)[1483]: gst_object_unref: assertion 'object != NULL' failed
Jun 02 11:58:06 x11 systemd[1]: v4l2-relayd.service: Deactivated successfully.
```

fcapano, 2024-06-06:
Same issue here. I'm unable to use the webcam now, even after downgrading the linux and linux-headers packages. Is anyone aware of other dependencies that should also be downgraded?

wille, 2024-06-10:
The install script will force install the latest linux-xxx-headers package, So I ended up with latest headers but an older kernel, I manually uncommented that line in the install script, it's what I did what back but forgot about it this time.

stefanpartheym, 2024-10-25:
Please refer to #90 and #92.

closed 2024-10-25 by stefanpartheym