stefanpartheym/archlinux-ipu6-webcam #86: Dell XPS 13 9340 on Manjaro 6.6.32-1

AlexiZ, 2024-06-07:
Hi,

I am trying to get the camera working on XPS 9340 with Manjaro 6.6.32-1 but can't find any way to do it.
As [issue 78](https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/78) suggests, I also tried to build `icamerasrc-git` to said commit but it did nothing much. His Dell is different than this one, 9320 instead of 9340, which can explain the failure.

Result of `./install.sh` seems fine but `$ ./test.sh` fails : `WARNING: erroneous pipeline: no element "icamerasrc"`.

Camera kinda seems to exist (?) as `$ sudo dmesg | grep -i camera` said :
<pre>
[   13.563217] intel-ipu6 0000:00:05.0: IPU camera mask = 0x0
[   13.580327] intel-ipu6 0000:00:05.0: Connected 1 cameras
</pre>

Can you help me with this please ?

Detailed outputs : 
<details>
<summary><code>$ ./install.sh</code></summary>
<pre>
# Install headers for: linux
# Install package from the AUR/repos: linux-headers
Sync Explicit (1): linux419-headers-4.19.315-1
avertissement : linux419-headers-4.19.315-1 est à jour -- ignoré
 il n’y a rien à faire
=> SUCCESS
# Wayland detected or explicitly requested. Installing 'gst-plugins-bad'.
# Install package from the AUR/repos: base-devel
Sync Explicit (1): base-devel-1-2
avertissement : base-devel-1-2 est à jour -- ignoré
 il n’y a rien à faire
=> SUCCESS
# Install package from the AUR/repos: intel-ivsc-firmware
 -> intel-ivsc-firmware-r10.10c214f-1 est à jour -- omission
 il n’y a rien à faire
=> SUCCESS
# Build and install package: intel-ipu6-dkms-git
~/Téléchargements/archlinux-ipu6-webcam/intel-ipu6-dkms-git ~/Téléchargements/archlinux-ipu6-webcam
==> Création du paquet intel-ipu6-dkms-git r165.cfb7af1e5-1 (ven. 07 juin 2024 15:22:38)
==> Vérification des dépendances pour l’exécution…
==> Vérification des dépendances pour la compilation…
==> Récupération des sources…
  -> Mise à jour du dépôt ipu6-drivers git…
==> Validation des fichiers source avec sha256sums…
    ipu6-drivers ... Ignoré
==> Extraction des sources…
  -> Création d’une copie de travail du dépot ipu6-drivers git…
Remise à zéro de la branche 'makepkg'
==> Lancement de prepare()…
Clonage dans 'ivsc-driver'...
remote: Enumerating objects: 382, done.
remote: Counting objects: 100% (178/178), done.
remote: Compressing objects: 100% (100/100), done.
remote: Total 382 (delta 90), reused 113 (delta 57), pack-reused 204
Réception d'objets: 100% (382/382), 153.78 Kio | 1.42 Mio/s, fait.
Résolution des deltas: 100% (148/148), fait.
==> Lancement de pkgver()…
==> ATTENTION : Un paquet a déjà été compilé, installation du paquet existant…
==> Installation du paquet intel-ipu6-dkms-git-fix avec pacman -U…
chargement des paquets…
avertissement : intel-ipu6-dkms-git-fix-r165.cfb7af1e5-1 est à jour -- ignoré
 il n’y a rien à faire
~/Téléchargements/archlinux-ipu6-webcam
=> SUCCESS
# Build and install package: intel-ipu6ep-camera-bin
~/Téléchargements/archlinux-ipu6-webcam/intel-ipu6ep-camera-bin ~/Téléchargements/archlinux-ipu6-webcam
==> Création du paquet ipu6-camera-bin r79.987b09a-1 (ven. 07 juin 2024 15:22:41)
==> Vérification des dépendances pour l’exécution…
==> Vérification des dépendances pour la compilation…
==> Récupération des sources…
  -> Mise à jour du dépôt ipu6-camera-bins git…
[... 157 lines cut ...]
0000:00:07.3 PCI bridge: Intel Corporation Meteor Lake-P Thunderbolt 4 PCI Express Root Port #3 (rev 10)
0000:00:08.0 System peripheral: Intel Corporation Device 7e4c (rev 20)
0000:00:0a.0 Signal processing controller: Intel Corporation Device 7d0d (rev 01)
0000:00:0b.0 Processing accelerators: Intel Corporation Meteor Lake NPU (rev 04)
0000:00:0d.0 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 USB Controller (rev 10)
0000:00:0d.2 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 NHI #0 (rev 10)
0000:00:0d.3 USB controller: Intel Corporation Meteor Lake-P Thunderbolt 4 NHI #1 (rev 10)
0000:00:0e.0 RAID bus controller: Intel Corporation Volume Management Device NVMe RAID Controller Intel Corporation
0000:00:12.0 Serial controller: Intel Corporation Device 7e45 (rev 20)
0000:00:14.0 USB controller: Intel Corporation Meteor Lake-P USB 3.2 Gen 2x1 xHCI Host Controller (rev 20)
0000:00:14.2 RAM memory: Intel Corporation Device 7e7f (rev 20)
0000:00:15.0 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #0 (rev 20)
0000:00:15.2 Serial bus controller: Intel Corporation Meteor Lake-P Serial IO I2C Controller #2 (rev 20)
0000:00:16.0 Communication controller: Intel Corporation Device 7e70 (rev 20)
0000:00:1c.0 PCI bridge: Intel Corporation Device 7e3f (rev 20)
0000:00:1e.0 Communication controller: Intel Corporation Meteor Lake-P Serial IO UART Controller #0 (rev 20)
0000:00:1f.0 ISA bridge: Intel Corporation Device 7e02 (rev 20)
0000:00:1f.3 Multimedia audio controller: Intel Corporation Meteor Lake-P HD Audio Controller (rev 20)
0000:00:1f.4 SMBus: Intel Corporation Meteor Lake-P SMBus Controller (rev 20)
0000:00:1f.5 Serial bus controller: Intel Corporation Meteor Lake-P SPI Controller (rev 20)
0000:71:00.0 Network controller: Intel Corporation Wi-Fi 7(802.11be) AX1775*/AX1790*/BE20*/BE401/BE1750* 2x2 (rev 1a)
10000:e0:06.0 PCI bridge: Intel Corporation Device 7e4d (rev 20)
10000:e1:00.0 Non-Volatile memory controller: KIOXIA Corporation NVMe SSD Controller XG8 (rev 01)
</pre>
</details>

hazimimdnazri, 2024-06-25:
Had a similiar  ```  could not link camerasrc0 to videoconvert0 ``` error with my XPS 9340 too. Has the camera able to work now? If yes, what is the solution?

AlexiZ, 2024-06-25:
@hazimimdnazri I haven't tried again since this post, camera still not working on my end too.

stefanpartheym, 2024-10-11:
Hi @AlexiZ,
are you still having issues? You may want to try with a newer LTS kernel. For instance: `6.6.56-1-lts`.
Best regards
Stefan

AlexiZ, 2024-10-12:
Hi @stefanpartheym,

I am running on kernel LTS 6.6.52-1, which is not  the .56 you are mentioning but I don't have the option to get this one for some reason. Maybe it's irrelevant though, since it's still the LTS.

Camera still doesn't work on my end. I tried to run the install script again from a fresh clone of your repository without success.
Install script breaks while trying to build / install `intel-ipu6ep-camera-bin` package :
```
error: transaction validation failed (file conflict)
intel-ipu6ep-camera-bin-fix: /usr/lib/firmware/intel/ipu/ipu6_fw.bin already exists in the file system (owned by linux-firmware)
intel-ipu6ep-camera-bin-fix: /usr/lib/firmware/intel/ipu/ipu6ep_fw.bin already exists in the file system (owned by linux-firmware)
intel-ipu6ep-camera-bin-fix: /usr/lib/firmware/intel/ipu/ipu6epadln_fw.bin already exists in the file system (owned by linux-firmware)
intel-ipu6ep-camera-bin-fix: /usr/lib/firmware/intel/ipu/ipu6epmtl_fw.bin already exists in the file system (owned by linux-firmware)
```

I tried several changes to make it work but they all failed :
- adding `--nocheck` option to `makepkg` command in install.sh:L.11
- playing with `intel-ipu6-camera-bin/PKGBUILD` to ignore already installed package
- manually installing `intel-ipu6ep-camera-bin` with yay => same error as above

I am not familiar with PKGBUILD and don't know how to solve this, even with the help of ChatGPT...

Can you help me on this please ?

stefanpartheym, 2024-10-12:
Hi @AlexiZ,

uff, honestly I have no idea, why you get this error. Seems to me like there is an issue with some of the packages you installed.
The error message says, the files are owned by `linux-firmware` package. I could be wrong, but suspect the `linux-firmware` package already contains upstreamed drivers for 6.10.x kernels. Really this is just a wild wild guess. Maybe you need to downgrade `linux-firmware`.

AlexiZ, 2024-10-14:
Hi @stefanpartheym,
It looks like your guess is right : [gitlab.com/kernel-firmware/linux-firmware/intel/ipu](https://gitlab.com/kernel-firmware/linux-firmware/-/tree/main/intel/ipu?ref_type=heads)
But I can't mess with package unfortunately : [archlinux.org/packages/linux-firmware](https://archlinux.org/packages/core/any/linux-firmware/)
I don't have the solution right now but when I do, I'll post it here.
Thank you for your time anyway :)

hazimimdnazri, 2024-10-14:
@stefanpartheym do you think it will work on fresh arch installation with kernel `6.6.56-1-lts` ?

stefanpartheym, 2024-10-14:
> Hi @stefanpartheym, It looks like your guess is right : [gitlab.com/kernel-firmware/linux-firmware/intel/ipu](https://gitlab.com/kernel-firmware/linux-firmware/-/tree/main/intel/ipu?ref_type=heads) [...]

Maybe you can try to radically rename the driver files temporarily. This would enable the you to install the AUR package. This way you could at least try if it works.

stefanpartheym, 2024-10-14:
> @stefanpartheym do you think it will work on fresh arch installation with kernel `6.6.56-1-lts` ?

yes

AlexiZ, 2024-10-14:
@stefanpartheym Your idea raised my hope and I tried it. Renaming .bin files worked to pass previous error ! I also had to make a few changes here and there.
Later in the process, I came across this new error : 

```
--   Package 'ia_imaging', required by 'virtual:world', not found
CMake Error at cmake/FindIA_IMAGING.cmake:28 (message):
  IA_IMAGING not found
Call Stack (most recent call first):
  CMakeLists.txt:285 (find_package)
```

Someone mentioned installation of [pkgconf](https://man.archlinux.org/man/pkgconf.1.en) here : [github.com/intel/ipu6-camera-hal/#18](https://github.com/intel/ipu6-camera-hal/issues/18), which I did but as you can tell, it wasn't very helpful.

Any hint on this one ?

PS : I'll share everything I did when we're done, to get relevant parts only.

stefanpartheym, 2024-10-17:
I have the exact same issue now too.

alexhulbert, 2024-10-18:
Same here, I tried a whole bunch of things, using many different combinations of earlier commits, etc, but I'm still getting stuck at this call to [find_package_handle_standard_args](https://github.com/intel/ipu6-camera-hal/blob/a2de9c2a4d1d1be5d6493350c8860810f7568352/cmake/FindIA_IMAGING.cmake#L86). A user on the Arch Linux forum mentioned that they were able to get a specific commit with specific patches working on a laptop with the same architecture & camera sensor as me using a patched version of Fedora Linux w/ kernel 6.11. However, I couldn't get it to build under arch.

[Here's the forum post](https://bbs.archlinux.org/viewtopic.php?pid=2198775#p2198775)

And the relevant rpm spec files for [ipu6-camera-bin](https://build.opensuse.org/projects/home:ateixeira:xps13/packages/ipu6-camera-bins/files/ipu6-camera-bins.spec?expand=1) and [intel-ipu6-camera-hal](https://build.opensuse.org/projects/home:ateixeira:xps13/packages/ipu6-camera-hal/files/ipu6-camera-hal.spec?expand=1).

I'm guessing that these spec files are doing something the PKGBUILD isn't, but I can't seem to figure out what. I'm on the latest kernel btw, so maybe my situation is different from all of you since I'm guessing most people here are using the LTS build.

stefanpartheym, 2024-10-19:
Ok, so i got it partially working on the latest LTS kernel again. However, I can't really use it, because it does not work in my browsers (neither Chrome/Chromium/Brave nor Firefox).
It works with the `test.sh` script though.
All I did is, I pinned the commits of the packages `intel-ipu6ep-camera-bin-fix` and `v4l2loopback-dkms-git-fix` as follows:
```diff
diff --git a/intel-ipu6ep-camera-bin/PKGBUILD b/intel-ipu6ep-camera-bin/PKGBUILD
index d80ecd4..3ec6d7d 100644
--- a/intel-ipu6ep-camera-bin/PKGBUILD
+++ b/intel-ipu6ep-camera-bin/PKGBUILD
@@ -14,7 +14,7 @@ depends=('intel-ipu6-dkms-git' 'glibc')
 makedepends=('git')
 provides=(ipu6-camera-bin intel-ipu6ep-camera-bin)
 conflicts=(intel-ipu6-camera-bin intel-ipu6ep-camera-bin)
-source=("git+${url}.git")
+source=("git+${url}.git#commit=af5ba0cb4a763569ac7514635013e9d870040bcf")
 sha256sums=('SKIP')
 
 pkgver() {
diff --git a/v4l2loopback-dkms-git/PKGBUILD b/v4l2loopback-dkms-git/PKGBUILD
index 1b9ddeb..d55ec83 100644
--- a/v4l2loopback-dkms-git/PKGBUILD
+++ b/v4l2loopback-dkms-git/PKGBUILD
@@ -13,7 +13,7 @@ depends=('dkms')
 makedepends=('git' 'help2man')
 conflicts=("${_pkgbase}-dkms" "${_pkgbase}-dkms-git")
 provides=("${_pkgbase}-dkms")
-source=("git://git.launchpad.net/ubuntu/+source/v4l2loopback#branch=ubuntu/devel")
+source=("git://git.launchpad.net/ubuntu/+source/v4l2loopback#commit=0e961819fc46f98017af02e84b7973106edd12f1")
 md5sums=('SKIP')
 
 pkgver() {
@@ -35,7 +35,10 @@ package() {
     cd "${srcdir}/${_pkgbase}"
     mkdir -p "${pkgdir}/usr/share/licenses/${_pkgbase}"
     cp -vf COPYING "${pkgdir}/usr/share/licenses/${_pkgbase}"
-    make DESTDIR="${pkgdir}" PREFIX="/usr" install-utils install-man
+    #make DESTDIR="${pkgdir}" PREFIX="/usr" install-utils #install-man
+    # NOTE: Builing and installing man pages fails for some reason, so I just
+    # dont build and install them.
+    make DESTDIR="${pkgdir}" PREFIX="/usr" install-utils
     mkdir -p "${pkgdir}/usr/src/${_pkgbase}-${pkgver}"
     cp -ar * "${pkgdir}/usr/src/${_pkgbase}-${pkgver}"
 }
```

When I run `./test.sh`, the camera works. I get also some errors, which seem to not have any negative impact on the camera itself:
```
Setting pipeline to PAUSED ...
[10-19 22:35:41.727] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-19 22:35:41.728] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-19 22:35:41.729] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-19 22:35:41.730] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev16 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock

(gst-launch-1.0:31895): GStreamer-CRITICAL **: 22:35:41.776: Registering meta implementation 'GstCamerasrcMeta' without init function
[10-19 22:35:41.776] CamHAL[ERR] sensor output sub device is not set
Redistribute latency...
ERROR: from element /GstPipeline:pipeline0/GstWaylandSink:waylandsink0: Output window was closed
Additional debug info:
../gstreamer/subprojects/gst-plugins-bad/ext/wayland/gstwaylandsink.c(858): on_window_closed (): /GstPipeline:pipeline0/GstWaylandSink:waylandsink0
Execution ended after 0:00:06.860112642
Setting pipeline to NULL ...
Enter ia_aiq_get_aiqd_data()
Out-aiqd data size: 41480
[10-19 22:35:48.738] CamHAL[WAR] Failed to open file /run/camera/ov01a10-uf_VIDEO.aiqd, error No such file or directory
Freeing pipeline ...
```

stefanpartheym, 2024-10-25:
It should work on LTS kernels now for most people with PR #92 now merged.
Please refer to issue #90 and PR #92 for further information.

closed 2024-10-25 by stefanpartheym

AlexiZ, 2024-10-28:
@stefanpartheym sadly, it does not work yet. Install script works fine now but test script shows this error : `ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.`.
Here is the full output :
<details>
<summary>./test.sh</summary>
<pre>
[10-28 09:12:01.447] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.447] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.447] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.447] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.447] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.447] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.447] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.447] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.448] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.448] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.449] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.449] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.449] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.449] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE eExposure time range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.450] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.450] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.451] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.451] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.451] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.451] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.451] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.451] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.451] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.451] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.451] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.451] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.451] CamHAL[ERR] Malformed ET range in exposure time range configuration
[10-28 09:12:01.451] CamHAL[ERR] Parse AE gain range failed
[10-28 09:12:01.452] CamHAL[ERR] Malformed ET range in exposure time range configuration
[... 144 lines cut ...]
[10-28 09:12:02.325] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-28 09:12:02.325] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[... cut ...]
[10-28 09:12:02.332] CamHAL[ERR] @configure Device Configure failed
[10-28 09:12:02.332] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.004131971
Setting pipeline to NULL ...
Freeing pipeline ...
</pre>
</details>

I did not find any helpful tip in #90.

hazimimdnazri, 2024-11-02:
@AlexiZ I don't think XPS 9340 will work, seems our sensor HM1092 is still not being supported.

alexhulbert, 2024-11-12:
> @AlexiZ I don't think XPS 9340 will work, seems our sensor HM1092 is still not being supported.

@hazimimdnazri @AlexiZ 

HM1092 is actually just the IR sensor. The camera itself is supported.

hazimimdnazri, 2024-11-12:
@alexhulbert Ah really? I did try to get the sensor of XPS 9340 from [this](https://hansdegoede.dreamwidth.org/28841.html) and it returns the HM1092.

alexhulbert, 2024-11-12:
@hazimimdnazri  https://hansdegoede.dreamwidth.org/28841.html?thread=5289#cmt5289

hazimimdnazri, 2024-11-13:
@alexhulbert Thank you for the clarification! So based on your comment and what I've read on the article the ivsc driver is the culprit? AFAIK the camera is working on ubuntu because oem kernel is being used. Are you using xps 9340 as well?

alexhulbert, 2024-11-14:
> @alexhulbert Thank you for the clarification! So based on your comment and what I've read on the article the ivsc driver is the culprit? AFAIK the camera is working on ubuntu because oem kernel is being used. Are you using xps 9340 as well?

Yeah exactly, I emailed Hans a while back and he said the following:

> The normal sensor not being visible under /sys/bus/i2c/devices typically is caused by something being amiss with the Intel VSC driver which needs to probe the iVSC chip successfully before the main sensor will show up under /sys/bus/i2c/devices

He couldn't give me anything beyond that though, since he's focusing on Fedora support.

As for the Ubuntu OEM kernel, I actually compiled the kernel with the relevant changes in my test. I followed [this post](https://bbs.archlinux.org/viewtopic.php?pid=2198775#p2198775) and tried to get the out-of-tree ipu6 drivers working on my laptop but i ran into issues compiling the specified commit of ipu6-camera-hal, even without the patches.