stefanpartheym/archlinux-ipu6-webcam #88: v4l2-relayd crashes (segfault) on 6.6.39-1-lts

gfarrell, 2024-07-15:
I upgraded my kernel to 6.6.39 (I'm on the LTS kernel) and suddenly the camera stopped working. I have "virtual camera" available, as long as v4l2-relayd.service is running, but the output is black, camera light never turns on, and I get the following in dmesg:

```
[  +0.193526] [Mon Jul 15 16:35:12 2024] v4l2-relayd[39547]: segfault at 0 ip 00007e6ba45f20c0 sp 00007fff7c298c60 error 4 in libc.so.6[7e6ba457b000+16c000] likely on CPU 7 (core 12, socket 0)
[  +0.000009] [Mon Jul 15 16:35:12 2024] Code: 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 42 55 48 89 e5 41 54 49 89 f4 53 48 89 fb e8 20 f3 ff ff <48> 8b 03 85 c0 74 19 48 8d 50 ff f0 48 0f b1 13 75 0e 5b 31 c0 41
[  +0.613429] [Mon Jul 15 16:35:12 2024] v4l2-relayd[39841]: segfault at 0 ip 00007b799c4c70c0 sp 00007fff0968b490 error 4 in libc.so.6[7b799c450000+16c000] likely on CPU 4 (core 8, socket 0)
[  +0.000008] [Mon Jul 15 16:35:12 2024] Code: 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 42 55 48 89 e5 41 54 49 89 f4 53 48 89 fb e8 20 f3 ff ff <48> 8b 03 85 c0 74 19 48 8d 50 ff f0 48 0f b1 13 75 0e 5b 31 c0 41
[  +0.717135] [Mon Jul 15 16:35:13 2024] v4l2-relayd[39904]: segfault at 0 ip 00007d3d7cbf30c0 sp 00007ffec27ff2c0 error 4 in libc.so.6[7d3d7cb7c000+16c000] likely on CPU 4 (core 8, socket 0)
[  +0.000010] [Mon Jul 15 16:35:13 2024] Code: 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 42 55 48 89 e5 41 54 49 89 f4 53 48 89 fb e8 20 f3 ff ff <48> 8b 03 85 c0 74 19 48 8d 50 ff f0 48 0f b1 13 75 0e 5b 31 c0 41
[  +0.760907] [Mon Jul 15 16:35:14 2024] v4l2-relayd[39941]: segfault at 0 ip 00007ac1085a60c0 sp 00007ffeac2091a0 error 4 in libc.so.6[7ac10852f000+16c000] likely on CPU 6 (core 12, socket 0)
[  +0.000008] [Mon Jul 15 16:35:14 2024] Code: 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 42 55 48 89 e5 41 54 49 89 f4 53 48 89 fb e8 20 f3 ff ff <48> 8b 03 85 c0 74 19 48 8d 50 ff f0 48 0f b1 13 75 0e 5b 31 c0 41
[  +0.720784] [Mon Jul 15 16:35:15 2024] v4l2-relayd[39976]: segfault at 0 ip 0000701a60f570c0 sp 00007ffcbbb524e0 error 4 in libc.so.6[701a60ee0000+16c000] likely on CPU 5 (core 8, socket 0)
[  +0.000010] [Mon Jul 15 16:35:15 2024] Code: 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 42 55 48 89 e5 41 54 49 89 f4 53 48 89 fb e8 20 f3 ff ff <48> 8b 03 85 c0 74 19 48 8d 50 ff f0 48 0f b1 13 75 0e 5b 31 c0 41
[  +0.772841] [Mon Jul 15 16:35:15 2024] v4l2-relayd[40042]: segfault at 0 ip 00006ffc8b3450c0 sp 00007ffd792f2780 error 4 in libc.so.6[6ffc8b2ce000+16c000] likely on CPU 7 (core 12, socket 0)
[  +0.000008] [Mon Jul 15 16:35:15 2024] Code: 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 42 55 48 89 e5 41 54 49 89 f4 53 48 89 fb e8 20 f3 ff ff <48> 8b 03 85 c0 74 19 48 8d 50 ff f0 48 0f b1 13 75 0e 5b 31 c0 41
[  +0.765359] [Mon Jul 15 16:35:16 2024] v4l2-relayd[40075]: segfault at 0 ip 000070c413dde0c0 sp 00007ffe2a7194d0 error 4 in libc.so.6[70c413d67000+16c000] likely on CPU 6 (core 12, socket 0)
[  +0.000009] [Mon Jul 15 16:35:16 2024] Code: 66 2e 0f 1f 84 00 00 00 00 00 f3 0f 1e fa 48 81 7e 08 ff c9 9a 3b 77 42 55 48 89 e5 41 54 49 89 f4 53 48 89 fb e8 20 f3 ff ff <48> 8b 03 85 c0 74 19 48 8d 50 ff f0 48 0f b1 13 75 0e 5b 31 c0 41
```

I tried uninstalling and reinstalling (which worked fine) but it still doesn't work.

gfarrell, 2024-07-15:
If I boot into the 6.9.9 kernel (non-LTS), then v4l2-relayd _does not_ crash, but the camera still doesn't work, and instead if I look at the journal (`journalctl -u v4l2-relayd -f`) when trying to use the webcam, I a lot of this sort of output:

```
ul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[ERR] invalid media format, default value used.
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[ERR] invalid media format, default value used.
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[ERR] invalid media format, default value used.
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[ERR] invalid media format, default value used.
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[ERR] invalid media format, default value used.
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
Jul 15 16:47:01 anaximander sh[1125]: [07-15 16:46:20.270] CamHAL[INF] aiqb file name AR0234_TGL_10bits.aiqb
```

gfarrell, 2024-07-17:
Inexplicably, after shutting down my 6.9.9 session and rebooting back into 6.6.39-1-lts, the camera works. I haven't changed anything (and did try several reboots before going into 6.9.9 to no avail).

I'm not sure if it's still interesting to leave this open in case anyone can see anything useful in the segfaults etc., or if I should just close it.

stefanpartheym, 2024-10-11:
Hi @gfarrell,

weird, but I can relate. Sometimes camera refuses to work after a fresh install. Then rebooting (several) times "fixes" the issue.

I'll close this one.

Best regards
Stefan

closed 2024-10-11 by stefanpartheym

stefanpartheym, 2024-10-25:
Hi @gfarrell ,

This issue should now be fixed with d1c471caf4099e25d392ad23a3410706692a8701 for v4l2-relayd on Wayland using one of the latest LTS kernels.