stefanpartheym/archlinux-ipu6-webcam #90: Webcam not working on LTS kernel

alba4k, 2024-08-12:
[image]

<details>

<summary>test.sh output</summary>

```
Enter Password or Place finger on fingerprint reader: 
          
Setting pipeline to PAUSED ...
[08-12 13:11:31.66] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-12 13:11:31.67] CamHAL[INF] aiqb file name ov01a10.aiqb
[08-12 13:11:31.68] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[08-12 13:11:31.68] CamHAL[ERR] GetControl: Device node /dev/v4l-subdev16 IOCTL VIDIOC_G_EXT_CTRLS error: Invalid argument
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:08.512265603
Setting pipeline to NULL ...
Caught SIGSEGV
#0  0x00007081ee50ec70 in ?? () from /usr/lib/libc.so.6
#1  0x00007081ee515b68 in pthread_mutex_lock () from /usr/lib/libc.so.6
#2  0x00007081ee8d8bf6 in gst_pad_stop_task () from /usr/lib/libgstreamer-1.0.so.0
#3  0x00007081ee048e91 in ?? () from /usr/lib/gstreamer-1.0/libgsticamerasrc.so
#4  0x00007081ee04f671 in ?? () from /usr/lib/gstreamer-1.0/libgsticamerasrc.so
#5  0x00007081ee8c9c1c in ?? () from /usr/lib/libgstreamer-1.0.so.0
#6  0x00007081ee8ca919 in gst_pad_set_active () from /usr/lib/libgstreamer-1.0.so.0
#7  0x00007081ee8aec9d in ?? () from /usr/lib/libgstreamer-1.0.so.0
#8  0x00007081ee8bb3fd in gst_iterator_fold () from /usr/lib/libgstreamer-1.0.so.0
#9  0x00007081ee934bbf in ?? () from /usr/lib/libgstreamer-1.0.so.0
#10 0x00007081ee8aed69 in ?? () from /usr/lib/libgstreamer-1.0.so.0
#11 0x00007081ee8af03e in ?? () from /usr/lib/libgstreamer-1.0.so.0
#12 0x00007081ee04cb4d in ?? () from /usr/lib/gstreamer-1.0/libgsticamerasrc.so
#13 0x00007081ee03699c in ?? () from /usr/lib/gstreamer-1.0/libgsticamerasrc.so
#14 0x00007081ee8ae026 in gst_element_change_state () from /usr/lib/libgstreamer-1.0.so.0
#15 0x00007081ee8ae932 in ?? () from /usr/lib/libgstreamer-1.0.so.0
#16 0x00007081ee88468e in ?? () from /usr/lib/libgstreamer-1.0.so.0
#17 0x00007081ee8ae026 in gst_element_change_state () from /usr/lib/libgstreamer-1.0.so.0
#18 0x00007081ee8ae0b9 in gst_element_change_state () from /usr/lib/libgstreamer-1.0.so.0
#19 0x00007081ee8ae932 in ?? () from /usr/lib/libgstreamer-1.0.so.0
#20 0x0000577b3a39aad3 in ?? ()
#21 0x00007081ee4a3e08 in ?? () from /usr/lib/libc.so.6
#22 0x00007081ee4a3ecc in __libc_start_main () from /usr/lib/libc.so.6
#23 0x0000577b3a398055 in ?? ()
Spinning.  Please run 'gdb gst-launch-1.0 18129' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
^C⏎                                                                                          
```

</details>

<details>

<summary>systemctl status v4l2-relayd</summary>

```
● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: disabled)
     Active: active (running) since Mon 2024-08-12 13:08:43 CEST; 5min ago
 Invocation: 1489ff05b19b409689846b696981338d
    Process: 1047 ExecCondition=/usr/bin/test -n ${VIDEOSRC} (code=exited, status=0/SUCCESS)
    Process: 1111 ExecCondition=/usr/bin/test -n $FORMAT (code=exited, status=0/SUCCESS)
    Process: 1137 ExecCondition=/usr/bin/test -n $WIDTH (code=exited, status=0/SUCCESS)
    Process: 1155 ExecCondition=/usr/bin/test -n $HEIGHT (code=exited, status=0/SUCCESS)
    Process: 1172 ExecCondition=/usr/bin/test -n $FRAMERATE (code=exited, status=0/SUCCESS)
    Process: 1182 ExecCondition=/usr/bin/test -n ${CARD_LABEL} (code=exited, status=0/SUCCESS)
   Main PID: 1190 (v4l2-relayd)
      Tasks: 5 (limit: 18691)
     Memory: 42.3M (peak: 45.8M swap: 672K swap peak: 2.5M zswap: 102.8K)
        CPU: 439ms
     CGroup: /system.slice/v4l2-relayd.service
             └─1190 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! v4l2sink name=v4l2sink device=>

ago 12 13:08:43 dell-xps systemd[1]: Starting v4l2-relay daemon service...
ago 12 13:08:43 dell-xps systemd[1]: Started v4l2-relay daemon service.
ago 12 13:08:43 dell-xps (sh)[1190]: v4l2-relayd.service: Referenced but unset environment variable evaluates to an empty string: SPLASHSRC
```

</details>

nathanael-h, 2024-09-04:
Hello, I downgraded to a previous linux-lts kernel and headers that are supported (cf. README) and it works (at least in Chromium and Firefox. 

`sudo pacman -U https://archive.archlinux.org/packages/l/linux-lts/linux-lts-6.6.31-1-x86_64.pkg.tar.zst https://archive.archlinux.org/packages/l/linux-lts-headers/linux-lts-headers-6.6.31-1-x86_64.pkg.tar.zst` 

Be careful when downgrading package, here is the doc https://wiki.archlinux.org/title/Arch_Linux_Archive#How_to_downgrade_one_package :wink:

alba4k, 2024-09-23:
I mean, sure, but it's weird that they won't work on the latest lts (currently 6.6.52)

alba4k, 2024-10-11:
Let me try.

alba4k, 2024-10-11:
```

  -> Creazione di una copia di lavoro del repository ipu6-camera-bins git in corso...
Clone in 'ipu6-camera-bins' in corso...
fatto.
==> Avvio di pkgver() in corso...
==> Versione aggiornata: ipu6-camera-bin r84.98ca6f2-1
==> Entro nell'ambiente fakeroot...
==> Avvio di package() in corso...
cp: impossibile eseguire stat di 'ipu6-camera-bins/lib/ipu_adl/pkgconfig': File o directory non esistente
==> ERRORE: Si è verificato un errore in package().
    L'operazione sta per essere interrotta...
ERROR: Failed to build/install: intel-ipu6ep-camera-bin
```
while building, not sure why ?

stefanpartheym, 2024-10-11:
Hmm, thats weird. 

I checked my local repo for any changes:
```diff
diff --git a/intel-ipu6ep-camera-bin/PKGBUILD b/intel-ipu6ep-camera-bin/PKGBUILD
index d80ecd4..29f4b54 100644
--- a/intel-ipu6ep-camera-bin/PKGBUILD
+++ b/intel-ipu6ep-camera-bin/PKGBUILD
@@ -4,7 +4,7 @@
 pkgbase=ipu6-camera-bin
 pkgname=intel-ipu6ep-camera-bin-fix
 _pkgname=ipu6-camera-bins
-pkgver=r73.af5ba0c
+pkgver=r79.987b09a
 pkgrel=1
 pkgdesc="Intel IPU6 camera binaries (Alder Lake)"
 arch=('x86_64')
```

It took a different `pkgver` for `ipu6-camera-bin`. I think this happens automatically with `yay`. Maybe you need the "new" `pkgver`.

But honestly, I have no idea...

alba4k, 2024-10-11:
I honestly don't care enough about the webcam to worry about this now ahahaha

stefanpartheym, 2024-10-12:
Hahaha yeah don't worry :smile:

hazimimdnazri, 2024-10-17:
@stefanpartheym having similar issue with this on fresh arch installation with kernel `6.6.56-1-lts`

emilrueh, 2024-10-19:
Hi, to chip in a bit, I'm on x1 carbon gen10 and kernel 6.11.4 so I thought to roll back to LTS kernel 6.6.57-1-lts.

When running `./install.sh` I get this:

`ERROR: Failed to build/install: intel-ipu6ep-camera-bin`

struan-robertson, 2024-10-21:
@stefanpartheym is this still working for you on kernel 6.6.57-1-lts? I have got all the packages building for the latest commits on their repos, but without a working camera. If it is still working for you on the latest lts kernel, we could use the package versions you have installed and it should work for everyone.

stefanpartheym, 2024-10-21:
@struan-robertson As mentioned in https://github.com/stefanpartheym/archlinux-ipu6-webcam/issues/86#issuecomment-2418748836, I have the exact same issue now. No camera working even on LTS kernel.

victornoel, 2024-10-24:
Hi @stefanpartheym, about that error `cp: cannot stat 'ipu6-camera-bins/lib/ipu_adl/pkgconfig': No such file or directory`, the reason is that they finally fixed that "shit" (cf the comment in the PKGBUILD: https://github.com/stefanpartheym/archlinux-ipu6-webcam/blob/main/intel-ipu6ep-camera-bin/PKGBUILD#L30).

Still, even with fixing this (I just removed the `ipu_adl` related lines), I'm getting now the following error from the kernel:
```
➜  ~ sudo dmesg| grep ipu
[   22.180146] intel-ipu6 0000:00:05.0: enabling device (0000 -> 0002)
[   22.180746] intel-ipu6 0000:00:05.0: Device 0x465d (rev: 0x6)
[   22.180773] intel-ipu6 0000:00:05.0: physical base address 0x6076000000
[   22.180775] intel-ipu6 0000:00:05.0: mapped as: 0x00000000c64ee155
[   22.181294] intel-ipu6 0000:00:05.0: Unable to set secure mode
[   22.181297] intel-ipu6 0000:00:05.0: IPU in non-secure mode
[   22.181299] intel-ipu6 0000:00:05.0: IPU secure touch = 0x0
[   22.181301] intel-ipu6 0000:00:05.0: IPU camera mask = 0xff
[   22.181819] intel-ipu6 0000:00:05.0: Skip ipc reset for non-secure mode
[   22.181822] intel-ipu6 0000:00:05.0: IPC reset done
[   22.181823] intel-ipu6 0000:00:05.0: cpd file name: intel/ipu6ep_fw.bin
[   22.183790] intel-ipu6 0000:00:05.0: Direct firmware load for intel/ipu6ep_fw.bin failed with error -2
[   22.183795] intel-ipu6 0000:00:05.0: Requesting signed firmware failed
[   22.184784] intel-ipu6: probe of 0000:00:05.0 failed with error -2
```

The firmware is present as expected (from that intel-ipu6ep-camera-bin-fix package) but I noticed that linux-firmware was also providing similar file (but compressed). So I tried not to include the firmware files from intel-ipu6ep-camera-bin-fix package but the error is the same.

I'm not sure what to do more now.

struan-robertson, 2024-10-24:
I have everything working for me now and have created a pull request #92. Hopefully it works for everyone :)

Mengesh, 2024-10-24:
I can confirm that it works on `Lenovo ThinkPad X1 Carbon Gen 10` kernel `6.6.58-1-lts`.

Thank you!!!

stefanpartheym, 2024-10-25:
Partially works for me: `test.sh` gives me camera output, but using the webcam in the browser crashes `v4l2-relayd` service.
But with the suggested workaround from @struan-robertson I can also use my webcam in the browser (without v4l2-relayd):
`sudo gst-launch-1.0 icamerasrc ! video/x-raw,format=NV12,width=1280,height=720 ! videoconvert ! v4l2sink name=v4l2sink device=/dev/video0`

I merged PR #92 .

stefanpartheym, 2024-10-25:
Ok, nevermind. Fixed the v4l2-relayd service file as suggested in #71 with d1c471c.
It seems as this was only an issue on Wayland, so I guess @struan-robertson and @Mengesh you guys are on X11 – right?

Anyways, please test the latest commit with your latest LTS kernel and let me know if it works for you.

struan-robertson, 2024-10-25:
@stefanpartheym strangely no, I'm on Wayland also. I use Sway so maybe it or its underlying library wlroots were not susceptible to the crash. 

Anyway, glad it is all working for you! I have tested the latest commit and all works for me as well.

stefanpartheym, 2024-10-25:
@struan-robertson Ok that's actually weird. But I'm using GNOME, so maybe that is the issue. Anyways, thanks for your PR 
 :)

Will close this issue, once enough feedback is available. @alba4k Let me know if it works for you – thanks!

alba4k, 2024-10-25:
Sure, I wasn't home this week so I couldn't check this out

will let you know this evening or tomorrow

alba4k, 2024-10-25:
Also, since I haven't used this in a while, are the flags for hibernation workaround and format conversion still needed? or does hibernation work fine now

victornoel, 2024-10-25:
I just tested latest main branch on a XPS 13 (9315) and it works well

alba4k, 2024-10-25:
with no install flags?

victornoel, 2024-10-25:
> with no install flags?

If the question is for me, then no, just `./install.sh` after removing any already built package to start fresh

ferpizza, 2024-10-25:
I'm getting some partial success on an ThinkPad X1 Gen 11th

Installed with `./install.sh -as`
Then rebooted twice, and still not working

So, getting inspiration on other issues, I started cycling the `ov2740` kernel module, running this command as `root`
```
modprobe -r ov2740 && modprobe ov2740
```

I had to run this command a number of times (between 5 and 10, maybe more) until I get
```
# modprobe -r ov2740 && modprobe ov2740
modprobe: FATAL: Module ov2740 is in use.
```

At this time, I have to restart the V4L Relay service with
```
# systemctl restart v4l2-relayd.service
```

And then, it work as expected

This survived rebooting, but it didn't survive a full poweroff ... so I have to run this on each new boot.

Not ideal, but much better than the non-working cammera situation I was on.

Any ideas on how to make this easier are appreciated :)

alba4k, 2024-10-25:
mind checking wether -a and/or -s are still needed?

ferpizza, 2024-10-25:
@alba4k , if that question is for me, not sure how I would do that (don't really want to mess this up)

I was just following comments on other closed Issues related to X1, but some time ago I got it working without any flags (but that time didn't stick) ... so not sure they are at all needed, but in my case there was no harm in using them

alba4k, 2024-10-25:
@stefanpartheym test.sh works, obs and firefox both get a black image

stefanpartheym, 2024-10-25:
> I'm getting some partial success on an ThinkPad X1 Gen 11th

@ferpizza How do you test, whether your camera works or not? 
I recommend always trying `test.sh` first, before checking in any browsers and other apps.

stefanpartheym, 2024-10-25:
> @stefanpartheym test.sh works, obs and firefox both get a black image

@alba4k
From what I learned, this seems to be an issue with `v4l2-relayd`.
I had the same issue earlier this day. That's why I committed d1c471caf4099e25d392ad23a3410706692a8701 as it was suggested in #71 .
Have you checked systemctl status `v4l2-relayd` or dmesg output?

ferpizza, 2024-10-25:
In addition to `./test.sh` I've been using https://webcamtests.com/ from a browser.

stefanpartheym, 2024-10-25:
Ok, so `./test.sh` gives you the expected camera output right?
Seems like the same issue that @alba4k has too. As mentioned in my comment above, I had the same issue earlier this day.
Please check the output of `systemctl status v4l2-relayd`.

alba4k, 2024-10-27:
Nothing unusual appears in that output

<details>
<summary>systemctl status v4l2-relayd</summary>

```
● v4l2-relayd.service - v4l2-relay daemon service
     Loaded: loaded (/usr/lib/systemd/system/v4l2-relayd.service; enabled; preset: disabled)
    Drop-In: /etc/systemd/system/v4l2-relayd.service.d
             └─override.conf
     Active: active (running) since Sun 2024-10-27 10:55:00 CET; 5min ago
 Invocation: 68df1e4b9888403eb119d4ea80f03d66
    Process: 930 ExecCondition=/usr/bin/test -n ${VIDEOSRC} (code=exited, status=0/SUCCESS)
    Process: 965 ExecCondition=/usr/bin/test -n $FORMAT (code=exited, status=0/SUCCESS)
    Process: 972 ExecCondition=/usr/bin/test -n $WIDTH (code=exited, status=0/SUCCESS)
    Process: 994 ExecCondition=/usr/bin/test -n $HEIGHT (code=exited, status=0/SUCCESS)
    Process: 1011 ExecCondition=/usr/bin/test -n $FRAMERATE (code=exited, status=0/SUCCESS)
    Process: 1022 ExecCondition=/usr/bin/test -n ${CARD_LABEL} (code=exited, status=0/SUCCESS)
   Main PID: 1035 (v4l2-relayd)
      Tasks: 5 (limit: 18697)
     Memory: 66.8M (peak: 81.7M)
        CPU: 229ms
     CGroup: /system.slice/v4l2-relayd.service
             └─1035 /usr/bin/v4l2-relayd -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 ! videoconvert ! video/x-raw,format=YUY2 ! v4l2>

ott 27 10:55:00 dell-xps systemd[1]: Starting v4l2-relay daemon service...
ott 27 10:55:00 dell-xps systemd[1]: Started v4l2-relay daemon service.
ott 27 10:55:00 dell-xps (sh)[1035]: v4l2-relayd.service: Referenced but unset environment variable evaluates to an empty string: SPLASHSRC
```

</details>

struan-robertson, 2024-10-27:
@alba4k what output do you get if you stop the v4l2-relayd service and then run `sudo GST_DEBUG=2 v4l2-relayd -d -i icamerasrc -o "appsrc name=appsrc caps=video/x-ra
w,format=NV12,width=1280,height=720,framerate=30/1 is-live=true ! queue ! videoconvert !
 v4l2sink name=v4l2sink device=/dev/video0"`

alba4k, 2024-10-28:
the following

<details>

```
0:00:00.094159661 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.131205422 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.236314899 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.295865131 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.296132213 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.296282363 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.296464763 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.296646363 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.296770424 35471 0x603d7c54a380 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.371346243 35470 0x5948f34807c0 ERROR           GST_PIPELINE subprojects/gstreamer/gst/parse/grammar.y:629:gst_parse_element_make: no element "w"
0:00:00.371366189 35470 0x5948f34807c0 WARN                 default subprojects/gstreamer/gst/parse/grammar.y:1566:priv_gst_parse_yyerror: Error during parsing: syntax error
0:00:00.371371134 35470 0x5948f34807c0 ERROR           GST_PIPELINE subprojects/gstreamer/gst/parse/grammar.y:1490:priv_gst_parse_yyparse: syntax error
0:00:00.371375385 35470 0x5948f34807c0 ERROR           GST_PIPELINE subprojects/gstreamer/gst/parse/grammar.y:1490:priv_gst_parse_yyparse: syntax error
0:00:00.371379729 35470 0x5948f34807c0 ERROR           GST_PIPELINE subprojects/gstreamer/gst/parse/grammar.y:1490:priv_gst_parse_yyparse: syntax error
0:00:00.371383626 35470 0x5948f34807c0 ERROR           GST_PIPELINE subprojects/gstreamer/gst/parse/grammar.y:1490:priv_gst_parse_yyparse: syntax error
0:00:00.371387267 35470 0x5948f34807c0 ERROR           GST_PIPELINE subprojects/gstreamer/gst/parse/grammar.y:1490:priv_gst_parse_yyparse: syntax error
0:00:00.419233237 35470 0x7c3098000f50 WARN            videodecoder gstvideodecoder.c:2799:gst_video_decoder_chain:<pngdec0> Received buffer without a new-segment. Assuming timestamps start from 0.
0:00:00.422006628 35470 0x7c3098001130 WARN             imagefreeze gstimagefreeze.c:1166:gst_image_freeze_src_loop:<imagefreeze0> error: Internal data stream error.
0:00:00.422024546 35470 0x7c3098001130 WARN             imagefreeze gstimagefreeze.c:1166:gst_image_freeze_src_loop:<imagefreeze0> error: streaming stopped, reason not-linked (-1)
0:00:00.422198448 35470 0x5948f34807c0 ERROR            V4L2_RELAYD src/v4l2-relayd.c:146:backend_pipeline_bus_call: Internal data stream error.
```

</details>

struan-robertson, 2024-10-28:
Sorry I accidentally added a space in the command. Try running `sudo GST_DEBUG=2 v4l2-relayd -d -i icamerasrc -o "appsrc name=appsrc caps=video/x-raw,format=NV12,width=1280,height=720,framerate=30/1 is-live=true ! queue ! videoconvert ! v4l2sink name=v4l2sink device=/dev/video0"`

alba4k, 2024-10-28:
will do in 10h or so once I'm back home

hazimimdnazri, 2024-10-28:
I got this when running ./test.sh

```
Setting pipeline to PAUSED ...
[10-29 03:21:20.960] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-29 03:21:20.960] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-29 03:21:20.961] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov13b10.aiqb
[10-29 03:21:20.961] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov8856.aiqb
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov8856.aiqb
[10-29 03:21:20.961] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov8856.aiqb
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov8856.aiqb
[10-29 03:21:20.961] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-29 03:21:20.961] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-29 03:21:20.962] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.962] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-29 03:21:20.962] CamHAL[INF] aiqb file name ov01a10.aiqb
[10-29 03:21:20.962] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.962] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-29 03:21:20.962] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-29 03:21:20.963] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_1BG203N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_1SG204N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV02C10_CIFME14_ADL.aiqb
[10-29 03:21:20.963] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] aiqb file name OV2740_CJFLE23_ADL.aiqb
[10-29 03:21:20.963] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-29 03:21:20.963] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.963] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-29 03:21:20.964] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HM2170_1SG205N3_ADL.aiqb
[10-29 03:21:20.964] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HM2170_CJFME18_ADL.aiqb
[10-29 03:21:20.964] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HI556_1BG502T3_ADL.aiqb
[10-29 03:21:20.964] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[10-29 03:21:20.964] CamHAL[INF] aiqb file name HI556_CJFLE25_ADL.aiqb
[10-29 03:21:20.964] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.964] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-29 03:21:20.964] CamHAL[INF] aiqb file name ov01a1s.aiqb
[10-29 03:21:20.965] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.965] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[10-29 03:21:20.965] CamHAL[INF] aiqb file name OV08A10_YHUT_ADL.aiqb
[10-29 03:21:20.965] CamHAL[ERR] invalid media format, default value used.
[10-29 03:21:20.965] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-29 03:21:20.965] CamHAL[INF] aiqb file name IMX390_HDR_TGL.aiqb
[10-29 03:21:20.965] CamHAL[ERR] invalid media format, default value used.
[... 61 lines cut ...]
[10-29 03:21:20.967] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.967] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.967] CamHAL[INF] (null), media format in tuning: 0, media format for aic 0.
[10-29 03:21:20.967] CamHAL[ERR] Failed to find DevName for cameraId: 0, get video node: ov13b10 , devname: /dev/v4l-subdev1
Pipeline is live and does not need PREROLL ...
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
[10-29 03:21:20.973] CamHAL[ERR] Get entity fail for calling getEntityById
[... cut ...]
[10-29 03:21:20.973] CamHAL[ERR] @configure Device Configure failed
[10-29 03:21:20.973] CamHAL[ERR] failed to config streams.
ERROR: from element /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0: src pad: Internal data flow error.
Additional debug info:
gstcambasesrc.cpp(3143): gst_cam_base_src_loop (): /GstPipeline:pipeline0/Gstcamerasrc:camerasrc0:
streaming task paused, reason not-negotiated (-4)
Execution ended after 0:00:00.003655492
Setting pipeline to NULL ...
Freeing pipeline ...
```

Dell XPS 9340, any idea?

struan-robertson, 2024-10-28:
@hazimimdnazri looks like an issue on the intel side of things. What do you get if you run `sudo dmesg | grep ipu6`

struan-robertson, 2024-10-28:
So for people who are unable to run the `test.sh`, sorry I dont think there is much that can be done. As mentioned in #89, you may now be able to get it working on the latest kernel. Also it might be worth trying the `latest_commits` branch from my [fork of the repo](https://github.com/struan-robertson/archlinux-ipu6-webcam/tree/latest_commits), which contains the latest commits from the relevant Intel repositories. This did not work for my webcam, by you may have success on different sensors.

@Mengesh I'm unsure why it would stop working after resuming from suspend. I do notice that your dmesg output says `IPU in secure mode`, whereas mine says `IPU in non-secure mode` and is works after suspending.

alba4k, 2024-10-29:
@struan-robertson not sure if still relevant but this is what I got

<details>

```
0:00:00.095423029 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.134893389 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.221714351 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.274610014 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.274931741 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.275077104 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.275281257 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.275470186 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.275588178 664228 0x602b94952780 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.418046483 664227 0x6034fd5ef640 WARN                 default v4l2-utils.c:189:gst_v4l2_error:<v4l2sink> error: Device '/dev/video0' is not a output device.
0:00:00.418065253 664227 0x6034fd5ef640 WARN                 default v4l2-utils.c:192:gst_v4l2_error:<v4l2sink> error: Capabilities: 0x5200001
0:00:00.418117862 664227 0x6034fd5ef640 ERROR            V4L2_RELAYD src/v4l2-relayd.c:392:output_pipeline_bus_call: Device '/dev/video0' is not a output device.

(v4l2-relayd:664227): GLib-CRITICAL **: 18:34:59.678: g_source_remove: assertion 'tag > 0' failed

(v4l2-relayd:664227): GLib-CRITICAL **: 18:34:59.678: g_source_remove: assertion 'tag > 0' failed

(v4l2-relayd:664227): GStreamer-CRITICAL **: 18:34:59.678: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed

(v4l2-relayd:664227): GStreamer-CRITICAL **: 18:34:59.678: gst_object_unref: assertion 'object != NULL' failed

(v4l2-relayd:664227): GStreamer-CRITICAL **: 18:34:59.678: gst_element_set_state: assertion 'GST_IS_ELEMENT (element)' failed

(v4l2-relayd:664227): GStreamer-CRITICAL **: 18:34:59.678: gst_object_unref: assertion 'object != NULL' failed
```

</details>

alba4k, 2024-10-29:
Right, sorry, my bad forgot about that

<details>

```
0:00:00.065804982 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.091794977 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.164586620 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.212129136 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.212387543 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.212532788 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.212710730 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:515:register_plugin: Failed to initialize plugin
0:00:00.212874388 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.213009507 785467 0x5c15b0528650 WARN                  frei0r gstfrei0r.c:485:register_plugin: Failed to load plugin
0:00:00.315580227 785466 0x798aa0000f50 WARN            videodecoder gstvideodecoder.c:2799:gst_video_decoder_chain:<pngdec0> Received buffer without a new-segment. Assuming timestamps start from 0.
0:00:00.328259195 785466 0x798aa0000b90 WARN           v4l2allocator gstv4l2allocator.c:276:gst_v4l2_memory_group_new:<v4l2sink:pool0:sink:allocator> Driver pretends buffer 0 is queued even if freshly created, this indicates a bug in the driver.
0:00:00.328292837 785466 0x798aa0000b90 WARN           v4l2allocator gstv4l2allocator.c:276:gst_v4l2_memory_group_new:<v4l2sink:pool0:sink:allocator> Driver pretends buffer 1 is queued even if freshly created, this indicates a bug in the driver.
0:00:00.328306714 785466 0x798aa0000b90 WARN          v4l2bufferpool gstv4l2bufferpool.c:853:gst_v4l2_buffer_pool_start:<v4l2sink:pool0:sink> Uncertain or not enough buffers, enabling copy threshold
0:00:00.333605031 785466 0x57d4de3d1f30 WARN             V4L2_RELAYD src/v4l2-relayd.c:376:output_pipeline_bus_call: V4L2_EVENT_PRI_CLIENT_USAGE not supported

[...]
```

</details>

struan-robertson, 2024-10-29:
@alba4k you are not running the patched version of v4l2-relayd. The patch [0001-Fix-V4L2_EVENT_PRI_CLIENT_USAGE-to-use-value-used-up.patch](https://github.com/stefanpartheym/archlinux-ipu6-webcam/blob/main/v4l2-relayd/0001-Fix-V4L2_EVENT_PRI_CLIENT_USAGE-to-use-value-used-up.patch) changes the macro defining `V4L2_EVENT_PRI_CLIENT_USAGE` so that it aligns with upstream `v4l2loopback-dkms`. 

Make sure that you are on the latest commit of this repository, then run `clean.sh` to delete all built `.zst` package files. Just to be sure things get replaced, I would also run `uninstall.sh` to remove all the old packages from your system. Once that has done, reboot, then run `install.sh` and things should all work :)

alba4k, 2024-10-29:
uhm yeah webcam works in obs and firefox too now

thanks I guess

closed 2024-10-29 by alba4k

Mengesh, 2024-10-29:
> @Mengesh I'm unsure why it would stop working after resuming from suspend. I do notice that your dmesg output says `IPU in secure mode`, whereas mine says `IPU in non-secure mode` and works after suspending.

It’s a tough guess, but could this be because my system is encrypted? 

I'm suspending to RAM (deep sleep). Could that be a problem?

struan-robertson, 2024-10-29:
Strange thing is my system is also encrypted, plus signed for secureboot. I tried temporarily disabling secureboot but it had no change. 
I think the main problem is that the Intel drivers are a buggy mess. Later versions with bug fixes will not be compatible with the 6.6 LTS kernel but also potentially not compatible with the latest kernel, since development seems quite Ubuntu focused. 
Your best bet is probably to try progressively later commits of all the packages and see if they work or not ,unfortunately. If you do this make sure to try and use commits from around the same time period as the packages need to be compatible with each other.

alba4k, 2024-10-29:
my system is also signed for sb but not encrypted (which doesnt really make sense, I know, but whatever)

Mengesh, 2024-10-29:
> Your best bet is probably to try progressively later commits of all the packages and see if they work or not ,unfortunately.

Oh man, I'd rather reboot before a meeting :sweat_smile: and hope the things are going to progress with 6.10+ kernel as I've already had partial success using [the guides from arch forums](https://bbs.archlinux.org/viewtopic.php?id=297262&p=3).