stefanpartheym/archlinux-ipu6-webcam #97: Latitude 7640, EndeavourOS

Martian-0007, 2025-04-11:
It all seemed nice and pretty until it all failed on some CMake thingy (see error below)

```
CMake Error at CMakeLists.txt:17 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway.

-- Configuring incomplete, errors occurred!
==> ERROR: A failure occurred in build().
    Aborting...
ERROR: Failed to build/install: intel-ipu6ep-camera-hal-git
```

I think the error message indicates that the script allows CMake versions earlier than 3.5, which was removed from CMake.

OS: EndeavourOS x86_64
Host: Latitude 7640
Kernel: Linux 6.14.1-arch1-1
CPU: 13th Gen Intel(R) Core(TM) i7-1365U (12) @ 5.20 GHz
GPU: Intel Iris Xe Graphics @ 1.30 GHz [Integrated]

huyhoang8398, 2025-04-29:
did you handle to fix this? @Martian-0007

Martian-0007, 2025-04-30:
Yes, actually!
Just not using the tool in this repository.
I ended up asking in the EndeavourOS Telegram channel, where I got pointed to [this Arch forum topic](https://bbs.archlinux.org/viewtopic.php?id=297262&p=4) (which I read before, just not far enough 😅), which fixed it for me.

## TL;DR
This is the post that ended up containing the required information:
https://bbs.archlinux.org/viewtopic.php?pid=2213350#p2213350

The dkms AUR is current and up-to-date.

All I needed to do was clean the yay cache, install the `intel-ipu6-dkms-git`, and load the driver.
```bash
rm -rf ~/.cache/yay/intel-ipu6-dkms-git/ 
yay intel-ipu6-dkms-git
sudo modprobe -v ov02c10 # Specific for each model - this worked for my Latitude 7640
```

## The extended version
Multiple posts in multiple places showed ways of checking which camera sensor you have and verifying with the Intel driver repo. (This is all in the forum post as well.)
This way, I discovered the camera sensor has been implemented, giving me hope!

After some more digging, I found the outdated AUR and almost gave up since I had already spent hours on it.
However, I decided to give the Telegram channel a shot, and I was pointed back to the Arch forums post, which I scraped before. 
There, I found this very detailed post with a guide, saying the out-of-date tag is false and that the new version of the drivers should be built.

Since I had no experience with Linux drivers, I checked out the PKGBUILD in the AUR and tried to understand it: 
- It builds directly from the [Intel GitHub repo](https://github.com/intel/ipu6-drivers), so it's safe to just use yay.
- The hardcoded revision number (used only by a helper to identify it) is obsolete, and the package has been marked as out-of-date on the AUR. **Neither of these is true.**   

The out-of-date label is believed to have been added in confusion, as it seemed like all IPU6 drivers were merged into the Linux kernel - now we wouldn't be here if that were true, would we?
The drivers existed, and the AUR package was usable. So, I installed it and activated it using modprobe, and voila, it worked!

closed 2025-04-30 by Martian-0007

Martian-0007, 2025-04-30:
No. Only _some_ drivers were merged into the kernel, but not *all*. That means you still have to install it.

I highly recommend reading the post. It explains the topic really well.

Martian-0007, 2025-04-30:
Also, I switched kernels between solving the issue and today. You must install all the dependencies listed in the post on every kernel, since some get built using dracut.

```fish
yay -Sy gst-plugin-libcamera pipewire-libcamera libcamera-tools libcamera-ipa libcamera intel-ivsc-firmware
reboot
```

Martian-0007, 2025-05-01:
And are you having this issue with the `intel-ipu6-dkms-git` or `archlinux-ipu6-webcam`?

I only needed the `intel-ipu6-dkms-git` and its dependencies.

huyhoang8398, 2025-05-01:
with intel-ipu6-dkms-git i can have webcam via gs-launch, but not in browser or any otherapp, i dont know if i need this https://aur.archlinux.org/packages/intel-ipu6ep-camera-hal-git ( this i cannot compile due to cmake )

huyhoang8398, 2025-05-01:
@Martian-0007 can you list me your package that you installed to have camera working?

Martian-0007, 2025-05-01:
sure thing
```
gst-plugin-libcamera 
pipewire-libcamera 
libcamera-tools 
libcamera-ipa 
libcamera 
intel-ivsc-firmware
```
and then `intel-ipu6-dkms-git`

Regarding the browser, this is what the post says:
> gstreamer can use the camera, too, but only with sudo:
> `$ sudo gst-launch-1.0 libcamerasrc ! queue ! glimagesink`
> For the first time I see a picture, hooray! But since it only works as root, I must apply the camera udev rules, as bjesus and others explained: I created a file at /etc/udev/rules.d/camera.rules with this line in it:
> `KERNEL=="udmabuf", TAG+="uaccess"`
> After another reboot, now it works without sudo.
> Cheese worked out of the box.
> Firefox worked after setting media.webrtc.camera.allow-pipewire = true in [about:config](http://about:config) and restarting.
> For Chrome, I enabled "PipeWire Camera support" in chrome://flags/#enable-webrtc-pipewire-camera. Chrome detects the camera, but all I get is a black picture. I have to investigate this further. So far, nobody seems to have this problem, but I admit the combination Wayland+Plasma+Chrome+Pipewire+MIPI camera with ov02e10 sensor is very special.

I use Firefox and didn't try it without this config. Haven't tried any other app yet.

huyhoang8398, 2025-05-01:
look like same as me, i'm be able with gst-launch but not any other app ( mostly chromium and electron app ),

huyhoang8398, 2025-05-01:
what kernel you're using?

Martian-0007, 2025-05-02:
I am using the Zen kernel:
```fish
> uname -r
6.14.4-zen1-1-zen
```

huyhoang8398, 2025-05-02:
well i guess i gave up to setup webcam to work with my dell 9320

sALTaccount, 2025-06-01:
If anybody else is reading this thread, I wasn't able to get my dell XPS 13+ 9320 to work with v4l2. However, libcamera seems to be working just fine (the colors are pretty horrible though). You can use libcamerify to launch an application with a compatibility layer that converts the v4l2 to libcamera.

You can test if this works for you by launching an application such as firefox or discord through the command line with `libcamerify <app>`, or you can test the camera itself with `qcam`. Just as a side note, this does not appear to work for cheese or OBS.

If it works properly, you can change your desktop icons to actually open libcamerify with the argument to launch the application, such as so

[image]